bigtreetech/BIGTREETECH-TouchScreenFirmware #1112: Ability to Scroll GCode Command Display Window

ptmy1, 2020-09-20:
**Is your feature request related to a problem? Please describe.**
After issuing a GCode command using Terminal in the TFT35 touchscreen, a display window is shown.  This window shows the GCode command issued and response received from the printer firmware.  Information often appears and disappears too quickly to be read.  

**Describe the solution you'd like**
The ability to scroll back the display window.  Also, the ability to switch back and forth between the command window and display window.  Currently, if I accidentally touch the display window, system goes back to showing the command window.  The only way to show the display window again is to issue another GCode command.

**Describe alternatives you've considered**
Send GCode commands using OctoPi

**Additional context**
Add any other context or screenshots about the feature request here.

radek8, 2020-10-15:
PR #1171

oldman4U, 2020-10-20:
Hello ptmy1.

Could you please check PR #1171 like radek8 mentioned 5 days ago, if this covers your feature request.

Could you please be so kind and close this ticket after that. Your feature request is alreadycovered in the pinned Feature Request ticket #1177 

Thank you

ptmy1, 2020-10-24:
I don’t understand this.  #1177 looks like a bug fix unrelated to my feature request.

ptmy1, 2020-10-24:
I don't understand.  Is my enhancement request a duplicate of a request posted earlier?

oldman4U, 2020-10-24:
This was a typo. Sorry!

Please check #1170 but for your fr already a PR exist, right?

oldman4U, 2020-10-24:
And #1171 includes what you have been asking for. So if you compile it you get the feature. Please test 1171 and if it does what you want you can close this ticket. If not let the owner of 1171 know.

oldman4U, 2020-11-02:
And now it is in master branch. 

Please help the community and close the ticket once you do not need it anymore. Thank you

closed 2020-11-02 by ptmy1