bigtreetech/BIGTREETECH-TouchScreenFirmware #1113: [BUG] Wifi stops working when TFT35 set to Marlin Mode

chilman408, 2020-09-20:
Not sure why everytime I switch to Marlin Mode, it makes the Wifi module stop working.

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

**Actual behavior**

### Hardware Variant

### TFT Firmware Version & Main Board Firmware details

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

oldman4U, 2020-09-20:
Hi. You have the wifi connected to the tft?

Which connection do you use via wifi which is interrupted?

chilman408, 2020-09-20:
Yes, Wifi on the TFT on the wifi connector (8pin).

In Marlin mode on the TFT, the webui does not work.  It shows up but printer is not connected.

oldman4U, 2020-09-21:
Hi,

I can confirm this and I guess it is so by default. Maybe one of the more advanced developers can help us.

@AnHardt Entschuldige die Störung, but maybe you can explain to us, if this is a bug or the expected behaviour. And if so, why.

Thank you ;-)

AnHardt, 2020-09-21:
Sorry. I have no idea.
I have a bit of experience with Marlin, but not enough with the displays firmware.

oldman4U, 2020-09-21:
Ok. Thank you and sorry again.

radek8, 2020-09-21:
WiFi connected to TFT35 E3 V3
it works for me and I don't care what mode the display is in

radek8, 2020-09-21:
You need to load config.ini and set the value
serial_always_on: 1

radek8, 2020-09-21:
After that, it will also fake in Marlin mode without interruption

radek8, 2020-09-21:
You can close this ticket :-)

oldman4U, 2020-09-21:
That's right radek8!!!

Changed it and it works- Thank you

@chilman408 Please change this setting and it will work.

Please close this ticket once you do not need it anymore.

chilman408, 2020-09-21:
Ok, thx everyone.  Let me try this out when I get home tonight.

chilman408, 2020-09-22:
Ok, I will say that it does work as radek8 described.

However, I'm having a different issue now.  When the 3D Printer is turned off in Marlin (therefore will boot up in Marlin mode), neither the Marlin mode nor Touch mode would boot up correctly.

In Marlin mode, I just get a blank screen, no response at all.

If I switch to Touch mode, the TFT does not connect to the printer.

oldman4U, 2020-09-22:
Then please close this and start a new ticket.

closed 2020-09-22 by chilman408

Talha909, 2020-12-01:
> WiFi connected to TFT35 E3 V3
> it works for me and I don't care what mode the display is in

Please help me.  Do I need to change any thing in Marlin for working of wifi , I have attached wifi with the BIGTREE SKR V3 E3. i also flashed the software into the esp 8266. Its works fine but did not communicate with the printer.

radek8, 2020-12-01:
What is it BIGTREE SKR V3 E3
I have not found such a device

Talha909, 2020-12-02:
I have this one. I did nothing on marlin. Lcd working good with my marlin. I installed the esp8266 esp-01. I got the printer setting on my screen. but printer did accept my command. If mean there is no communication between esp8266 and lcd. 

https://github.com/bigtreetech/BTT-TFT35-E3-V3.0

radek8, 2020-12-02:
So from your communication I understood this:
You have a wifi module ESP-1 connected to TFT35 E3V3.0
and does not communicate with the printer. It is so?

Write what kind of board you have with Marlin.
Attach the Marlin configuration files. (configuration.h, configuration_adv.h).
Attach config.ini from TFT.
Thank you.

Talha909, 2020-12-02:
[Marlin1.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5628214/Marlin1.zip)
I have BIGTREETECH SKR Mini E3  V2 
[config.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5628274/config.zip)
for my ender. 

I have attach the files. I want to put esp wifi module into my tft screen.

oldman4U, 2020-12-02:
And if you have an ESP-1 or ESP-1S.

Talha909, 2020-12-02:
I have esp-1s. I flashed the software and also upload the files. I got the printer screen on my pc. But the tft did not accept the commands which i gave from the pc.

oldman4U, 2020-12-02:
Ok. This exact same combination works on my printers.

You get the information that the wifi is connected once the printer is turned on?

Send a real picture from Tft and also from computer screen.

Talha909, 2020-12-02:
> Ok. This exact same combination works on my printers.
> 
> You get the information that the wifi is connected once the printer is turned on?
> 
> Send a real picture from Tft and also from computer screen.

did you make change in your Marlin for that setup.  Please upload your marlin setting.  It difficult to send picture. printer is at home.

oldman4U, 2020-12-02:
Send me a dropbox or WeTransfer link.

Talha909, 2020-12-02:
> […](#)

I need your marlin config files and config.ini for the tft. I will compare that config with mine config. make the adjustment.

oldman4U, 2020-12-02:
I am not allowed to attach it here. So I need a way to send it to you.

Talha909, 2020-12-03:
> […](#)

You can email me. My address is robotics459@gmail.com

Talha909, 2020-12-03:
my email address is robotics459@gmail.com. Did you make any change in the marlin or tft screen. Or any change in pin files.

radek8, 2020-12-03:
I had to uncomment in Marlin
#define ESP3D_WIFISUPPORT
#define WIFI_SSID "WiFi SSID"
#define WIFI_PWD "WiFi Password"

Password and SSID are without quotes