bigtreetech/BIGTREETECH-TouchScreenFirmware #1114: [BUG] Fan status icon always shows 0% under certain conditions

ptmy1, 2020-09-21:
### Description

Fan status icon shows 0% after an M106 Gcode command (with no optional parameters), and the fan can't be turned off using touch screen when this happens.  However, if the M106 command is issued with an "S" parameter, everything works as expected.  For example, send "M106 S128."  Fan turns on at half speed and status icon shows 50% as expected.  Send "M106 S255."  Fan turns on at full speed, and status icon shows 100% as expected.  But send just "M106."  Fan turns on at full speed, but status icon continues to show 0%.  The only way to turn fan off at this point is to send gcode "M107." 

### Steps to reproduce

1. send gcode command M106, and observe parts cooling fan turning on at full speed, but fan speed on status screen remains at 0%.  
2. try turning fan off with touch screen, but nothing happens.
3. send gcode M107 to turn fan off
4. send gcode command M106 S255, and observer parts cooling fan turning on at full speed, and fan speed on status screen shows 100% as expected
5. turn fan off using touch screen or send gcode M107

**Expected behavior**
status icon should always show actual fan speed

**Actual behavior**
Please see above

### Hardware Variant
TFT35 E3 V3
### TFT Firmware Version & Main Board Firmware details
TFT firmware: Compiled using Master Branch
Marlin: 2.0.6.1

radek8, 2020-09-21:
I can confirm this behavior, but it is expected behavior.
Enter command M106 parameter limit to set the full power of the cooling fan
https://marlinfw.org/docs/gcode/M106.html

radek8, 2020-09-21:
The problem with updating the value on the status screen has been resolved before.
If you are using the current FW for and you have Marlin enabled
#define EXTRA_FAN_SPEED
Your value should update correctly.

radek8, 2020-09-21:
The display can ask Marlin for speed because it would set full speed when asking for M106.
That's why the option was recently added to Marlin
#define REPORT_FAN_CHANGE
which tells all the peripherals that the speed has changed.
if you do not enable it, the display will not know about the change and still shows the previous value (0%)

ptmy1, 2020-09-21:
@radek8 Uncommenting #define REPORT_FAN_CHANGE in Marlin worked as you described.  Thanks you.

closed 2020-09-21 by ptmy1

oldman4U, 2020-09-23:
I can not believe that a user closed the ticket with me asking to do so. 🤣

Thank you👍🏻👍🏻👍🏻👍🏻👍🏻👍🏻