bigtreetech/BIGTREETECH-TouchScreenFirmware #1115: [FR] Add option for temp sensors.

BlackDragon2020, 2020-09-21:
Short and sweet, I would like to add extra sensors to the printer for for things like ambient temperature, stepper driver temperatures, maybe drivers or motherboard temperature.  

I am currently looking into how to setup this source code to compile, I have some coding background, a bit rusty not used in a few years. I have 2 ideas:

1: find some sensors that will just plug in to the touch screen and just need code to read and display.
2: use a microcontroller  like an Arduino to handle the sensors. then use one of the extra uart inputs to talk back and forth.

I would love to do this, but it is a bit overarching atm maybe I could have it functional in a few months. Have to clear that rust out and oil the gears!

oldman4U, 2020-09-23:
Hi Dragon.

Developers are always very welcome;-)

The problem I can see is more, where to end ( how many additional functions you would like to get) and how they are shown on screen. Usually space is very limited on screen and iterate a single icon through more than 2-3 values will not work, because the time you have to wait for a single value to be shown will be too long. And if you show them at the same time, the values will be hard to read, especially on the smaller screens. Of course you can make a "for me only" version.

I use an external sub 15.- Euro device which allows me to see two temp. values and to control the fans with it. For me this works perfectly to turn on the extruder heater at 50ºC and the driver fan at 30ºC.

In case you still want to implement this, i would start with the interface - but I am not a developer...

Hope this helps

Happy printing

BlackDragon2020, 2020-09-24:
It would not need to be displayed, it could have its own menu to check, or seen via wifi. This way you could check on things inside an case.

I got the source code loaded, I have ran into some issue I am going to try and find if I have time, if not I will post about them if the newest code does not fix it.

oldman4U, 2020-09-24:
Understand

oldman4U, 2020-10-20:
Hi Dragon.

Could you please be so kind and close this ticket. Your feature request is covered in the pinned Feature Request ticket #1177. Let me know in case you are working on it and you believe you could get it done.

Thank you

BlackDragon2020, 2020-10-21:
> Hi Dragon.
> 
> Could you please be so kind and close this ticket. Your feature request is covered in the pinned Feature Request ticket #1170 . Let me know in case you are working on it and you believe you could get it done.
> 
> Thank you

So we are consolidating request in to one ticket for easy of searching as and tracking progress?

I have not taken on this task yet, I do plan to give it a go at some point but not sure when.

Let me know if I understand correctly his correctly and I will close it.

oldman4U, 2020-10-21:
You are 100% right. And this is done because of the two reasons mentioned
in #1170.

Thank you

oldman4U, 2020-10-22:
Please do not forget to close this ticket.

THANK YOU

closed 2020-10-22 by BlackDragon2020