bigtreetech/BIGTREETECH-TouchScreenFirmware #1117: Outdated Data

ikn369330, 2020-09-21:
### Description
Found Outdated Data error message on screen, message says 
Insert the SD card with the required files and press the reset button to update

### Steps to reproduce

1. Compiled latest Firmware for the TFT35 and added the TFT35 folders containing bmp and font folder as per instructions
2. inserted SD card with BIGTREE_TFT35_V3.0.26.x.bin file that I had compiled in Platformio]
3. TFT screen performs update and updates the firmware, bmps and fonts
4. error message appears
5. removed SD Card and formatted, then repeated the above steps from a clean install
6. same error

**Expected behaviour**
 Firmware Updated and working correctly 

**Actual behaviour**
Keeps saying Outdated Data and I can get no further response 

### Hardware Variant
 TFT35
[Files used.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5255366/Files.used.zip)

TFT Firmware Version TFT35_V3.0.26.x.26

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

ikn369330, 2020-09-21:
If I use the precompiled BIGTREE_TFT35_V3.0.26.x.bin file, I still get the same error

oldman4U, 2020-09-21:
Hello.

Unfortunately BTT is very lacy when it comes to update outdated information, even we prepare everything.

Could you please follow the instructions here and let me know if this helps you: https://github.com/oldman4U/BIGTREETECH-TouchScreenFirmware/tree/Oldman4U

Thank you

radek8, 2020-09-21:
When updating, you must have 3 things on the SD card:
bin file with update
TFT35 folder
and the cofif.ini file

So you are missing an ini file with the configuration on the card!

ikn369330, 2020-09-21:
Yes, that worked.

Thank you

oldman4U, 2020-09-21:
👍🏻

Please help the community and close the ticket once you do not need it anymore. Thank you

closed 2020-09-21 by ikn369330