bigtreetech/BIGTREETECH-TouchScreenFirmware #1118: [BUG] When serial_always_on is enabled, booting into Marlin mode seem to fail connection to printer

chilman408, 2020-09-22:
### Description

When serial_always_on:1 under config.ini, if the printer is turned off when TFT is in Marlin mode.  The next time the printer starts, it doesn't look like the TFT can connect to printer.  Under Marlin Mode (as it boots up), the screen is blank.  When switching to Touch mode, a "no printer attached" message is shown.  This requires shutting down the printer in Touch mode to get things properly working again.

### Steps to reproduce

1. Set serial_always_on:1 in config.ini and flash the TFT
2. Shut down the printer while TFT is in Marlin mode.
3. Restart the printer.

**Expected behavior**

**Actual behavior**

### Hardware Variant

### TFT Firmware Version & Main Board Firmware details

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

oldman4U, 2020-09-22:
Hi again.

Just tested it with my TFT35 E3 v3, also with WIFI module and SKR E3 DIP and I can not reproduce the issue.

Which mainboard and firmware are you using?

chilman408, 2020-09-22:
I am on Creality V2.2 mainboard.  The TFT firmware is latest, v26.

radek8, 2020-09-23:
Did you use only one 10-pin connector?
if so, is it connected to the EXP3 connector on the display?

oldman4U, 2020-09-23:
Hi radek8.

The system has been working fine incl. Marlin mode, until chilean408 activated "serial_always_on" to get the WIFI module (which is connected to the TFT) to work also after switching to Marlin mode.

Thank you for joining us;-)

radek8, 2020-09-23:
I know that, but I want to make sure it's properly connected.
EXP1 and EXP3 partially use the same pins and may work under certain circumstances if the connectors are interchanged. But EXP3 includes a reset pin and can loop through the reset pin on the UART connector.
Then Marlin would constantly reset and the display would show nothing.

chilman408, 2020-09-23:
I am using EXP1 + EXP2, not using EXP3.  Amd yes, Marlin has been working fine.

radek8, 2020-09-23:
Then I see no reason why this behavior should occur. I won't help you.

oldman4U, 2020-10-06:
Hi. What is the status of this issue?

chilman408, 2020-10-06:
Still the same.

oldman4U, 2020-10-07:
Have you tried to contact the mainboard manufacturer. Seems to works wit
BTT mainboard.

chilman408, 2020-10-07:
The main board is running fine.  To test, I could connect an Octopi via USB even when the screen crashed (when booting via Marlin Mode first).

oldman4U, 2020-10-07:
I tested Skr E3 DIP and SKR Pro v1.1 and I can not reproduce the issue. Seems no other user has the same issue so it is unlikely a TFT problem. Or there are not so many users using this kind of hardware combination. I am running Marlin 2.0.7 and latest firmware from 10 days ago.

Wish you the best

chilman408, 2020-10-07:
Thanks, no problem.  I'm thinking the issue may be the sequence or speediness with respect to trying to get a serial connection from the main board when booting in LCD simulation mode.  Because the Octopi would make this connection until 30 seconds + into the booting process.

Unfortunately, I don't have enough reasons to move to an SKR board at the moment.

oldman4U, 2020-10-07:
If everything else works I wouldn't change it also.

Please help the community and close the ticket once you do not need it anymore. Thank you