bigtreetech/BIGTREETECH-TouchScreenFirmware #1119: BTT TFT35 V3 found outdated data

SamTeixeira, 2020-09-22:
Description

Found Outdated Data error message on screen, message says
Insert the SD card with the required files and press the reset button to update

Steps to reproduce

Compiled latest Firmware for the TFT35 and added the TFT35 folders containing bmp and font folder as per instructions
inserted SD card with BIGTREE_TFT35_V3.0.26.x.bin file that I had compiled in Platformio]
TFT screen performs update and updates the firmware, bmps and fonts
error message appears
removed SD Card and formatted, then repeated the above steps from a clean install
same error
Expected behaviour
Firmware Updated and working correctly

Actual behaviour
Keeps saying Outdated Data and I can get no further response

Hardware Variant

TFT Firmware Version TFT35_V3.0.26.x.26

Additional Information

I have 3 things on the SD card:

bin file with update
TFT35 folder
config.ini file

oldman4U, 2020-09-22:
Hi.

Please read this new version of the ReadMe and let me know if this does not help.

Thank you

@bigtreetech When can we expect you to merge the pull requests waiting since days. This kind of  tickets could be avoided with a minimum of maintenance from your side.

SamTeixeira, 2020-09-22:
> Hi.
> 
> Please read this new version of the ReadMe and let me know if this does not help.
> 
> Thank you
> 
> @bigtreetech When can we expect you to merge the pull requests waiting since days. This kind of tickets could be avoided with a minimum of maintenance from your side.

Hi, thanks for your reply.

 I’ve been following step by step and keep getting the same message.

oldman4U, 2020-09-22:
How many files and folders have you copied onto the root of the SD card before you put the card into the TFT for the firmware update??

SamTeixeira, 2020-09-22:
I have 3 things on the SD card:

bin file with update
TFT35 folder
config.ini file

oldman4U, 2020-09-22:
Perfect.

And the three elements are all from the same firmware?
Are you using the precompiled firmware or home brewed using VSC and PIO?
When you do the update, do you see the firmware first, followed by the Font, BMPs next and last but not least the config.ini keyword update shown as a progress bar?

Please let me know

Thank you

SamTeixeira, 2020-09-22:
Yep. They’re all from the same firmware, home brewed, using VSC and Pio. 
It does complete the firmware update and get stuck on the font and bmp. Showing that message.
[image]

oldman4U, 2020-09-22:
What is the size and format of the sd card you use for the update?

Seems data in TFT folder is corrupted.

SamTeixeira, 2020-09-22:
It’s an 8gb fat32.

radek8, 2020-09-22:
Do the TFT35 folder contain?

SamTeixeira, 2020-09-22:
Bmp and fonts folders.

radek8, 2020-09-22:
ok

radek8, 2020-09-22:
font no fonts!

SamTeixeira, 2020-09-22:
> font no fonts!

Thanks for correcting me. Valuable contribution.

radek8, 2020-09-22:
use this bin file
[BIGTREE_TFT35_V3.0.26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5264428/BIGTREE_TFT35_V3.0.26.x.zip)

SamTeixeira, 2020-09-23:
It didn’t work out. Still getting the same message.

oldman4U, 2020-09-23:
Looks like the firmware update works properly but the TFT35 fails. Have you renamed the folder or changed anything inside?

Maybe we have to call for the storm trooper 😁

SamTeixeira, 2020-09-23:
I’ll contact the support.
Thanks for you help!

oldman4U, 2020-09-23:
Ok.

Please help the community and close the ticket once you do not need it anymore. Thank you

radek8, 2020-09-23:
did TFT35 contain this?
[Uploading TFT35.zip…]()

oldman4U, 2020-09-23:
Exactly. Here a picture which shows the content of the SD Card how it should look like before inserted into the TFT.

[image]

ETE-Design, 2020-09-23:
And you have edited the Config.ini to match your Configuration.h?

oldman4U, 2020-09-23:
Config.ini overwrite what is defined in configuration. So there is no need to match.

I would recommend to download and use it as it is to see if it works like this. Changes can be made uploading an edited config.ini later on (without the need to upload new firmware or TFT folder).

radek8, 2020-09-23:
use the FTP35.zip I sent

oldman4U, 2020-09-23:
Usually I use self compiled firmware, but I wanted to give it a try and used the pre compiled firmware.

I was able to update my 2 TFT35 E3 v3 without any problem.

When you do so, make sure to use the E§ version in case you have a E3 TFT. In case it is a TFT35 v3, make sure to use the TFT35 v3 version.

Maybe it is also worth to try the 128MB card which came with the mainboard with a SD to micro SD adapter.

closed 2020-09-24 by SamTeixeira

oldman4U, 2020-09-24:
What solved the problem. Please let us know 

Thank you

Bokoshi, 2020-09-26:
I had the same issue, During update it informs me that there are icon files missing. Once the update finishes it ends up on the outdated data screen and is unable to progress.
I was trying the precompiled 'normal' version.
However when I used the 'Unified Menu Material theme' version the entire installation completes without failure.

oldman4U, 2020-09-26:
@Bokoshi 

Please open a ticket and attach all information and pictures.

radek8, 2020-09-26:
Why, he solved the problem :-)

oldman4U, 2020-09-26:
We do not, know but the user closed the ticket.

chepo92, 2020-10-29:
Same issue with precompiled bins in "Copy to SD Card root directory to update" folder, hardware TFT35 E3 V3

oldman4U, 2020-10-29:
Try self compiled firmware.

bwachtel89, 2021-02-25:
Think i found the issue people are having. I was having the same exact issue until i switched to an 8gb sd card instead of the 128gb i was trying to use and it worked. Just thought i would mention incase others experience the same issue.

oldman4U, 2021-02-25:
I believe this is mentioned in the ReadMe🤔

XeroDay99, 2021-03-18:
> did TFT35 contain this?
> 
> [TFT35.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5269443/TFT35.zip)

Files missing

radek8, 2021-03-19:
> > did TFT35 contain this?
> > [TFT35.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5269443/TFT35.zip)
> 
> 5 bmp Files missing
> *BRIGHJTNESS.bmp
> *MESH_EDITOR.BMP
> *MARLIN_MODE.bmp
> *CASE_LIGHT.bmp
> *Z_ZALIGN.bmp

Are you kidding me?
Of course the files are missing.
I added Archyv 1/2 year ago, FW went through a lot of development.
Use the current icon folder.

radek8, 2021-03-19:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update

radek8, 2021-03-19:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/THEME_Unified%20Menu%20Material%20theme