bigtreetech/BIGTREETECH-TouchScreenFirmware #1120: getting Marlin Mode working with TFT35 E3 V3 and SKR E3 V2

Grazuncle, 2020-09-22:
I bought these two boards together.. but cannot get the TFT35 to work in Marlin Mode (I have a plain Ender 5 with no options added) I downloaded the last update (had to reverse the stepper wiring on all axis except extruder) but it works ok

I press the touch screen or hold the know down to get the option and it does give me the two options. The know will not toggle between the two but pressing the Marlin icon does operate the option but only to show a black screen with the word LCDxxxx (some numbers)
Nothing happens but can get back to touch mode again.

I have a 10pin cable and the 5 pin serial attached and can use the printer in touch mode successfully.. Just not in Marlin Mode..

Some are saying you need two 10 pin cables attached and other say only one 10 pin and serial are needed.

I seem to have a recent Firmware on the TFT35 E3 V3
sorry if this is the wrong place to post this.

Help please?

oldman4U, 2020-09-23:
One cable is fine. 

Have you checked the read me first on top of this ticket system? Maybe you find the answer there. 

Please let us know

radek8, 2020-09-23:
Connect the 10-pin connector on the printer to the EXP3 connector

Grazuncle, 2020-09-23:
I'm sorry I didn't think to search but appreciate the acknowledgement of the number of 10 pin cables.

I do have the 10 pin to the EXP3..... it works fine and my preferred mode (for most things) but won't go into Marlin mode.. I must have something else not right in my system pairing.

If there is another thread already covering this I'll go there and read up on it.. Thanks

oldman4U, 2020-09-23:
No reason to be sorry, it is not very prominent placed at the moment.....

Can you attach the configuration.h from your mainboard firmware, please.

Thy

radek8, 2020-09-23:
If you have it connected properly, make sure you are using the correct display driver in Marlin.

You should only have one active !!
CR10_STOCKDISPLAY

radek8, 2020-09-23:
> 
> Můžete připojit Configuration.h z firmwaru základní desky, prosím.
> 

that would help

Grazuncle, 2020-09-23:
I did not compile it myself.. just downloaded the ender5 bin from this location

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V2.0/firmware-ender5.bin

If i could back track I might find it.. I have not tried a compile yet. 

Thanks for looking at this

oldman4U, 2020-09-23:
Under the link you sent us, there is a new version which is 4h old. Maybe they found a problem and fixed it...

radek8, 2020-09-23:
If you disconnect the 5-pin UART cable.
the display will only be on a 10-pin cable, the display will be powered, will the Marlin mode be displayed?

Grazuncle, 2020-09-23:
I'll try that!

Grazuncle, 2020-09-23:
This is just weird..

Before switching on I pulled the 10 pin out of the TFT35 and it still displayed in touch mode and the LED under the knob, which has never ever lit, despite selecting a colour option.. is glowing!

Interestingly, whilst the knob changes values in the touch mode menus it won't select between the touch or marlin mode (which i have seen it do on Teaching Tech YT videos. I can touch select Marlin mode but it just displays 'LCDxxxx simulator'

firmware for TFT35 says June 2020

oldman4U, 2020-09-23:
10pin cable is only used for Marlin mode.

The firmware is VERY old and you should update to the latest master.

Grazuncle, 2020-09-23:
Wow a months is a long time in programming!

I'll try and find it..

grateful

oldman4U, 2020-09-23:
Even more if the one month means 3-4 months. 

😁

Make sure you update firmware, TFT folder AND config.ini. 3 elements.

Grazuncle, 2020-09-23:
You must be psychic!

You just pre-empted my next question.. i have a message saying I need to update  fonts and config before proceeding..
I don't know where those are.. feel stupid now :(  If I switch off I'll have to start again?

radek8, 2020-09-23:
you didn't answer my question.

There must be 3 items on the mole
bin file
this file
TFT35 folder

Grazuncle, 2020-09-23:
apologies..

The Marlin mode didn't display after removing the 10 pin ribbon but the touch mode did... Oldman4U said I have greatly outdated firmware! I tried ot update with the bin file and  missed the suggestion to put the ini file in there.. I thought i found it but cannot select it as a down load.. i see the text of the .ini... 

could i copy it into notepad and save as config.ini?  

I'm out of my depth here :(

oldman4U, 2020-09-23:
No need to edit at the moment.

Could you please read the Read Me from here and let me know if this helps.

Thank you

https://github.com/oldman4U/BIGTREETECH-TouchScreenFirmware/tree/Oldman4U

Grazuncle, 2020-09-23:
Thanks for that I'll certainly look at that 

I'm just finding it hard to get everything together for this update

Thanks for you're patience

Grazuncle, 2020-09-23:
I think i have gone down a rabbit hole and cant get out.. broke my printer into the bargain.. 

It was going to be so easy..

I can't compile anything so thought the firmware .bin was ok as it was for the SKR.. :(

radek8, 2020-09-23:
I asked the other way around.
Disconnect the 5pin connector, connect the 10pin connector!

all files can be found here
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update

radek8, 2020-09-23:
edit config.ini for your own needs

oldman4U, 2020-09-23:
Updating the printer all you need is one firmware file.

Updating the TFT you need 
the firmware file of your specific TFT
the TFT folder which matches the size of your TFT and the
config.ini.

They all must be from the same firmware version, otherwise see your comment from 2h ago...

Grazuncle, 2020-09-23:
Thanks..

I started again with the Config.ini and although it says some bmp are missing,  it is still asking fonts: updated required 

when i look at your link I see folder called font.. There is not much inside to folder.. or not what i expected:

Do i just paste the font folder into the root of the memory card or the contents folder?

Sorry to be so slow/thick.. 

I'm lost!

radek8, 2020-09-23:
OMG
This unzip to SD card
[Copy to SD Card root directory to update.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5269372/Copy.to.SD.Card.root.directory.to.update.zip)

radek8, 2020-09-23:
or whatever you like
[Uploading Copy to SD Card root directory to update - Unified Menu Material theme.zip…]()

oldman4U, 2020-09-23:
This is the structure on the SD Card.

The orange line is a placeholder for all the bmp's which I deleted so everything fits on a single screenshot.

Hope this helps

[image]

Grazuncle, 2020-09-23:
well I managed to do something.. 👍 

Thanks to you and Radek8..

I edited the config and put the file in the root..

I still don't have Marlin mode but I checked the firmware date and now have April 2020.. that is 20 years old in programming time.. it is working but I seem to have made things worse.

Could someone point me to the youngest firmware please for TFT35 E3 V3 (I thought i was at that place but obviously i can't see the woods for the trees

Thanks again for your patience..

radek8, 2020-09-23:
use the FTP35.zip I sent

radek8, 2020-09-23:
[image]

radek8, 2020-09-23:
[image]

Grazuncle, 2020-09-23:
Thanks you so much..I have had a break for something to eat.. Then I'll go
back and put it right with your zipped files.

Sorry for being a pain.. Appreciate your and oldman4U's efforts.

oldman4U, 2020-09-23:
👍🏻

Once this is working you have to explain your nickname.

Grazuncle, 2020-09-23:
> 👍🏻
> 
> Once this is working you have to explain your nickname.

will do..

I tried Radek's zip files directly to a blank SD card.. it missed a few bmp files as before and now it has stopped again ..

There is some text under 'found outdated  data'. then a line which I can't fully read ..
Then it says insert the card with the required files.... 

I'll bet you never wished you got up this morning...

oldman4U, 2020-09-23:
I had worse days before!

Disconnect everything from the TFT except for the 5 pin RS232 cable and try
it again.

If it fails take the SD card and check if the filenames changed.

👍🏻

oldman4U, 2020-09-23:
Usually I use self compiled firmware, but I wanted to give it a try and used the pre compiled firmware like radek8 described above.

I was able to update my 2 TFT35 E3 v3 without any problem.

When you do so, make sure to use the E2 version in case you have a E3 TFT. In case it is a TFT35 v3, make sure to use the TFT35 v3 version.

Maybe it is also worth to try the 128MB card which came with the mainboard with a SD to micro SD adapter.

radek8, 2020-09-23:
try this bin, I disabled the check.
[BIGTREE_TFT35_V3.0_E3.26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5271311/BIGTREE_TFT35_V3.0_E3.26.x.zip)

Grazuncle, 2020-09-23:
The file names did change..

It definitely says TFT35 E3 V3 on the back of the TFT.. have i used the wrong firmware then

I tried again without the 10 pin header and it did update without any messages  I don't recognise the display icons and it doesn't seem to be the home screen.. 

Saying that though..  the touch doesn't work when i press the screen.. 

I could buy another screen.. (or go back to the April firmware?) but will have to wait a bit

oldman4U, 2020-09-23:
You could give v26.one (not v26.x) a try. And try another SD card.

radek8, 2020-09-23:
You either have a faulty SD card or a faulty Flash in the display

radek8, 2020-09-23:
He read what you wrote again.
You never mentioned what kind of record you had with Marlin.
Disconnect the 10-pin cables and try the current FW update in the display with the firmware file TFT35.zip that I sent

radek8, 2020-09-23:
a photo can also help

Grazuncle, 2020-09-23:
> You either have a faulty SD card or a faulty Flash in the display

that's a valid point.. I'll try and locate the original or at least something else

Grazuncle, 2020-09-23:
> He read what you wrote again.
> You never mentioned what kind of control board you have with Marlin.
> Disconnect the 10-pin cables and try the current FW update in the display with the firmware file TFT35.zip that I sent

I'm not exactly sure what you mean by control board? the SKR or the TFT35 E3 V3?

I did try the disconnect the 10 pin ribbon idea and it did go through with no erors but although i did get a display of the old style BMP icons the touch operation didn't do anything..

I'll try and find another SD card tomorrow.

Thanks for sticking with me on this..
psI'd prefer the unified interface.. i see you can select it in the config.ini but where are the BMPs?

I'll post images in follow ups

radek8, 2020-09-23:
SKR....?

oldman4U, 2020-09-24:
> You never mentioned what kind of control board you have with Marlin.
> 

It is in the headline. Skr 
E3 mini v2

oldman4U, 2020-09-24:
So you can update to a non working display with the firmware which has the check disabled. This means the update did not work. 

With the firmware which checks the versioning the update ends with the warning. 

We are still at the point where the update does not work. SD card is most likely the problem. 

I will attach pictures to show how a update has to look like.

Grazuncle, 2020-09-24:
I found a 1Gb Sd card formatted it (not quick option)

Then i found the 8 day old firmware list on Github.. selected the unified option and downloaded the zip. (I actually learned and remembered something!!!) I admit I find this hard to navigate.. I remember when modems started to become consumer items with no help whatsoever.. it was like a black art!

I edited the config just to turn on the knob LED and selected the green option just so i would know if anything has actually worked.  It went through the routine with no BMP missing this time and next thing i know there is the unified screen and the know in glowing with green LED.... no touch working though
[image]

I did the upload with just serial cable attached.. ironically when pressing the know for three seconds now, to get to marlin or touch mode screen you can use the knob to switch between the the two (never had that work!) it still doesn't let you go into it though.. you can press the knob for another three seconds ot get back to BTT mode but all you can do or see is the attached photo

I'm still way off with something... try another card?

Grazuncle, 2020-09-24:
more weirdness..

I really needed to print something today.. and thought.. what if I..

Put the old Ender 5 screen back on?.. well I have all of the functions.. 

[image]

bit clunky in relation to the TFT touch.. However it may help us in some way.. the SKR is not ballsed up too.. LOL

Any thoughts?

radek8, 2020-09-24:
Does the touch not work?
The screen touches the metal frame too tightly and shorts the touch layer. Loosen the screws and put a washer there. :-)

Grazuncle, 2020-09-24:
> Does the touch not work?
> The screen touches the metal frame too tightly and shorts the touch layer. Loosen the screws and put a washer there. :-)

Thanks.. I took the TFT35 out completely. rebooted it and it still does not operate on touch.. the old display is next to the TFT35  (it is not connected as well btw) 
[image]

ps. all three files/folders have their extensions changed to CUR.. Not really sure if that means anything now

radek8, 2020-09-24:
after a successful update, the extension is changed to CUR so that it is not installed at the next reset

radek8, 2020-09-24:
Upload an empty "resert.txt" file to the SD card and restart the display. Calibration is performed.

radek8, 2020-09-24:
> > Nikdy jste nezmínil, jaký druh ovládacího panelu máte s Marlinem.
> 
> Je to v nadpisu. Skr
> E3 mini v2

There is no "mini" in the description
I don't know if this is a new type ...

Grazuncle, 2020-09-24:
Bloody Hell !!!!!

booteed with the reset.txt file and now the damn thing just woke up and working......

there is are a lot more options to set in this new update.. My God!

I tried the 10 pin ribbon (marlin mode) in the screen again and whilst the knob does switch between options all I get is LCDxxxx simulator.

I think I'll count my chickens and be satisfied with a working TFT35.. 

I appreciate you all having the patience to stay with this.. I might revisit Marlin mode sometime.. just a little happier with it working.. Thanks guys

radek8, 2020-09-24:
Did you connect the 10pin connector to the EXP3 connector?

radek8, 2020-09-24:
What does LCDxxxx mean? Describe it all.

radek8, 2020-09-24:
If you connect only the 10pin connector, disconnect the 5pin connector. Does the display light up?

Grazuncle, 2020-09-24:
> What does LCDxxxx mean? Describe it all.

I could not remember the numbers sorry so put xxxxs
[image]

yes I did try the 10 pin ribbon. previously the knob would not let me turn it to toggle between BTT touch or Marlin mode on display but it does now.. But the screen capture above is all you get.. pressing the knob or turning it does nothing so you have to back out of it into touch mode.

oldman4U, 2020-09-24:
Grazuncle

We are not giving up, sorry;-) Now we want the Marlin work as well.

[51 more replies]

closed 2020-09-25 by Grazuncle