bigtreetech/BIGTREETECH-TouchScreenFirmware #1121: [FR] Runout Sensor On Board

ETE-Design, 2020-09-23:
When you have a Runout Sensor connected direct to the mainboard nothing seems to happend on the TFT when the runout is triggered... Would be nice with support for it so the TFT Will pop-up with a messages when runout is triggered...

blueeagle69, 2020-09-27:
I did think that with;
#define HOST_ACTION_COMMANDS, #define HOST_PROMPT_SUPPORT along with M114 Detail and Emergency Parser would have fixed it.
Clearly not it appears.
The TFT just carries on like nothing has happened although the printer has paused waiting for filament change.

empakoso, 2020-10-01:
I would be interested in this enhancement as well.  My Octoprint recognizes 'Wait for user input'.

blueeagle69, 2020-10-07:
I have an update to this.
With Marlin 2.061 and the latest TFT source the Smart sensor is working when connected to the onboard connector.
I tested last night and even in TFT mode I got alerted to a runout state with buttons to respond to it.
Basically everything Marlin would prompt during a runout was on the TFT!

Make sure you have these settings in  Marlin config_adv.

#define M114_DETAIL         // Use 'M114` for details to check planner calculations
#define REPORT_FAN_CHANGE  (not related but handy never the less)
#define HOST_ACTION_COMMANDS
 #define HOST_PROMPT_SUPPORT
#define EMERGENCY_PARSER

empakoso, 2020-10-07:
I will try this.  Thanks for the update.  When you define “EMERGENCY_PARSER”, did you get a fault on in Platformio stating that the BTT board doesn’t support it?  I should mention that I am using the BTT SKR E3 v2

blueeagle69, 2020-10-07:
Never noticed it bud.

blueeagle69, 2020-10-07:
Although I am running the SKR1.4 in my B1

labotecno, 2020-10-07:
i dont understand when i have get conected runout on the tft with skr 1.4 turbo is working just like inverted , if filament is not an the sensor is working and if filament is inside it tell me runout filament ... i try to change this and nothing to do ....
  ```
Filament runout inverting - invert the logic of the sensor.
 Options: [true: 1, false: 0]
```
someone have the same issue ? 
fil_runout_inverting:0on` the config.ini and

blueeagle69, 2020-10-07:
Weird. Never really tried it connected to the TFT.
Daft question but I have to ask.
What sensor is it?
Is it the BTT smart one?
If the latter, have you selected Smart in the TFT runout options.
But saying that I have heard that the smart sensor doesn't appear to be working correctly when connected to the TFT.
In fairness I disconnected the runout cable from the TFT and connected to the SKR 1.4.
Works so much better now in TFT mode and Marlin mode

blueeagle69, 2020-10-20:
With the latest Marlin and TFT FW runout support appears to be working.
I have my BTT Smart Filament Sensor connected to my SKR 1.4 with the following options enabled in Marlin.
#define M114_DETAIL // Use 'M114` for details to check planner calculations
#define REPORT_FAN_CHANGE (not related but handy never the less)
#define HOST_ACTION_COMMANDS
#define HOST_PROMPT_SUPPORT
#define EMERGENCY_PARSER

Make sure the filament runout options are disabled on the TFT.
I have tested, and the TFT does show a filament runout. The printer pauses, chucks out the remaining filament, and the TFT has a clickable button to load filament and resume etc.
Tested twice, appears to work fine.

oldman4U, 2020-10-20:
Hi ETE-Design.

Could you please check if this is really working and let us know. Your request is currently in the pinned FR list.

Thank you blueeagle69 for letting us know

blueeagle69, 2020-10-20:
@oldman4U No problem 😉
The last time I tested with the above config/firmware, the TFT appeared to echo what you would expect to see in Marlin mode, with the clickable option to resume etc. It also showed a popup showing it was waiting for the nozzle to reach temp before loading the filament.
Again, everything Marlin shows.
I hope I didn't dream it 😴

oldman4U, 2020-10-20:
Hope so too;-) Otherwise, we have it on our FR list...

blueeagle69, 2020-10-20:
😁

blueeagle69, 2020-10-20:
Just did a very quick test with no filament loaded.
[image]
It brought up the popup above, carried on for a couple of seconds (runout distance), paused and threw out the imaginary filament!
So yes, it works on mine!
But of course @ETE-Design has to confirm.

oldman4U, 2020-10-21:
I guess the filament max length  in Marlin is too low. Change it from 200
to 800.

blueeagle69, 2020-10-21:
Mine was increased to 480, but will increase to 800.
But hey at least the runout works when the sensor is connected to the mainboard instead of the TFT.
I have successfully performed a filament change while in TFT and appeared to work flawlessly.

oldman4U, 2020-10-21:
I don't have such a hardware, but I am happy to hear that it works. Don't know who fixed it;-)

@ETE-Design. Could you please confirm and close this ticket.

Thank you

ETE-Design, 2020-10-21:
@oldman4U Dosen't seem to work here... I have an MKS Gen L + MKS TFT28 with BigTreeTech firmware on... With M114 the filament show Triggered / open depending on there is filament in or not, but nothing on the screen... I'm using Marlin 2.0.7.2

blueeagle69, 2020-10-21:
Strange.
Daft question, have you got the following enabled in Marlin?

#define M114_DETAIL // Use 'M114` for details to check planner calculations
#define REPORT_FAN_CHANGE (not related but handy never the less)
#define HOST_ACTION_COMMANDS
#define HOST_PROMPT_SUPPORT
#define EMERGENCY_PARSER

Because that's all I did to get mine working with an onboard sensor, was to update Marlin and the TFT firmware.
I had the above settings enabled beforehand, and it started working.
The only difference I can see is I'm using an SKR 1.4 and TFT35 e3.
What sensor is it? Smart or normal?
Is it working in Marlin mode?

ETE-Design, 2020-10-22:
@blueeagle69 Yes it is enabled in Marlin... Are you using Marlin 2.0.7.2 or Marlin Bugfix 2.0.x?
It is a normal filament runout sensor, and not possible to switch between moden on the MKS TFT 28, so only touch mode is possible...

blueeagle69, 2020-10-22:
I'm using 2.0.7.2.
I'll be surprised if it's because I'm using the Bigtreetech Smart sensor.
Or is it because the screen doesn't have the option to switch between modes.
It should still receive commands using 
#define HOST_ACTION_COMMANDS
#define HOST_PROMPT_SUPPORT
#define EMERGENCY_PARSER

That's annoying 😣

blueeagle69, 2020-10-22:
This is with my setup. Just tested.
[image]
[image]
[image]
[image]

Sorry for the blur in the last photo. In between them is a Nozzle Parked notification also.

mustang5269, 2020-10-25:
I’m running a tft35 e3, skr e3 mini v2 running the latest marlin bug fix using the eo-stop pins and it’s not working for me. I’ve tried to trigger it multiple times and the printer kept printing. I have not tried this on the marlin mode.

blueeagle69, 2020-10-25:
Yep, same here.
I do my printing from the onboard SD.
I think that's why it worked for me and not others.
Thanks for mentioning the onboard SD.

blueeagle69, 2020-10-25:
With the Biqu B1 I have always printed default from the Onboard SD, so I can slice straight to the SD Card without taking it out, so I never thought of what SD Card I was using.
But clearly it has a bearing on the runout sensor!

mustang5269, 2020-10-25:
Yup, looks like the only way for the sensor to work is using the onboard sd. I was so used to using the other and didn't think of it.

oldman4U, 2020-10-30:
@ETE-Design Are you printing from OnBoard or TFT SD card?

ETE-Design, 2020-10-31:
@oldman4U No, printing direct from the TFT

mustang5269, 2020-11-22:
So lately I've been trying to print a lithophane and I keep finding my printer at a heater timeouts message. After rolling back firmware's and several other things I believe it's related to the sensor triggering a halt. What would cause this when there is no break in filament?

empakoso, 2020-11-22:
Not sure if we are hijacking this thread.  Move this if necessary.  But I will answer using my experience.
Depending on bed or nozzle.  Nozzle would be; loose sensor/thermocouple.
Bed and/or Nozzle; Loose connection to the heater element.
I would check these things first.  It just happened to me last week where my bed was faulting.  It was the terminals in the board became loose and the wire was just sitting there.  This was caused by the very bad practice of pretinning with solder, all the strands of wire to keep them together.  Please ensure strands are all individually compressed by a screw terminal.  And don’t use ferrules, they are the worst for loosening up.
Next its a faulty element or sensor.  Check using a multimeter set to continuity.

mustang5269, 2020-11-22:
I certainly don't want to hijack anyone's thread but the reason I'm posting this is because I feel it's related. The only thing that fixes my issue was to remove anything dealing with the smart sensor in marlin to disable it and it works fine. I think a setting must be off triggering the smart sensor to cut the heaters.

empakoso, 2020-11-22:
I definitely don’t want to discredit this relationship between your two issues.  My apologies for for that.
I have both the BTT smart runout sensor/BLTouch/neopixel/speaker(gcode songs play nicely) working on my machine using the BTT SKR mini E3 v2 and the TFT35 E3 v3 setup running Marlinfw 2.0.7.2 release.
As mentioned the screw terminals (and they are not at issue here, just to be clear) became loose connecting my bed heater element.
I tried messing with the firmware’s thermal protection settings by doubling the values without success.
It was my physical connections.  But I’ve also run into this before with a loose thermocouple sensor in the nozzle.  And another time, discovered a broken solder joint on my heated bed.
Just some advice.  Hope it helps.

mustang5269, 2020-11-22:
I'm going to replace both the thermocoupler and heat sensor in the block and check connections to rule that out also. Thanks for the tips!

empakoso, 2020-11-22:
Just to add, I use OctoPrint for all my prints.  I have not tried using either SD slits for printing.

mustang5269, 2020-11-22:
I've been using octoprint also. I ran it in safe mode and rolled back firmware to 2.0.7.2. Same fail. Maybe it's just a coincidence but it sure didn't want my to print this lithophane until i disabled the sensor. Thai may give me the excuse now to move my spool back up top and not use the sensor until it's more supported.

empakoso, 2020-11-22:
I wish you luck.  I would recommend you take a base line (from your good/new/working elements  and sensors) resistance reading and compare it to what’s on your machine (disconnect at least one wire so you don’t add any other board resistances to your result.  If they are the same, then save yourself the hassle of replacement.

mustang5269, 2020-11-22:
So I've finally was able to get a fault while I was in the room. Ran over to octoprint and it's a heating failed. Heater is bed. I noticed the bed temp taking off the preset temp of 60. I redid the power wires but I'm wondering if something wrong with the sensor.

empakoso, 2020-11-23:
if the bed wires look good, check the sensor is not loose.  You can check resistance of the sensor based on a good one.  But the short answer is yes, the sensor could be damaged.

mustang5269, 2020-11-23:
I made the mistake of leaving the wires tinned so I redid the bed wires. The red/black wires look good on the bed but I have no way of testing the bed thermistor. I've never touched it since owning it so I don't know how it would have gotten damaged. I have been doing pid tuning but the numbers are always changing.

empakoso, 2020-11-23:
Actually, what you are doing by PID tuning is a great indicator.  Those changing values indicate you will need take a closer look at your Bed sensor.  Flip the bed and take a look at if you can replace it.  Otherwise, it maybe a bed replacement.

mustang5269, 2020-11-23:
Thanks for all your help! I think we're on to something. It's probably worth just replacing the bed at this point since the sensor is taped over in the middle so it has to come off. I feel like I'm hijacking at this point so is there another way I can continue chat with you?

empakoso, 2020-11-23:
Sure thing.  Let’s finish this here when you finally get a good result.  The mods can move this as they please.

oldman4U, 2021-01-12:
Hi.

I learned from kisslorand that you have to make Marlin know that the print has started by adding an "M75" to your start gcode. Once done, the reported issue should be fixed.

Could you please check and let me know, because I have no such sensor

Thank you

ETE-Design, 2021-01-12:
@oldman4U So with filament runout connected to mainboard and printing from SD-Card at TFT Screen a M75 should make it is this correct understod? Cause the I close this issue :-) Mabye it should be added to the "Manual"?

oldman4U, 2021-01-12:
Yes. Please verify this but it already worked for another user. Once you
confirm this I would add it to config.ini so other users do see it also.

ETE-Design, 2021-01-13:
@oldman4U I'n not near my printer ATM :-( @mustang5269 Do you have time to test it?

mustang5269, 2021-01-13:
I wound up taking it off due to not getting it to work properly, sorry.

oldman4U, 2021-01-19:
Any news ETE-Design?

ETE-Design, 2021-01-19:
@oldman4U Don't have a printer with BTT TFT atm :-(

oldman4U, 2021-01-22:
So I am the only one who cares, even I don't use this hardware because it is useless in my opinion!!??

ETE-Design, 2021-01-22:
@oldman4U So you don't use the BTT TFT? :-D

oldman4U, 2021-01-22:
I do. But no runout sensor.

ETE-Design, 2021-01-22:
@oldman4U Okey, I can just close this if you want?

oldman4U, 2021-01-22:
If you are not able to maintain it it would be better. Yes.

closed 2021-01-22 by ETE-Design