bigtreetech/BIGTREETECH-TouchScreenFirmware #1122: default extruder speed?

BlackDragon2020, 2020-09-24:
So I am in the middle of tuning my tmc2209, with my skr1.4. I go to do e step calibration and it start skipping steps.

I spent a good hour playing with current and nozzle temps etc. after some testing I came to the conclusion it was the tft35 and what it was sending.

after some digging in source codes if found this:
The stock ender 3 manual feed rate for the extruder is 120mm /min

tft extruder speed
slow: 60 mm /min
normal: 600 mm / min
fast: 1200 mm /min

I dont know the limits of other printers, but @ 600 mm/ min, extra temp on nozzle, extra tension on extruder screw, max current for extruder stepper, I got the skipping or kick backs down to about 2 in a 100mm feed @ 600mm /min.

you might consider adjusting the default speeds, most definitely for the tft35 e3 as it is targeted @ the ender 3.

Just my findings do with them what you want.

thanks for your time.

radek8, 2020-09-27:
You can change this value by the user in the Config.ini file

This value is mainly intended for calibrating the extruder.
I don't know about you, but I'm calibrating with the bowden cable disconnected. I don't want to stand at the printer for a few minutes and wait for the 100mm filament to pop out :-)

[image]

BlackDragon2020, 2020-09-27:
> You can change this value by the user in the Config.ini file
>
> This value is mainly intended for calibrating the extruder.
> I don't know about you, but I'm calibrating with the bowden cable [...]

I do calibration at temp connected because as far as I know pushing filament through hot nozzle vs nothing will change esteps a bit.

radek8, 2020-09-27:
OK, set your own value that suits you

traffic-light, 2020-09-28:
You now you can use the tuning menu for estep calibration?
The speed is at the right speed for calibrating e-steps.

oldman4U, 2020-09-28:
;-)

Please help the community and close this ticket once you do not need it anymore.

Thank you

closed 2020-09-29 by BlackDragon2020