bigtreetech/BIGTREETECH-TouchScreenFirmware #1124: [Q] Making Touch Mode work with SKR1.4

siteswapjuggler, 2020-09-25:
The Touch Screen keep prompting "No Printer Attached" despite corresponding baudrate configuration. BTT TFT35-E3 v3 and BTT SKR 1.4 Turbo. I searched for similar issues on this github without founding any valid solution. Already updated to the latest BIGTREE_TFT35_V3.0_E3.26.x firmware.

What should I check to make it work.

traffic-light, 2020-09-25:
Did you connect the black ribbon cable correctly?
What did you do to solve this problem?
Is the baudrate the same on the screen AND the tft?
Is the serialport set correctly?
Should be:
#define SERIAL_PORT -1
#define SERIAL_PORT_2 0

siteswapjuggler, 2020-09-25:
Hello and thanks for your reply, yes ribbon cable is connected correctly, baudrate was the same on the tft and skr. But you were right about serial port not being connected in Marlin. I was gotting it by checking the pinout while your were replying.

So it works perfectly now, thanks a lot.

closed 2020-09-25 by siteswapjuggler