bigtreetech/BIGTREETECH-TouchScreenFirmware #1125: [FR] (Correct report from marlin firmware\skr board for M117 messages via terminal etc)

radfordwill, 2020-09-25:
**Is your feature request related to a problem? Please describe.**
Marlin firmware reports m117 messages to the lcd in Ender 3. I am assuming it is that way with most printer\lcd combos. The firmware for the tft35 v2.0, at least from the BTT SKR DIP v1.1 in my case, does not process M117 messages from anywhere but internally sent messages.

**Describe the solution you'd like**
I read python and some c++ but I can't find anywhere to start adding support for this problem. Can this be added? Octoprint plugins like [Display Layer Progress](https://plugins.octoprint.org/plugins/DisplayLayerProgress/) would be an asset to this lcd's functionality. Any Ideas?

**Additional context**
[M117 - Set LCD Message](https://marlinfw.org/docs/gcode/M117.html)

oldman4U, 2020-09-25:
Hi.

Searching for M117 gives me the following results:
[image]

So there are already several tickets related to this topic, maybe you want to join one and close this ticket after that.

Thank you

radek8, 2020-09-25:
I don't know what news you are missing.
Unwanted M117 messages can be suppressed in the parseACK.c file
Search there if you want to display a message.

radfordwill, 2020-09-25:
> Hi.
> 
> Searching for M117 gives me the following results:
> [image]
> 
> So there are already several tickets related to this topic, maybe you want to join one and close this ticket after that.
> 
> Thank you

Reading through those now. I've read several and they don't cover it. It is not a bug.  I am asking for a feature added.

radfordwill, 2020-09-26:
> I don't know what news you are missing.
> Unwanted M117 messages can be suppressed in the parseACK.c file
> Search there if you want to display a message.

What port on the tft are you using? 1 or 3?

radek8, 2020-09-26:
[image]

radfordwill, 2020-09-26:
> I do not understand the question
> 
> [image]

I followed this to the letter with the V2.0 of the TFT35. It does not appear to have the same options the 3.0 board has without the EXP3 port receiving the commands maybe.  In ant case, probably an old board that is not going to work even if they add it to the newer boards. Closed request for TFT35 V2.0. Out of date board Bigtree should quit selling. 

@radek8 - Thanks for the pointers and directions in the [other thread](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1050)

closed 2020-09-26 by radfordwill

radek8, 2020-09-26:
That's right TFT35 V2.0 is a failed model