bigtreetech/BIGTREETECH-TouchScreenFirmware #1127: [BUG] (short description)

feiticeir0, 2020-09-25:
### Description

When printing from the onboard SDCARD, the status screen is not updated (sometimes, only after the nozzle reaches the target temperature) until the bed reaches the desired temperature and after that, it only gets updated after the nozzle reaches the target temperature
### Steps to reproduce

1. load a gcode from the onboard TFT
2. wait

**Expected behavior**

Expecting the screen to be updated everytime the temperature is increasing

**Actual behavior**

only gets updated after the target temperature is reached..

### Hardware Variant

### TFT Firmware Version & Main Board Firmware details
Board: BIGTREETECH TFT35_v2.0
Firmware: TFT35_v2.0.26 Apr 11 2020

Marlin 2.0.6
[configurations.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5285549/configurations.zip)

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

radek8, 2020-09-25:
You have an old FW on the display. Update it ...

feiticeir0, 2020-09-25:
Where can I find it ? I've searched, but for this TFT, I can't find

radek8, 2020-09-25:
I don't know what your display is, you have to know

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update%20-%20Unified%20Menu%20Material%20theme

radek8, 2020-09-25:
hey, version TFT35V2.0 is lost somewhere.
here it is if you need it.
[BIQU_TFT35_APP1_V2.0.26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5285771/BIQU_TFT35_APP1_V2.0.26.x.zip)

radek8, 2020-09-25:
Caution TFT35V2.0 may not include a botloader.
if you do not have it, you must first upload it according to the instructions
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/TFT35_V2%20Bootloader%20fix

feiticeir0, 2020-09-25:
> I don't know what your display is, you have to know
> 
> https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update%20-%20Unified%20Menu%20Material%20theme

It was the first thing I did when installed the TFT - I have TFT35 v2.0 - Already tried (with the respective folder):

-  BIGTREE_TFT35_V3.0.26.x.bin
- BIGTREE_TFT35_V3.0_E3.26.x.bin 
none of them was updated. 

I have version 2 of the TFT screen .

I finally tried BIQU_TFT35_APP1_V2.0.26 and it was updated. But it kept complaining about a missing file to complete the update, that I ended up copying all the files to the SDCARD and let it update. 

I know have TFT35_v2_0.26.x from September 2020

But, the issue remains... No update of the status of bed and nozzle...

bigtreetech, 2020-09-26:
Hello, You can manually send "M155 S1" (automatically report temperature every 1s) from the Terminal menu after each restart to enable Marlin's automatic temperature report function, or add "M155 S1" to the start gcode of the print file. In this way, Marlin will report the temperature once every second, which can solve the problem that the temperature does not update due to Marlin not responding to M105 during heating up

feiticeir0, 2020-09-26:
> Caution TFT35V2.0 may not include a botloader.
> if you do not have it, you must first upload it according to the instructions
> https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/TFT35_V2%20Bootloader%20fix

TFT it did update the firmware - I have a new version now - BIQU_TFT35_APP1_V2.0.26

feiticeir0, 2020-09-26:
> Hello, You can manually send "M155 S1" (automatically report temperature every 1s) from the Terminal menu after each restart to enable Marlin's automatic temperature report function, or add "M155 [...]

Why is this done automatically when reading from the TFT SDCARD and not when is reading from the  SDCARD onboard of the SKR 1.4 ??

radek8, 2020-09-26:
Check the Marlin configuration

[image]

feiticeir0, 2020-09-26:
I have it set in the configuration !

oldman4U, 2020-09-27:
Have you added it to the start gcode like recommended by BTT?

feiticeir0, 2020-09-27:
> Have you added it to the start gcode like recommended by BTT?

Yes ! It's done - but my question remains - when printing from TFT sd card it works, when printing from SKR onboard SD CARD it does not and it's necessary to add the GCODE

radek8, 2020-09-27:
There are things between God and the printer that don't need to be understood, especially that it works. :-)
After all, when I print from touch mode, I print from the SD that is on the display, when I print in Marlin mode, I print from the card that is on the Marlin board. Thus, in both cases, the temperature updates well.
It is not good to print from a card on a Marlin board in touch mode.
because this will make the daledo use the serial line more, and when using the speed of 115000 and printing a part that contains a lot of short paths, the line may be overloaded and the print may jam ...

Each card has its own use.

jalensailin, 2020-09-30:
Can I add this gcode at the end of config.ini? Or does it have to be done in the slicer? This is what I was thinking: 

`#--------------------------------------------------------------------
# Start, End & Cancel G-codes
#--------------------------------------------------------------------

#### Default Start Gcode Status
# Options: [enable: 1, disable: 0]
start_gcode_enabled:1

#### Default End Gcode Status
# Options: [enable: 1, disable: 0]
end_gcode_enabled:0

#### Default Cancel Gcode Status
# Options: [enable: 1, disable: 0]
cancel_gcode_enabled:0

#### Start G-code - run this G-code before starting print
# maximum length 50 characters
start_gcode:M155 S1\n`

jalensailin, 2020-10-02:
The above does not seem to work for me and I don't know why

feiticeir0, 2020-10-02:
> The above does not seem to work for me and I don't know why

It also does not work for me anymore... I've put the code in cura startup gcode, but won't work

oldman4U, 2020-10-22:
Just to make sure, I also add the movie here. This is from printing of the mainboards SD Card, using a SKR E3 DIP and BTT TFT35 E3 v3, running Marlin 2.0.7.2 and the TFT firmware from Oct. 19th, no start gCode related to M155 in Cura 4.7.1 and no start gCode defined at all on the TFT.

I have define Auto_Report_Temperatures activated in Marlin.
[Temp Changing print from SD card.mov.mov.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5422223/Temp.Changing.print.from.SD.card.mov.mov.zip)

oldman4U, 2020-10-22:
Short update. Disabling Auto_Report_Temperatures does not make a difference on my machine. Temp always updates, also while printing from mainboard SD card

feiticeir0, 2020-10-22:
With the latest Marlin firmware, I know have this problem fixed. 
Printing from the internal SD Card .  No start GCODE defined in any place - CURA or TFT screen

oldman4U, 2020-10-28:
> With the latest Marlin firmware, I know have this problem fixed. 
> 
> Printing from the internal SD Card .  No start GCODE defined in any place - CURA or TFT screen.
> 
> 
> 
> I will attach the configurations and the version when I get home

When will you get home!!??

🤪

feiticeir0, 2020-10-29:
> 
> 
> > With the latest Marlin firmware, I know have this problem fixed.
> > Printing from the internal SD Card .  No start GCODE defined in any place - CURA or TFT screen.
> > I will attach the configurations and the version when I get home
> 
> When will you get home!!??
> 
> 🤪

Sorry ! With so many things, had forgotten. 👍 
Today will past it here and close the issue !

oldman4U, 2020-10-29:
👍🏻

feiticeir0, 2020-10-29:
Here are my configs.

[Marlin-configs.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5460711/Marlin-configs.zip)

oldman4U, 2020-10-29:
Ok. Is it fixed for you?

feiticeir0, 2020-10-29:
Yes ! I will close the ticket

closed 2020-10-29 by feiticeir0

blueeagle69, 2020-11-09:
Just looked at feiticeir0 configs.

He is using Marlin 2.0.6 and a Ramps board according to the configs. Also there is no SD source defined in config_adv.
Is warmup temps working when printing from onboard media working from other boards? Such as SKR1.4.
Because this has stumped me🙄

feiticeir0, 2020-11-10:
> Just looked at feiticeir0 configs.
> 
> He is using Marlin 2.0.6 and a Ramps board according to the configs. Also there is no SD source defined in config_adv.
> //#define SDCARD_CONNECTION LCD
> 
> Is warmup temps working when printing from onboard media working from other boards? Such as SKR1.4.
> Because this has stumped me🙄
> 

I'm using SKR1.4 , not ramps... 

I've enabled sd card and in config_adv (I think is there were is defined) that I'm using the onboard SD card... 

I will check the configs... Unless I've uploaded my old configs (I came from a RAMPs  board) , it should be defined SKR1.4 and not RAMPs

oldman4U, 2020-11-10:
But you have to use Marlin v2.0.7.2 or higher.

feiticeir0, 2020-11-10:
> 
> 
> But you have to use Marlin v2.0.7.2 or higher.

Yes you do !

 Tonight I have to check the configs I've uploaded..

oldman4U, 2020-11-11:
Please check out PR #1244 for a possible solution.

Thank you

feiticeir0, 2020-11-12:
I'm sorry all. The configs uploaded were from the previous version. I've deleted them. 
This ones are for version 2.0.7
[Configurations.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5527483/Configurations.zip)

PR #1244 (2020-11-18): Temperature update when printing from onboard SD
After printing from the onboard SD, the bed and nozzle temperatures are not updated until they reach the target temperature.
This adjustment solves it.
Adding code "M155 S1" to start G-code does not work.
Adding a code to the start code in Cura also does not work.
First the heating is started and only then the start code is activated.
Therefore, the values on the display are not updated until the target temperature is reached.

resolves #1127
resolves #1225
resolves #1070

Add a comment to Config.ini, with information about the required settings [... cut]