bigtreetech/BIGTREETECH-TouchScreenFirmware #1133: [BUG] Tuning menu = screen freeze

zonalimitatore, 2020-09-27:
In TFT mode with last firmw,with tft24 screen.
When I click on tuning menu,all freeze,and I'm constrict to reset the printer.

oldman4U, 2020-09-27:
Pre or self compiled firmware?

zonalimitatore, 2020-09-27:
Pre,because I've errors with autobuild

oldman4U, 2020-09-27:
The precompiled version seems to have serious issues. So I would strongly recommend to compile your own firmware and see if this solves the problem.

zonalimitatore, 2020-09-27:
Allright. Now I try to recompile it,hoping errors leave!
Let you know if the problem still

traffic-light, 2020-09-27:
Please post the error's here when it go's wrong ;-)

oldman4U, 2020-09-27:
And this is from the document of our traffic-light.

Q: Unexpected freezes when entering a menu after newly compiled firmware

A: Did you also update the config.ini on you're screen after the update? For example the freezing in preheat menu is caused by using a old config.ini.
Please try if this work. When it didn't work place an issue and tell that you did update config.ini ;-)

zonalimitatore, 2020-09-27:
Well. I've compile the firmware,and the config too.
But the problem still... When I press tuning screen freeze with a little white point. I post the photo soon

oldman4U, 2020-09-27:
Which firmware version? Check the Info of the screen.

zonalimitatore, 2020-09-27:
[image]
[Uploading VID_20200927_200044.mp4…]()

radek8, 2020-09-27:
When you uploaded the FW to the display, the icons were also recorded. Did you miss one icon?

radek8, 2020-09-27:
do you miss the Tune_extruder.bmp icon on the SD card?

zonalimitatore, 2020-09-27:
i check

zonalimitatore, 2020-09-27:
Thank you so much....Solved

closed 2020-09-27 by zonalimitatore

oldman4U, 2020-09-27:
And the winner is... RADEK8!!

So the missing icon caused this!!?? Have you ignored the warning during updating the firmware?

zonalimitatore, 2020-09-27:
Honestly, I didn't care (idiot)

oldman4U, 2020-09-27:
We all learned it step by step.

😎

zonalimitatore, 2020-09-27:
Yeah!!!Thanks at all