bigtreetech/BIGTREETECH-TouchScreenFirmware #1135: [BUG] On-screen Layer info not working printing from onboard SD

blueeagle69, 2020-09-27:
Hi.

When using the .26 firmware (latest build) and the build from the Biqu B1 repo the on-screen layer information is not working when printing from the onboard SD.

However if I print from any media connected to the TFT it works as expected.

My screen is a Biqu custom TFT35 e3 V3.

Thanks.

oldman4U, 2020-09-28:
Please add the information of the Slicer and which firmware worked well (for some time at least...)

Sebazzz, 2020-09-30:
I have the same issue. Cura 4.6.2 and Marlin 2.0.6 - although I have ported all changes from the BIQU fork from Marlin. I will reflash an older Marlin when I have the time and report back.

blueeagle69, 2020-09-30:
> Please add the information of the Slicer and which firmware worked well (for some time at least...)

I'm using Prusaslicer, with;
;LAYER: [layer_num]
;Z: [layer_z]
in Before_Layer_Change.

I'm sure now the only reason it worked before was due to the fact I was printing from media in the TFT.
If you print from the onboard (SKR1.4) SD card it doesn't work.

Sebazzz, 2020-09-30:
Is the number the same as the Z position in emulation mode? Because that does work - even when printing from the SD card in the printer.

Met vriendelijke groet,
Sebastiaan Dammann

blueeagle69, 2020-09-30:
At the moment I'm stuck on an older .26 build from the B1 repo due to the screen switching from Marlin to TFT when I start a print, so I can't confirm.  But I know on my current B1 build if I print from the tft card the layer info shows firstly the layer height (0.200mm) and increases every layer afterwards.  But from the onboard it stays at 0.00mm. If I could stop my screen switching from Marlin mode to tft and locking me there I would re update and check . I mostly use Marlin mode so I can use  my smart filament sensor.

oldman4U, 2020-09-30:
@Sebazzz 
The problem is in the slicer and Marlin. Cura gcode should work straight forward. Somewhere I saw information how to change Marlin to make this work but can not find it anymore.

blueeagle69, 2020-10-02:
Just to clarify.
I'm currently running firmware dated 220-8-18 due to the #1058 bug.

With this firmware I have sliced using Cura and Prusaslicer.
But when printing from the printer onboard SD the layer information never updates.
If this has been fixed in recent updates then apologies.

oldman4U, 2020-10-05:
Marlin mode or Touchscreen? Can you attach a picture how this looks like on screen and also Menu-Settings-Info from TFT.

blueeagle69, 2020-10-06:
In Touch mode.
Here are the screenshots.
[image]
[image]

Maker-M, 2020-10-08:
> In Touch mode.
> Works fine from TFT media but not onboard (mainboard).

Up, help me !

oldman4U, 2020-10-08:
I have to correct myself from above (sorry!!!) and can confirm this behaviour.

In addition changing the temperature while preheating is ignored (M109 vs. M104).

Wouldn't it be good to make one (Known Issue) ticket for all the issues related to printing from OnBoard SD card?

traffic-light, 2020-10-29:
This is a odd bug that's been there for a while. I'll never print from mainboard sd card.

One other thing: you're using the wrong firmware for your tft. It's telling you're using a tft35 e3. That's not the one you have.

traffic-light, 2020-10-29:
> I have to correct myself from above (sorry!!!) and can confirm this behaviour.
> 
> In addition changing the temperature while preheating is ignored (M109 vs. M104).
> 
> Wouldn't it be good to make one (Known Issue) ticket for all the issues related to printing from OnBoard SD card?

That could be something for in the readme first. 
I'm almost done with my exams. Then I got a little more space to breath

oldman4U, 2020-10-29:
Its already march!!??

👍🏻

traffic-light, 2020-10-29:
Not yet. But in march I've even more time :-)

oldman4U, 2020-10-29:
Sounds promising. 😁

blueeagle69, 2020-10-29:
> This is a odd bug that's been there for a while. I'll never print from mainboard sd card.
> 
> One other thing: you're using the wrong firmware for your tft. It's telling you're using a tft35 e3. That's not the one you have.

Weird, that's the one Biqu told me to use with the Biqu B1. I have tried both though TFT35 V3 and the e3 variant.
I print from the onboard SD so I can slice straight to the card without  having to take it out. Especially as the B1 insists on using micro SD cards.

blueeagle69, 2020-11-05:
I will tidy up my end and close my issues/bugs that you have kindly added to the pinned list 😀

closed 2020-11-05 by blueeagle69

oldman4U, 2020-11-05:
👍🏻

sadbass, 2020-11-16:
There is some news for this problem?

oldman4U, 2020-11-17:
In contrast to what I wrote above, I believe this is a bit by design. Since I have the touchscreen TFT, I mainly print from the U disk of the TFT and so this all works perfectly. But before I bought the TFT, I used the standard Ender 3 screen and using this screen there was also no layer hight information coming from Marlin. So Marlin by default seems NOT to know anything about the layer and the layer hight it is printing at the moment. To make this work under Marlin, an extension has to be activated in Cura, where this information is written into the gcode and read out from there by Marlin.

So the question is, how the TFT can get this information while printing from the SD card of the mainboard, and therefore depending on the information which comes from Marlin.

My 2 cents

oldman4U, 2020-11-19:
Please check PR #1272 to see if it solves the reported issue.

blueeagle69, 2020-11-19:
Cheers @oldman4U I will compile and flash tonight or tomorrow and report, unless someone beats me to it!

oldman4U, 2020-11-19:
👍🏻

blueeagle69, 2020-11-20:
@oldman4U 
Layer info now works perfectly with #1272 when printing from onboard SD!
Also warmup temps are also working perfectly with the previous commit.
I have noticed though that if there is no mesh saved (ABL not active) the ABL icon does not show in the Movement screen.
If I go to Marlin mode and run ABL and save the EEPROM, when I go back to TFT mode the ABL icon appears.
I'm pretty-sure this was mentioned elsewhere.
Also, if you cancel an onboard SD card print you get the black Loading screen forever again.
I'm pretty-sure this was fixed, but it appears to have crept back in.
Not sure when though.

oldman4U, 2020-11-20:
Please comment all issues related to this PR directly using #1272.

Thank you

blueeagle69, 2020-11-20:
@oldman4U 

Yeah I am going to :)

All the best.