bigtreetech/BIGTREETECH-TouchScreenFirmware #1137: [FR] Add ability to get to home screen with 1 click from any screen please

krudoy, 2020-09-28:
Can you please add ability to get to home screen with a single click from any screen. It is really boring to click multiple time 'Back' button

traffic-light, 2020-09-28:
There is not really space for a button that returns you to home.

What, maybe, could be done is pressing a button for 3 sec. But then, when you press for 5sec, the screen goes into mode selection

oldman4U, 2020-09-28:
traffic-light. Wouldn't it be good to collect all those FR in a single ticket like your ReadMe FIRST...

oldman4U, 2020-09-28:
@krudoy 

I agree with you, but the underlying problem here is the depth of menu items. Currently some functions are 4 (or if you also count the function itself - 5) clicks away. If this could be reduced to 3 and maybe the most important once to 2 clicks - your FR would mean 1 or 2 clicks to the home button. I have a concept for reducing the clicks needed to reach all menu items, hope this will be available once.

traffic-light, 2020-09-28:
> traffic-light. Wouldn't it be good to collect all those FR in a single ticket like your ReadMe FIRST...

No not really (atleast, that's my opinion).
You can search for functionrequest with the label enhancement ;-)

> @krudoy
>
> I agree with you, but the underlying problem here is the depth of menu items. Currently some functions are 4 (or if you also count the function itself - 5) clicks away. If this could be [...]

Sometimes it's needed to have a multiple layers of menu's. There are only 8 buttons max per page.

The only easy way i can think of is pressing the return button for a longer time....

(by the way, marlin is also without a return to home button ;-) )

krudoy, 2020-09-28:
> (by the way, marlin is also without a return to home button ;-) )

That's why we like to have a touchscreen with different functionality. And Marlin has an option to get to Info screen after some period of inactivity

I like idea of long or double click. I also consider that 5sec long click to switch between modes can be fully replaced by long click on a wheel.

traffic-light, 2020-09-28:
I don't agree with long clicking on the wheel. You shouldn't use the encoder too much when in touchmode.

Go back to the homescreen after x seconds as an setting could be possible ;-)

You're right. It's a diffrent thing ;-)

oldman4U, 2020-09-28:
Now we are 3 persons with 3 opinions. Add 2 persons more for a total of 6 or seven opinions 😁

traffic-light, 2020-09-28:
And add 3 more on that and it's infinity 😝

oldman4U, 2020-09-29:
Exactly

😎

um 23:13:

cbaugher, 2020-09-30:
> 
> Sometimes it's needed to have a multiple layers of menu's. There are only 8 buttons max per page.
> 
> The only easy way i can think of is pressing the return button for a longer time....

Why is there an 8 button maximum per page?

An auto-return to main menu after timeout would ok if adding another button is not possible.

C|

makmillion, 2020-10-05:
Auto-return to main menu after x- seconds of inactivity sounds like a fantastic idea.

oldman4U, 2020-10-06:
#296 covers this already.

oldman4U, 2020-10-20:
Hello krudoy.

Could you please be so kind and close this ticket. Your feature request is covered in the pinned Feature Request ticket #1177 

Thank you

closed 2020-10-20 by krudoy

kisslorand, 2021-01-06:
A press on the title bar would be ideal but that will remove the notification menu that is assigned to title bar pressing.
What if pressing the title bar goes to notification menu only in the home screen?
Is notification so important that it should be available from any menu?

oldman4U, 2021-01-12:
Before the ability to see the notification by pressing the title bar has been added, the last notifications received has only been visible in the little echo window of the main screen.

So I could imagine that using the title bar to home could work nicely!

kisslorand, 2021-01-18:
@krudoy 
Check PR #1531.

PR #1531 (2021-01-30): Home, sweet home
### Requirements

BTT or MKS TFT

### Description

Long press "Back" button brings you to the home screen. It takes 1/3 of the long press time needed for the Marlin/Touch mode change. 
Holding Back button will not trigger the Marlin/Touch mode change menu.
This function works only in menus where there are icons, not applicable for menus like Notification, Terminal, File selection, Parameter setting, etc. 

### Benefits

Everyone.

### Related Issues

Fixes #1137

### Additional info

This PR doesn't bring any new popups, there are no user selectable/adjustable parameters. It d [... cut]