bigtreetech/BIGTREETECH-TouchScreenFirmware #1139: [BUG] in print fan % change issue

BlackDragon2020, 2020-09-28:
### Description

If you change the fan % while printing you have to change with full, half, off or you have to change in small increments of 1% then wait a few moments to increase another 1%. If you try to hit the increase button more than once it looks like it work but after a few seconds it will change to show it only it increases 1%.

### Steps to reproduce
Start a print
Change fan speed with 1% steps add 5% or 10%

**Expected behavior**
To change the fan speed by increased amount

**Actual behavior**
% jumps back down to original value or original value +1%

### Hardware Variant
Tft35 e3 v3
Current build download today and updated no change.

Will make video if need be.

oldman4U, 2020-09-28:
This is "normal" behavior and there have been other tickets related to it. It affects a more busy system more and also happens with other settings like print speed and flow.

traffic-light, 2020-09-28:
What oldman said. 

The mainboard is busy processing data. A microcontroller is not like a computer. It can't run tasks next to each other.

(Close the ticket when not needed anymore ;-) )

BlackDragon2020, 2020-09-28:
I did not have this issues with my old screen and stock board. Not sure if it's related to the tft or the skr. 

Seems to me after reading @oldman4U the code just needs tweaked. Instead of reading the user input and changing the % instantly it should do something like this rough code:
If Input change
Loop Read input, end after input stops changing for x seconds
Take last input send that as the change.

That seems it should fix it, I will poke around in the code. I have been but it's been years and I'm rusty and never got into GUI, but I maybe able to address this.

traffic-light, 2020-09-29:
I serial always on for you?

#843

oldman4U, 2020-09-30:
Resources are very limited on the TFTs and a numpad would be the best solution I believe also.

BlackDragon2020, 2020-09-30:
@oldman4U I don't know if it's because I'm rusty and not remembering how to do stuff 100% or b cause it just so full already, but a small function with 3 variables added to the fan menu cause a flash overflow problem. What it did was do a quick loop that counts the number of times you push the increase or decrease button then pass that to the switch code, but I can't get it to compile due to flash over flow and it's 20 bytes lol. I will keep playing around see if I can find out what's up.

oldman4U, 2020-09-30:
Well. You are a black dragon and I am a old man. So you have to know and do it...😁

traffic-light, 2020-09-30:
The flash overflow is a known issue. Btt is working on it by removing all theblanguages on compile. At least that's sort of what they told.

They're making it that you need to set the language when compiling

BlackDragon2020, 2020-09-30:
> I serial always on for you?
> 
> #843

Yep, first thing I tried. Thanks for the reply.

BlackDragon2020, 2020-09-30:
> The flash overflow is a known issue. Btt is working on it by removing all theblanguages on compile. At least that's sort of what they told.
> 
> They're making it that you need to set the language when compiling

I was not sure if it was because I have not set up an IDE in 8 years, that I was new to platform IO, if flash was full, or there was some settings that needed adjusted in one of the source codes that stopped adding of code.

radek8, 2020-10-01:
Allow in Marlin
#defune REPORT_FAN_CHANGE
if it is not enabled, it can cause problems

oldman4U, 2020-10-01:
> The flash overflow is a known issue. Btt is working on it by removing all theblanguages on compile. At least that's sort of what they told.
> 
> They're making it that you need to set the language when compiling

This should be available very soon I heard;-)

traffic-light, 2020-10-01:
> > The flash overflow is a known issue. Btt is working on it by removing all theblanguages on compile. At least that's sort of what they told.
> > They're making it that you need to set the language when compiling
> 
> This should be available very soon I heard;-)

Soon is a word that can be 2days~2months with btt :-P
Verry soon is especially longer ;-)

oldman4U, 2020-10-01:
Did not know BTT is working on it....

oldman4U, 2020-10-01:
> I did not have this issues with my old screen and stock board. Not sure if it's related to the tft or the skr.

The encoder wheel is by design not very reliable. Sometimes turning it slowly does nothing and moving it fast - you never know where it will end. But you are right, values did not change once they have been shown up, without turning the wheel again.

;-)

BlackDragon2020, 2020-10-02:
> The flash overflow is a known issue. Btt is working on it by removing all theblanguages on compile. At least that's sort of what they told.
> 
> They're making it that you need to set the language when compiling

So I redone the language code, deleted the menu in screen settings and only included the header and array need for my specified language and it has dropped flash usage from    87.3% down to 59.8%. I still need to go through the function that reads the config.ini to remove the option to change it there, but its up and working and all I have to do is tell it before compile what language. 

I am rusty and been out the game for a bit, but that rust has gotten a lot of oil. Now I just need to learn how to share on git hub :).

oldman4U, 2020-10-02:
@traffic-light 

Lets wait over the weekend and lets discuss and think about a feature request list and which developer does what after that, OK.;-)

@BlackDragon2020 Cant see even the smallest bit of rust!

traffic-light, 2020-10-02:
@oldman4U it would be great if it is possible to mark something that someone is working on it. It was a bit disappointing that digant73 also made the bedlevel grid. Because I was already working on it. I spent time on it and can't get it back...

Maybe some sort of list where people are working on

oldman4U, 2020-10-02:
A Ticket?😎

um 18:43:

oldman4U, 2020-10-02:
> Soon is a word that can be 2days~2months with btt :-P
>
> Verry soon is especially longer ;-)

Sometimes it happens very fast my friend.

#1138

_update icons, fix ABL G-code sending on mode change, Fix text overlap in update screen **Move language translations to SPI flash to reduce main memory usage**_

traffic-light, 2020-10-02:
Gurmeet is a hero.

BlackDragon2020, 2020-10-04:
> Allow in Marlin
> #define REPORT_FAN_CHANGE
> if it is not enabled, it can cause problems

just checked it as I was turning on linear advanced, it was on.

I been thinking, I got an idea to fix it with the push button setup, but I have been turning over in my head an idea to create a them and rearrange the menu layout. It would be more convenient to just change fan speed with a pop text number input. then you could have a core function home page that is just a tap icon input settings. I don't know yet  as there is lots of different setups and what layout works for one printer maynot work for others. At the end of the day the touch code is there and it would remove some bytes from the flash and with a cleaned menu system it could allow for the community to make custom menus / themes.

oldman4U, 2020-10-06:
Have you tried gurmeets PR which reduces the memory usage?

oldman4U, 2020-10-09:
Have you?

BlackDragon2020, 2020-10-09:
Nope, after I learned how to upload to github and add a pull request, I seen the other people's updates. Unfortunately I have not been able to do much, hurt my back remodeling. Will be back at soon.

oldman4U, 2020-10-09:
Welcome to the club. I did my first PR 3 weeks ago.

oldman4U, 2020-10-15:
#1138 seems to at least reduce this issue. Could you please check asap. 

Thank you

BlackDragon2020, 2020-10-17:
> #1138 seems to at least reduce this issue. Could you please check asap.
> 
> Thank you

so, I know of a few bugs, if I am using that one, do I post bugs to him or on main? also, do I just go and get his whole upto date fork or just that part? still new to this whole github working group thing.

oldman4U, 2020-10-18:
Hi. Use the whole commit and report directly if the bug is related to his code.

oldman4U, 2020-10-22:
Hi.

Have you been able to test the new version in the meantime. The fix is now part of the master branch.

Thank you

oldman4U, 2020-10-26:
@BlackDragon2020 

Do we still need this ticket?

Please reply. Thank you

BlackDragon2020, 2020-10-28:
> @BlackDragon2020
> 
> Do we still need this ticket?
> 
> Please reply. Thank you

just now getting to testing. this screen is a replacement, it should be able to do the basic stuff the old one did without issues so until its fixed I believe the ticket is needed.

don't get me wrong I love the screen  and what it offers and don't find bug testing or help fixing if I have the time, but I do think the big bugs and basic functions should be fixed for end consumers.

thanks will report back soon.

oldman4U, 2020-10-28:
Yes it is merged. So you still see the reported issue in the same amount like before using the latest master branch from today?

BlackDragon2020, 2020-10-28:
Yep, if I try to go up or down by 4,5, 6 % same issue. 1% pause then another works, but having to change fan settings like that can be a pain. 

Tbh I only mess with it if I'm testing something or trying to fix an issue on the fly, but a big none the less.

oldman4U, 2020-11-09:
OK. So now that the language is removed from memory, it would eventually be possible to implement what you suggested as a solution to the problem you reported. Do you want to implement it and contribute to the firmware? Would you be able to do so?

Would be great, thank you

BlackDragon2020, 2020-11-10:
> OK. So now that the language is removed from memory, it would eventually be possible to implement what you suggested as a solution to the problem you reported. Do you want to implement it and contribute to the firmware? Would you be able to do so?
> 
> Would be great, thank you

I would love to, but I am swapped atm. I just got my dual 5015 Installed I have had for 2 months.

From playing with the new fans, there is something crazy going on for sure, even using half / full buttons depending on the print I get crazy lag in-between it switching. I am using btt skr 1.4t and printing from skr sd.

There is also some issues with the loop back function. Depending on what's happening where you are in menu when you back out to main there is a delay on updating the hot end temp.

There is also some craziness that happens if you print back to back, you most stop a completed print, and if you do not preheat before starting new print you are locked out off setting temps on hot end or heat bed, it will only heat one after it is done with that task it loads the setpoint for the next heating task. You can change until it's done, but if you preheat first both can be heated at the same time.

Me or someone will need to go through the code of the tft firmware or Marlin and see if there is a way to "interrupt" to send a command without effecting the print. There is a issue somewhere.

The pause print also lags like crazy for me, on the original ender screen it would pause almost instantly with the new tft it lags and waits for the last command to finish.

It maybe something to do with the tft uses a different port vs the original.

kisslorand, 2020-11-25:
Check out [this](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1283) PR. Commit 524de01 includes also a fix for this bug.

closed 2020-11-27 by bigtreetech

oldman4U, 2020-12-21:
And a complete rework can be found here: #1417.

BlackDragon2020, please test this new PR and let us know.

Thank you

oldman4U, 2020-12-23:
@BlackDragon2020

Please test the PR, it works GREAT!!!

BlackDragon2020, 2020-12-23:
Can't wait, I have been crazy busy. Will test it as soon as I can. TBH I am 1000% more interested in going through the code hehe.

oldman4U, 2020-12-23:
Let me know once you know what he did, but kisslorand did a miracle job;-)