bigtreetech/BIGTREETECH-TouchScreenFirmware #1140: [Q] vs code and platformio cant find linker

BlackDragon2020, 2020-09-29:
it builds and work, but complains about the linker.
I took a look in to the  platformio.ini (not I am new to platformIO) the board for the tft35 v3 e3 and all the others are STM32F103VC, but the tft35 has am MCU STM32F206. not sure if thats the issue or if the linker files are messed up. not sure it matters.

just looking for why? can someone explain?
thanks for your time.

digant73, 2020-09-29:
it's due to binary max size reached. The max size configured for your variant is 256K - 32K. So at almost 90% of the flash size (256K) you will get the problems. On the other variants the max size is 256 - 28K (just a little bit more).
Bigtreetech is aware of the problem. They must reduce the languages provided in the binary.

oldman4U, 2020-10-09:
@BlackDragon2020 

Does this answer your question?

Please help the community and close the ticket once you do not need it anymore. Thank you

BlackDragon2020, 2020-10-09:
> @BlackDragon2020
> 
> Does this answer your question?
> 
> Please help the community and close the ticket once you do not need it anymore. Thank you

Nope, I removed the language, have 50% ish flash free, still get message and still don't know why PIO has the wrong dev board set.

For being educated, I sure feel very uneducated at times lol.

oldman4U, 2020-10-09:
I haven't seen this linker issue on my mac using the September Release of VSC and everything updated to the latest and greatest. Especially the TFT firmware compiles without any problem and reliable, but have you tried to delete everything in .platformio? In case I had a problem in the past, deleting the packages and platforms and all the other files and folders solved my problems (at least in this area;-))

Can you attach a screenshot how this looks on your screen?

BlackDragon2020, 2020-10-21:
> Any news?

Sorry, between hurting my back and other things I have been crazy busy. I will try and get a screen soon, and will also do a reinstall.

I would still love an answer from someone on the platformio.ini and the development board being listed as an stm32f1 and not a stm32f2. I know pin outs and things are different l. It does work tho lol

oldman4U, 2020-10-21:
No reason to be sorry.

Sure you know VSC and PIO better than me!

I just compiled the firmware from yesterday, it works great and I could not see any linker related error or warning during compiling. So I copied the Text and searched for linker and this gave me a result...;-)

Warning! Cannot find a linker script for the required board! Firmware will be linked with a default linker script!

Do you mean this?

oldman4U, 2020-10-30:
I saw that you started a ticket there also and could only find this: https://community.platformio.org/t/platformio-can-not-find-linker-script-file/6012

closed 2021-01-05 by stale[bot]