bigtreetech/BIGTREETECH-TouchScreenFirmware #1141: Did I cook my TFT 3.5 E3 V3.0?

Quickset45, 2020-09-29:
I have an ender 3 pro.
I am using the SKR-Mini E3 V2.0 board and the TFT 3.5 E3 V3.0 screen.
I've updated the board firmware using the ender .bin file. -no problem
I started to flash the TFT screen but neglected to included the TFT or config.ini files on the first try. Shortly, the error screen came up "found outdated data: fonts, config.  Please insert the SD card with the files and press restart button"
I was being called for dinner and leaving the display it in its error state, I powered down the unit. <<<<<
I came back later, and properly included all three files, but now it always stops after the first file with the same error message.
even after I insert the SD card with the correct files and hit the reset button.

I tried all three files from both the root directory to update, and the root directory to update - Unified Menu Material theme.
neither set of files will budge the error message. In fact when I power the unit with no card at all, I get the same error message.
I've reset the board a dozen times,.
Here are the files/folder in the root director of the SD card:
BIGTREE_TFT35_V3.0_E3.26.x.bin
TFT35 (folder)
config.ini

any help would be appreciated! I'm going to leave it unplugged for a while.

radek8, 2020-09-29:
Does the TFT folder contain songs and icons, or has it just uploaded an empty folder?

oldman4U, 2020-09-29:
The TFT does not like songs from Michael Jackson😎

radek8, 2020-09-29:
> The TFT does not like songs from Michael Jackson😎

I do ;-)

Thijsvanzwam, 2020-09-29:
format the SD card and then try again, that worked for me once.

oldman4U, 2020-09-29:
> > The TFT does not like songs from Michael Jackson😎
> 
> 
> 
> I like him ;-)

Me too. 😎

Quickset45, 2020-09-29:
Thanks for all the comments... you are all Number Ones!
so... I re-formatted my 128 gig SD card Fat... (3 hrs later)
I  re-installed the 3 components... same error message.
yes radek 8, I did check the folder on the SD card to verify that the fonts and .bmp files were in the folders.
So, I have another screen on the way,  and an 8 GB SD card.

radek8, 2020-09-30:
FAT or FAT33?

radek8, 2020-09-30:
Does TFT SD card support 128G?

oldman4U, 2020-09-30:
For firmware update you have to use <8GB SD card. Use the card which came with the hardware.

Quickset45, 2020-09-30:
Windows 10 format gave me the option of Fat x, or NTFS.
No card came with my display,  just cables.
The micro SD card that came with my SKR mini board won't fit on the display, SD only.
will 8G be ok?

radek8, 2020-09-30:
8G is fine

radek8, 2020-09-30:
> Žádné ocenění za odkaz na album Michaela Jacksona „číslo jedna“? :100:

Quickset45, 2020-09-30:
8GB did the trick!
Now to tackle the issue where LCD12864 is all I see when trying to go to Marlin mode.

radek8, 2020-09-30:
Do you have a connected display and a SKR 10pin connector?
The display connected to the EXP3 connector?

oldman4U, 2020-09-30:
Q: Switching to Marlin mode is giving me a empty screen

A: Check your ribbon cable connections to the TFT screen.
When this problem doesn’t occur when first booting in Marlin mode, you should try rotating the encoder wheel. This way Marlin is forced to redraw the screen

Using an "E3" based mainboard, make sure that ONLY the "CR10_STOCKDISPLAY" is activated in Configuration.h.
For all other mainboards, ONLY the "REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER" has to be activated.

Quickset45, 2020-09-30:
all is now working well.  I had the 10pin cable connected to EXP2 instead of EXP3 on my SKR mini board. 
Many thanks.

closed 2020-09-30 by Quickset45