bigtreetech/BIGTREETECH-TouchScreenFirmware #1142: [Q] (Your Question?)

clon1958, 2020-09-30:
Hello
I have a question
How are the coordinates for Manual Leveling set
my dimensions are:
// Size of machine
#define X_MIN_POS -37.5
#define Y_MIN_POS -3
#define Z_MIN_POS 0
#define X_MAX_POS 250
#define Y_MAX_POS 198
#define Z_MAX_POS 250

In Marlin there are:
// The size of the print bed
#define X_BED_SIZE 250
#define Y_BED_SIZE 198

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS -37.5
#define Y_MIN_POS -3
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 250

#define NOZZLE_TO_PROBE_OFFSET {-25, 0, -2.14}

// Most probes should stay away from the edges of the bed, but
// with NOZZLE_AS_PROBE this can be negative for a wider probing area.
#define PROBING_MARGIN 20

// X and Y axis travel speed (mm / min) between probes
#define XY_PROBE_SPEED (133 * 60)

// Feedrate (mm / min) for the first approach when double-probing (MULTIPLE_PROBING == 2)
#define Z_PROBE_SPEED_FAST HOMING_FEEDRATE_Z

// Feedrate (mm / min) for the "accurate" probe of each point
#define Z_PROBE_SPEED_SLOW (Z_PROBE_SPEED_FAST / 2)

Everything is fine in ABL mode

In Manual Leveling, it does not even stand in the center of the table and is shifted along the X axis

oldman4U, 2020-09-30:
Edit the headline of the ticket to something useful bb

closed 2020-09-30 by clon1958

clon1958, 2020-10-01:
where does the table zero point come from when manually leveling

oldman4U, 2020-10-01:
Not sure if you closed the ticket by mistake or not and I am not sure if this is the right repository for your ticket (believe it would be better in Marlin).

I believe you mean the Z-Axis, right? As far as I know, the zero point is the first levelling point. Because of this, you can get plus and minus corrections.