bigtreetech/BIGTREETECH-TouchScreenFirmware #1146: [BUG] (short description)

rasmus-wallin83, 2020-10-02:
I got this error when I tried to update the software on my tft35 screen

[image]

### Description

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

**Actual behavior**

### Hardware Variant

### TFT Firmware Version & Main Board Firmware details

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

oldman4U, 2020-10-02:
Please read the manual on the front page and update the title to something meaningful.

BlackDragon2020, 2020-10-02:
Take the time to post correctly, but close the ticket after your solution is found.

You need to copy the config.ini to the sd card.

closed 2020-10-02 by rasmus-wallin83