bigtreetech/BIGTREETECH-TouchScreenFirmware #1148: [Q] Switch to Marlin/12864/Emulation mode

shuffleznl, 2020-10-02:
Hi there! 
**TL;DR:** how can I switch to Marlin/12864/Emulation mode w/o touchscreen?
Is there a SW/FW option or short-circuit (like with bootloader update) shortcut to get this?

**Background:** I managed to get my GTR v1.0 working on CoreXY (much less documentation vs SKR :)).
Sadly whilst fitting a printed case for TFT35 the touchscreen broke.
I still have image in BTT Touchmode, and the right third of screen (near the flex cable) still works.

Thanks!

zonalimitatore, 2020-10-02:
Keep press knob,until select menu
Or load a config.ini file, where indicate use touch mode (but if your touch not work,im not sure you can do a lot)

shuffleznl, 2020-10-03:
Thanks @zonalimitatore :) I know **how** to switch if the touchscreen worked :)
Keeping the button pressed shows the option for _Marlin Mode_ but because touchscreen area doesn't work I can't actually select it :) The area to select _Touch Mode_ does still work. I'll look into the config-file!

zonalimitatore, 2020-10-03:
You can choose with knob?
If no,no way,need to edit the config file

oldman4U, 2020-10-03:
Upload new config.ini withe zhe default mode set.

shuffleznl, 2020-10-04:
> You can choose with knob?
> If no,no way,need to edit the config file

Pressing knob for 3 sec shows the two options yes, but rotating the knob doesnt work there.

shuffleznl, 2020-10-04:
> Upload new config.ini withe zhe default mode set.

Have tried this, but now keep getting Found Outdated data, Insert SD with required files and reset notification.
I have cloned the entire git-folder, so all files are actual. Checking now with default config.ini.

closed 2020-10-04 by shuffleznl