bigtreetech/BIGTREETECH-TouchScreenFirmware #1150: Include G34 Auto-Alignment to Movement/BedLevel UI

R0bin3D, 2020-10-03:
Please include the command G34 into the bedlevel menu for better user-experience for user with more than one Z-axis.

ETE-Design, 2020-10-05:
Would like this option :-) Also G34 Bed Tramming would be great as an option...

radek8, 2020-10-20:
What's your idea?

R0bin3D, 2020-10-20:
Extra buttons in the bedlevel-menu-point for G34 Auto-Align und G35 Assist-Tramm. That makes it easyer to use. I missed the buttons and must use the cute little terminal on the display atm.

radek8, 2020-10-20:
G34 is started with some parameter, or are the parameters stored in eeprom and just run the G34 command without parameters?
  The parameters could then be edited in the Printer Parameters Menu

radek8, 2020-10-20:
Can you save the output of the M503 and M422 commands from a printer on which you have two stepper motors on the Z axis?

oldman4U, 2020-10-20:
Hello R0bin3D.

Could you please be so kind and close this ticket. Your feature request is covered in the pinned Feature Request ticket #1177 

Thank you

closed 2020-10-20 by R0bin3D

radek8, 2020-10-20:
oldman, Why?

oldman4U, 2020-10-20:
Radek8. 

Read the pinned ticket to see why. One place which covers all.

radek8, 2020-10-21:
@R0bin3D Can I ask for an answer to my question? Thank you

radek8, 2020-10-21:
> Radek8.
> 
> Read the pinned ticket to see why. One place which covers all.

That's why he had to stay open

oldman4U, 2020-10-21:
Typo. Ticket is #1170

SORRY

oldman4U, 2020-10-21:
You can and should still use the ticket. Even it is closed. So we get a single place with all FRs, a usable ticket system and a place where people can discuss about the solution. 

Alternativ we have a ticket system with feature requests on page 2-5 noone reads, duplicate requests and several developers working on the same issue without knowing.

Chad541121, 2020-12-02:
I would like this menu feature also.

Dtcreation, 2021-01-02:
I would like this menu feature too

BlackDragon2020, 2021-01-31:
I was thinking about this, but in a different way. Some people only have one stepper driver but 2 motors. 

What if the tft preformed the g34 and could use some type of relay or multiplex setup that would ground out one of the stepper motors so the other could be adjusted.

I was thinking go over the gcode to see how it works, come up with the parts list and wire gauges so they work for correct amps. I think it would be a great feature and an awesome addon kit BTT could end up selling.

radek8, 2021-01-31:
This isn't real. The solution is two pulleys and a strap.

radek8, 2021-01-31:
Such a connection would be too complicated and difficult to operate from the display. The solution of two motors on one driver is synchronization using a toothed belt

[image]

digant73, 2021-07-27:
@oldman4u this feature was added. G34 support is configurable in config.ini. Once enabled, the g34 feature is available in Bed Leveling two Layer menu. So this FR can be removed from the list

PR #1483 (2021-01-12): Add Z Steppers Auto-Alignment (G34)
Enable/Disable via config.ini
Disabled by default.

Resolve #1150