bigtreetech/BIGTREETECH-TouchScreenFirmware #1151: [BUG] leveling not on right point

Tillema, 2020-10-03:
### Description

To start, imagine a printer with endstops that are not on coordinates (0, 0), but at e.g. (-5, -18) (in Marlin, this is configured with X_MIN_POS, Y_MIN_POS, X_MAX_POS, and Y_MAX_POS).
I've set the leveling to 20 mm from edge. After flashing, the leveling points are not on (20, 20), but on (15, 2).
I fixed it with setting the leveling positions in Leveling.c, but there are a lot of users who don't know this shit.
Could you please update the positions, that is moves to the spot you told to go to?
Currently, on the X-axis, the first leveling point is set as `infoSettings.machine_size_min[X_AXIS] + infoSettings.level_edge`. For the min point it is quite easy to fix: just use `infoSettings.level_edge`. Don't know how to do this on the max position, because that could be outside the bed as well.
### Hardware Variant
Used the latest firmware with the TFT35 E3 V3.0

If you need more information, or testing, I would like to help you.

clon1958, 2020-10-05:
Hello
I also have the same problem machine zero does not match table zero
Show me where to revise the code
Instead of this:
{infoSettings.machine_size_min [X_AXIS] + infoSettings.level_edge, infoSettings.machine_size_min [Y_AXIS] + infoSettings.level_edge}
Write like this:
{infoSettings.level_edge, infoSettings.level_edge}

zonalimitatore, 2020-10-05:
I've no understand.... Where? in marlin mode or touch mode? Ubl,bilinear or corner?

Tillema, 2020-10-05:
C'mon dudes. Just ping the search bar to find the phrase. It's in TFT/src/User/Menu/Leveling.c (for completeness: lines 6 - 11). Pretty clear it regards manual leveling points in Marlin mode.

zonalimitatore, 2020-10-05:
Ok,so it don't keep the marlin's offsets?

Tillema, 2020-10-05:
It's very clear it uses the absolute minimum, which could be negative as well. This is maybe because some printbeds has their origin in the center (mainly delta printers). Somewhere a function to add real bed size and home offsets (from origin) need to be specified in order to get the right spots on where to level.

clon1958, 2020-10-05:
And the solution to the problem will be

oldman4U, 2020-10-08:
Sounds to me like you would know the solution to the problem you reported also....

So what's next?

Tillema, 2020-10-08:
> Sounds to me like you would know the solution to the problem you reported also....

Not exactly. It is possible to set an upper bound that is higher than the size of the bed in Marlin. In that case, points 2, 3, and 4 are not on the right spot.
I know a little bit about programming, so, I was able to figure out how to fix the lower bound, but the problem is the upper bound, which I'm not able to fix professionally.

oldman4U, 2020-10-08:
@guruathwal Sorry to bother you, but could you please show us the right way - if possible.

Thank you

guruathwal, 2020-10-09:
The leveling edge distance is not a specific coordinate. As the name itself specifies, It is the distance the nozzle will move from the bed edges.
The edge distance will be added to the bed min and subtracted from the bed max.
If you set the edge distance to 20 and your bed size is min(X -10,Y -15) max(X 200,Y 240).
The coordinates for leveling will be:
Point 1: 10, 5
Point 2: 180, 5
Point 3: 180, 220
Point 4: 10, 220

oldman4U, 2020-10-09:
Tilema. Does this help somehow?

Tillema, 2020-10-09:
@guruathwal Great you wrote a summary of my initial question. No offense.

@oldman4U Let me try to explain my issue with a picture (which could say more than 1000 words):
[image]
Red square is the bed. Green dots are the correct leveling points while orange are the wrong leveling points. The black square is what is the border as set with the position of the limit switches. This could be different from the bed size, meaning, the min limit switches could be lower than zero and the max limit switches could be higher than bed size.
Currently, it takes the position from the edge of the limit switches, but I want it to take the position from the edge of the bed.

My initial question included a solution for the limit on the min side "Currently, on the X-axis, the first leveling point is set as `infoSettings.machine_size_min[X_AXIS] + infoSettings.level_edge`. For the min point it is quite easy to fix: just use `infoSettings.level_edge`.", but not on the max side.

Sorry that I had to explain it again a different way. I'm not a native speaker, which could make it hard to follow me. Hopefully, this clarifies what I mean.

oldman4U, 2020-10-09:
Hi Tillema.

I understand what you want and what your "problem" is.

For me this is an edge case, because most users are happy to know the size of their beds and the max Z and to type it in at the right place. To be forced to work with negative values, you have to have a special setup and such a special setup requires special knowledge. Of course it would be possible to make a setting like "Take absolute values 0, Take relative values 1, Value X, and let the user define, but this could cause many issues, just to cover an edge case. The question how far such a software should go and how such situations should be handled is always present - i guess.

Leaving the highway you should always expect bumping roads.

Tillema, 2020-10-09:
> For me this is an edge case, because most users are happy to know the size of their beds and the max Z and to type it in at the right place.
Those users are the ones who waste our times (after upgrading their printer with another hotend, causing limit switch offsets), because they don't know where to put the correct coordinates (instead of using LEVELING_EDGE_DISTANCE) in order to level on the right spots. That's why I opened this issue.
I think, setting the right bed size and the limit switch offsets is way easier than struggling with this. Note that I'm not a programmer. Just found what to edit by logical thinking.

guruathwal, 2020-10-09:
@Tillema The TFT needs to know the actual bed coordinates/ build area. You need to specify the actual bed coordinates/build area in the config.ini file, because your axis min/max are beyond the bed coordinates. 

[image]

oldman4U, 2020-10-09:
> Those users are the ones who waste our times (after upgrading their printer with another hotend, causing limit switch offsets), because they don't know where to put the correct coordinates (instead [...]

But imagine where things like this would be, if they would have been made just for the Pro's. They simply would not exist because nobody could afford them.

clon1958, 2020-10-11:
Hello
And, do not ask to drive the data points directly,
as done by MKS-TFT ?

like this:
#the point number of manual leveling: (3,4,5 point available)
> cfg_point_number: 5

#the coordinates of 5 point on manual leveling
> cfg_point1: 30.30
> cfg_point2: 190.30
> cfg_point3: 190,190
> cfg_point4: 30,190
> cfg_point5: 110,110

maybe it will be easier this way

closed 2020-12-17 by stale[bot]