bigtreetech/BIGTREETECH-TouchScreenFirmware #1152: Is BTT Monitoring this GitHub Project?

ptmy1, 2020-10-03:
All I see are users trying to help other users in this GitHub project.  Are any of the replies coming from a BTT employee?  Is anyone from BTT even monitoring this project, or working on fixing reported bugs?

oldman4U, 2020-10-03:
Occasionally.

Sometimes yes, not that i am aware.

Your help is welcome.

oldman4U, 2020-10-03:
I would like to add that the hardware is great but the support here from BTT sucks. Look at the SKR v1.3 repository where many tickets haven't been answered a single time even after months. Or look at all the open tickets where the owner never answered, even someone of the community tried to help. There is no planning and even no list of features users asked for and engineers are working on the same things - but not together - waisting valuable time.

My 2 cent s

CBDesignS, 2020-10-03:
reading my mind @oldman4U, you are spot on. they only went open source so that the end users can band together to fix their short commings. Fixing problems will drag the only coder away from desiging new hardware so they can punt new hardware to punters and make money so they can get paid next month..

oldman4U, 2020-10-04:
There is nothing wrong with making it open source and also with making money with the hardware. We, the users also have advantages and personally I prefer this way over a "black box" solution. The problem is, that BTT does not take care and the same time they are the only one, being able to do so. 

Look at the PR's and the versioning. They are released almost by mistake it seems. No concept, no structure. And because of this, people trying to contribute and help are leaving frustrated after a short time - except some heros like Gurmeet and co.

CBDesignS, 2020-10-04:
I thought I was on the MKS git posting this. my bad as I had both open at the same time and both suffer from the same problems.

oldman4U, 2020-10-04:
MKS offered no open source hardware 15 months ago. So it was never an option for me. Not idea how they do, but it is sad to hear, that it is the same there. But maybe it is just that my expectations are wrong?!

traffic-light, 2020-10-04:
@oldman4U i think it's a good idea to make one ticket where all the feature request come in and who is working on it ;-)

It's easier to see for other users if their feature request is already there and if someone is working on it ;-)

traffic-light, 2020-10-04:
Oh. And maybe better version control... That once a weak there will be a release with the binaries, images, fonts and config files.

And add one to the sub version number.
Keep the master for development. (only a stable release ;-))

thisiskeithb, 2020-10-04:
Using proper GitHub releases/tagging would be fantastic.

nextux, 2020-10-05:
I do find the current github a little confusing for this project.  Would be good to just have source only, and save the binaries for the release tab.

ETE-Design, 2020-10-05:
Remember BitTreeTech need to make money, they do that on constant making new hardware, so that is why the firmware halt... But it's open source, so great people in here make the changes they seems to need... Bad thing is of cause us who don't know anything about programming get lost in the middle...

oldman4U, 2020-10-05:
I have no idea about programming so I try to help otherwise. Problem is lack of structure and that everything finally depends on BTT releasing it.

oldman4U, 2020-10-05:
> Using proper GitHub releases/tagging would be fantastic.

Keith. You have been a very active user some time ago but not so active here the last couple of weeks. What happened?

thisiskeithb, 2020-10-05:
> What happened?

Other priorities, but I’m still testing code/hardware for BigTreeTech and Biqu as well as Marlin.

oldman4U, 2020-10-05:
👍🏻

oldman4U, 2020-10-08:
@ptmy1 

So what's your conclusion? Why did you ask?

ptmy1, 2020-10-08:
@oldman4U 
I was hoping someone from BTT would respond to this, but guess not.  I'm all for open source, but a project needs an owner that actually does something other than riding on the backs of the community.  Since this firmware only works on BTT hardware, BTT should do most of the heavy lifting.  For a company to release Beta software specific to their hardware and call it open source, expecting users to do all the debugging is just unethical in my opinion.

traffic-light, 2020-10-08:
Maybe @bigtreetech should make someone sub-owner(moderator) if they won't do anything to make their firmware better/ more clear for end users....

oldman4U, 2020-10-08:
If this is unethically or not I can not say, what I know for sure is, that it will be very very hard for them to grow their business on that. I bought a Polar fitness tracker almost a year ago and the last update which brought heavily needed functions is 3 months ago. The difference is, that there was a clear roadmap and I knew that it is not fully featured. Here BTT let the potential buyers believe that everything works and that it is easy to install and use. And that's simply not the case.

@traffic-light . I believe we can only help ourself and try to get a kind of structure. Yesterday I got the first time a respond from a user who told me to stop spamming by asking to close a ticket which has not been in use for almost a year. This user does not understand that if everybody does what he wants, this project will be destroyed in no time. The first hardware was released a bit more than a year ago and look at the SKR v1.3 Issues area. This is exactly what happens if you do not take care. But taking care takes time, as you know. So little steps, a few cool team members and BTT delegating others to do certain thing and this all could work very well.

My 2 cents - like always

oldman4U, 2020-10-12:
Finally BTT is very active in other repositories. Seems the TFT buyers did something wrong;-)

@ptmy1 Please be so kind and close this ticket once you do not need it anymore.

Thank you

closed 2020-10-12 by ptmy1

nextux, 2020-10-12:
Why wants an OctoPi UI?  If you are using OctoPi you probably aren't using a screen.

I would love to see the Marlin Touch UI implemented on the BTT TFT but from what I can understand it would take a lot to implement as this is a standalone board.