bigtreetech/BIGTREETECH-TouchScreenFirmware #1153: [Q] How to flash directly without SD card?

flolas, 2020-10-03:
**Your question or details you need. Please describe.**
After a upgrade following instructions from here, the "Menu" button show white screen with glitches and SD Card is not working, then I cant re-upgrade the tft... ¿What can I do for fixing the issue? 
Thanks

**specify the device variant related to your question.**

TFT43 V3 flashed with latest .bin previously compiled from github repository

oldman4U, 2020-10-03:
Follow exactly the Readme and do not ignore errors and warnings during update. This will solve the problem

oldman4U, 2020-10-03:
No upgrade without SD card - just to answer your question. 😎

flolas, 2020-10-03:
@oldman4U I followed the readme and did not ignore errors.
The problem was the SD card provided by SKR board. Really bad quality

closed 2020-10-03 by flolas