bigtreetech/BIGTREETECH-TouchScreenFirmware #1155: [BUG] (ABL Menu has disappeared)

z00mantwo, 2020-10-06:
### Description

Compiled latest version as 10/5/20 with 

default_envs = BIGTREE_TFT35_E3_V3_0

ABL menu has disappeared from "Movement" menu. Appeared on previous version compiled from newest code 9/23/20. Also get following compile errors:

In file included from C:\Users\Garey\.platformio\packages\framework-stm32cube\f1\Drivers\STM32F1xx_HAL_Driver\Inc/stm32f1xx_hal_def.h:46:0,
                 from C:\Users\Garey\.platformio\packages\framework-stm32cube\f1\Drivers\STM32F1xx_HAL_Driver\Inc/stm32f1xx_hal_rcc.h:45,
                 from C:\Users\Garey\.platformio\packages\framework-stm32cube\f1\Drivers\STM32F1xx_HAL_Driver\Inc/stm32f1xx_hal_conf.h:234,
                 from C:\Users\Garey\.platformio\packages\framework-stm32cube\f1\Drivers\STM32F1xx_HAL_Driver\Inc/stm32f1xx_hal.h:46,
                 from C:\Users\Garey\.platformio\packages\framework-stm32cube\f1\Drivers\STM32F1xx_HAL_Driver\Src\stm32f1xx_hal_can.c:161:
C:\Users\Garey\.platformio\packages\framework-stm32cube\f1\Drivers\STM32F1xx_HAL_Driver\Src\stm32f1xx_hal_can.c: In function 'HAL_CAN_GetTxTimestamp':
C:\Users\Garey\.platformio\packages\framework-stm32cube\f1\Drivers\CMSIS\Device\ST\STM32F1xx\Include/stm32f1xx.h:210:32: warning: implicit declaration of function '__CLZ' [-Wimplicit-function-declaration]
 #define POSITION_VAL(VAL)     (__CLZ(__RBIT(VAL)))
                                ^
C:\Users\Garey\.platformio\packages\framework-stm32cube\f1\Drivers\STM32F1xx_HAL_Driver\Src\stm32f1xx_hal_can.c:1097:23: note: 
in expansion of macro 'POSITION_VAL'
     transmitmailbox = POSITION_VAL(TxMailbox);
                       ^~~~~~~~~~~~

In file included from C:\Users\Garey\.platformio\packages\framework-stm32cube\f1\Drivers\STM32F1xx_HAL_Driver\Inc/stm32f1xx_hal_def.h:46:0,
                 from C:\Users\Garey\.platformio\packages\framework-stm32cube\f1\Drivers\STM32F1xx_HAL_Driver\Inc/stm32f1xx_hal_rcc.h:45,
                 from C:\Users\Garey\.platformio\packages\framework-stm32cube\f1\Drivers\STM32F1xx_HAL_Driver\Inc/stm32f1xx_hal_conf.h:234,
                 from C:\Users\Garey\.platformio\packages\framework-stm32cube\f1\Drivers\STM32F1xx_HAL_Driver\Inc/stm32f1xx_hal.h:46,
                 from C:\Users\Garey\.platformio\packages\framework-stm32cube\f1\Drivers\STM32F1xx_HAL_Driver\Src\stm32f1xx_hal_pwr.c:43:
C:\Users\Garey\.platformio\packages\framework-stm32cube\f1\Drivers\STM32F1xx_HAL_Driver\Src\stm32f1xx_hal_pwr.c: In function 'HAL_PWR_EnableWakeUpPin':
C:\Users\Garey\.platformio\packages\framework-stm32cube\f1\Drivers\CMSIS\Device\ST\STM32F1xx\Include/stm32f1xx.h:210:32: warning: implicit declaration of function '__CLZ' [-Wimplicit-function-declaration]
 #define POSITION_VAL(VAL)     (__CLZ(__RBIT(VAL)))
                                ^
C:\Users\Garey\.platformio\packages\framework-stm32cube\f1\Drivers\STM32F1xx_HAL_Driver\Src\stm32f1xx_hal_pwr.c:114:92: note: in expansion of macro 'POSITION_VAL'
 #define CSR_EWUP_BB(VAL)         ((uint32_t)(PERIPH_BB_BASE + (PWR_CSR_OFFSET_BB * 32U) + (POSITION_VAL(VAL) * 4U)))
                                                                                            ^~~~~~~~~~~~
C:\Users\Garey\.platformio\packages\framework-stm32cube\f1\Drivers\STM32F1xx_HAL_Driver\Src\stm32f1xx_hal_pwr.c:403:22: note: in expansion of macro 'CSR_EWUP_BB'
   *(__IO uint32_t *) CSR_EWUP_BB(WakeUpPinx) = (uint32_t)ENABLE;
                      ^~~~~~~~~~~

traffic-light, 2020-10-06:
I think it's moved. @digant73 you made some changes here.

z00mantwo, 2020-10-06:
It isn't moved. Went through every menu. It's just gone. I'm guessing the detection method has broken.

digant73, 2020-10-06:
as far as I know that kind of bug was found and fixed in #1138.
In the last code the BL menu (in move menu) is shown if a BL type (ABL, UBL, MBL) is detected during M503 and M115. For some Marlin FW, the BL type was not detected so no BL menu was shown.
As I workaround, from the terminal menu type M503. After that you should see the BL menu

z00mantwo, 2020-10-06:
Also note that the notifications that ABL is on and fade height never go off like in the old version. If I go to the machine settings and turn on ABL manually then those notifications popup. The ABL menu still does not show up. I believe the issue may be related to this. ABL is on but the new version is not detecting it. M420 returns:

Send: M420
Recv: echo:Bed Leveling ON
Recv: echo:Fade Height 2.00
Recv: ok

So the printer has ABL turned on. The new version never seems to detect this.

z00mantwo, 2020-10-06:
Also if I go back to the 26.1 version everything works and the alerts on ABL and fade height show up as soon as I turn on the printer and the ABL menu is there.

digant73, 2020-10-06:
M420 is to activate/deativate the use of ABL not to detect what kind of BL is provided by Marlin FW (ABL, UBL, MBL etc...).
Your problem should be fixed in #1138.

z00mantwo, 2020-10-06:
I'm on the "master" as of 3 days ago. So I believe that should include the 1138 changes.

digant73, 2020-10-06:
no #1138 is a current PR

z00mantwo, 2020-10-06:
Okay. It's fixed by pulling your #1138 apparently that hasn't been rolled in to the "master".

Sorry for the fire drill.

digant73, 2020-10-06:
good, however it's not my PR

Zowen79, 2020-10-06:
The latest update has killed my TFT35_E3_V3_0. It updated like normal, but as soon as it finished updating the icons my screen went black and said: 
"Found outdated data:

Config: Update required
Insert the SD card with the required files and press the reset button to update."
I recompiled with some changes (changed to Classic Menu), it updated like it normal again, then right back to that same message. I restarted the whole machine, but still didn't work. Help

[image]

z00mantwo, 2020-10-06:
Did you put the config.ini file on the SD? I've seen this myself and just put the config.ini on the SD then it went away. Config.ini had to match the updated firmware.

Zowen79, 2020-10-06:
I will try that. Let you know in a few

Zowen79, 2020-10-06:
That worked. Thanks a lot

discip, 2020-10-07:
I think after the [latest commit](https://github.com/guruathwal/BIGTREETECH-TouchScreenFirmware/commit/462ded278c3f275fb0582d4b26dc0b131f4e87f7) this should be closed, because it is resolved.

closed 2020-10-07 by z00mantwo