bigtreetech/BIGTREETECH-TouchScreenFirmware #1156: TFT35 V2.0 wifi support

ssombra, 2020-10-06:
Hi. I have the TFT35 V2.0 and I wonder if currently it is possible to add a wifi module to it (as I use the SKR 1.3 board I cant add the module there). When I got it several months ago the support was not implemented. Now I did a search on the issues and didn't find specific new information on this. I understand that this screen doesn't have the socket for the module but, if the support in the firmware is there, it should be an easy solder. Anyone could point me to the correct direction?

Slimgig, 2020-10-06:
The screen does have a socket for the wifi module there are plenty of videos on YouTube for it I have the wifi module but have not hooked it up yet just got it the other day

Slimgig, 2020-10-06:
But I have seen videos for it too that I am going to go off of

ssombra, 2020-10-06:
> The screen does have a socket for the wifi module there are plenty of videos on YouTube for it I have the wifi module but have not hooked it up yet just got it the other day

The V2.0? 
I have mine in a case right now, but I think there is not one in my screen, just the space and the pins to solder one like in the manual:

[image]

radek8, 2020-10-06:
You have 3 options
1. You can solder the connector to the WIFI position
2. use the USART3 connector
3. connect WIFI to SKR1.3

https://www.youtube.com/watch?v=nEhDPCGki80

[image]

Slimgig, 2020-10-06:
I have the 3.0 sorry about that

ssombra, 2020-10-06:
> You have 3 options
> 
> 1. You can solder the connector to the WIFI position
> 2. use the USART3 connector
> 3. connect WIFI to SKR1.3
> 
> https://www.youtube.com/watch?v=nEhDPCGki80
> 
> [image]

Thank you very much for that. If I buy the module from BTT (ESP01S) I guess the easiest way is option #1. How is the wiring for option #2? I only see 3 pins while the modules usually have 8.

oldman4U, 2020-10-06:
What kund of functionality would you expect from the WIFI connection?

radek8, 2020-10-06:
connection for flash is here
[image]

radek8, 2020-10-06:
[image]

ssombra, 2020-10-06:
> What kind of functionality would you expect from the WIFI connection?

Basic controls (temperature, movement, a terminal, print files from the board SD). The modules are cheap so why not adding it to the printer.

ssombra, 2020-10-06:
> connection for flash is here
> [image]

Thanks for all the information Radek8! I will definitely take a look into it.

oldman4U, 2020-10-07:
> > What kind of functionality would you expect from the WIFI connection?
> 
> 
> 
> Basic controls (temperature, movement, a terminal, print files from the board SD). The modules are cheap so why not adding it to the printer.

The web interface is ok and offers most functions Pronterface has. So it should do what you want. You have no SKR PRO, right!?

ssombra, 2020-10-07:
> The web interface is ok and offers most functions Pronterface has. So it should do what you want. You have no SKR PRO, right!?

No, SKR 1.3. I already ordered the module from BTT along with a writer.

oldman4U, 2020-10-07:
Perfect. 

Please help the community and close the ticket once you do not need it anymore. Thank you

ssombra, 2020-10-07:
Sure, thanks everyone for the support.

closed 2020-10-07 by ssombra

oldman4U, 2020-10-07:
Reopen it if needed.

Thank you and happy printing

SerjPr, 2020-10-08:
I just soldered the WIFI module to the contacts. After pre-flashing it. Everything works great.
[image]

SerjPr, 2020-10-08:
ESP07, ESP12E, ESP12F, без проблем подходят. Только под них надо подложить прокладку. Чтоб плата не касалась контактов.

oldman4U, 2020-10-08:
Congratulations. Looks great!

ssombra, 2020-10-08:
If I am not wrong, the module that BTT sells has only 8 pins, right? This seems to have 16.

radek8, 2020-10-08:
the car has 4 wheels and a motorcycle 2, and you will drive on both :-)

SerjPr, 2020-10-09:
> If I am not wrong, the module that BTT sells has only 8 pins, right? This seems to have 16.

Only "TX, RX, VSS, GND, Ch_PD" is used. You can only solder them. Well, and the resistor on GPIO 15, or just short-circuit to the case.

SerjPr, 2020-10-09:
But there is a flaw in my way. If you need to reflash it, then the module needs to be soldered. Or at least unsolder "RX ,TX". And they will be soldered to them for firmware

ssombra, 2020-10-26:
Well, I received the ESP module from BTT and it was way more easy than I imagined. On the one hand, it turns out that my TFT (TFT53 V2.0) did have the socket already soldered, I guess it is a newer version that the one shown in the manual. On the other hand, the ESP modules sold by BTT are already flashed with ESP3D, so it was pretty much a plug and play operation. Thanks everyone for the support.