bigtreetech/BIGTREETECH-TouchScreenFirmware #1157: SKR 1.4 Turbo not connecting to TFT 35 v2.0

IIIxpHunterIII, 2020-10-06:
I'v got an BIGTREETECH SKR 1.4 Turbo connected to the TFT 35 V2 using the RS232 Port on the TFT and the TFT port on the board. Updated the screens firmware, set the baudrate @ 250000 on both Marlin (of which I'm using the 2.0.x) and the screen.

Could anybody help me get the darn thing to communicate please?

Regards

z00mantwo, 2020-10-06:
Make sure you plug is orientated correctly on both sides. Are you just using the RS@#@ port or are you using the display port (10 pin) connector also? If so is that working?

oldman4U, 2020-10-06:
You get No Printer Attached?

Can you make a picture showing the connection between mainboard and TFT and attach it here. 

Thank you

radek8, 2020-10-06:
SERIAL_PORT in Marlin must be set to 0
Do you have it set?
#define SERIAL_PORT 0
or
#define SERIAL_PORT_2 0

radek8, 2020-10-06:
> Make sure you plug is orientated correctly on both sides. Are you just using the RS232 port or are you using the display port (10 pin) connector also? If so is that working? Also on the back of the TFT35 make sure you are plugged in to the correct port.

TFT 35 V2.0 does not have a 10pin connector

zonalimitatore, 2020-10-07:
wrong baud. Need set it to 115200,in marlin firmw,and in the screen

radek8, 2020-10-07:
I have 250000, and everything is fine

oldman4U, 2020-10-07:
As long as it is the same it will work.

zonalimitatore, 2020-10-07:
On a 1.4 turbo?

oldman4U, 2020-10-07:
@zonalimitatore 

Is there a problem with the baudrate on the Turbo? I have a non turbo.

zonalimitatore, 2020-10-07:
I have it. And if i set a different baud,the screen don't connect to printer

oldman4U, 2020-10-07:
Different baudrate can not work. But using the same baudrate for firmware and TFT, the baudrate used should not matter.

zonalimitatore, 2020-10-07:
Yep...I know it. But I've set 250000 for both,and never work  With 115200 yes

IIIxpHunterIII, 2020-10-07:
I've gotten everything just as you said apart from both Baudrates at 0, the firt is set to -1 and the second to 1 (tried 0 to for the second). I'll give it a try if you think that'll help.

oldman4U, 2020-10-07:
In our heads we replace baudrate with serial port😁

Change one to 0 to allow the TFT to talk with the mainboard. -1 is USB which you can leave activated as well.

zonalimitatore, 2020-10-07:
😂😂😂
#define SERIAL_PORT      -1
#define SERIAL_PORT_2   0
#define AUDRATE 115200

IIIxpHunterIII, 2020-10-07:
That's how I did it at first, doesn't work. @zonalimitatore

zonalimitatore, 2020-10-07:
How have connect the wire on motherboard?

radek8, 2020-10-07:
send Marlin configuration files
send photo of engagement

IIIxpHunterIII, 2020-10-07:
[image]
[image]
Now it's beeping like crazy saying ekk.

Here's my config:
/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2020 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <https://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 */
#define CONFIGURATION_H_VERSION 020007

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * https://reprap.org/wiki/Calibration
 * https://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * https://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * https://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * https://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer start with one of the configuration files in the
// config/examples/delta directory and customize for your machine.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
[... 2328 lines cut ...]

/**
 * R/C SERVO support
 * Sponsored by TrinityLabs, Reworked by codexmas
 */

/**
 * Number of servos
 *
 * For some servo-related options NUM_SERVOS will be set automatically.
 * Set this manually if there are extra servos needing manual control.
 * Leave undefined or set to 0 to entirely disable the servo subsystem.
 */
//#define NUM_SERVOS 3 // Servo index starts with 0 for M280 command

// (ms) Delay  before the next move will start, to give the servo time to reach its target angle.
// 300ms is a good value but you can try less delay.
// If the servo can't reach the requested position, increase it.
#define SERVO_DELAY { 300 }

// Only power servos during movement, otherwise leave off to prevent jitter
//#define DEACTIVATE_SERVOS_AFTER_MOVE

// Allow servo angle to be edited and saved to EEPROM
//#define EDITABLE_SERVO_ANGLES

oldman4U, 2020-10-07:
So we see some progress. No "No printer attached" anymore?

You can attach the files here directly, not as text but the files. This will help to keep this ticket readable, thank you.

Are you sure you have all those axes and extruders?

#define X_DRIVER_TYPE TMC2208
#define Y_DRIVER_TYPE TMC2208
#define Z_DRIVER_TYPE TMC2208
#define X2_DRIVER_TYPE TMC2208
#define Y2_DRIVER_TYPE TMC2208
#define Z2_DRIVER_TYPE TMC2208
#define Z3_DRIVER_TYPE TMC2208
#define Z4_DRIVER_TYPE TMC2208
#define E0_DRIVER_TYPE TMC2208
#define E1_DRIVER_TYPE TMC2208
#define E2_DRIVER_TYPE TMC2208
#define E3_DRIVER_TYPE TMC2208
#define E4_DRIVER_TYPE TMC2208
#define E5_DRIVER_TYPE TMC2208
#define E6_DRIVER_TYPE TMC2208
#define E7_DRIVER_TYPE TMC2208

Most likely this has to be like this:
#define X_DRIVER_TYPE TMC2208
#define Y_DRIVER_TYPE TMC2208
#define Z_DRIVER_TYPE TMC2208
//#define X2_DRIVER_TYPE TMC2208
//#define Y2_DRIVER_TYPE TMC2208
//#define Z2_DRIVER_TYPE TMC2208
//#define Z3_DRIVER_TYPE TMC2208
//#define Z4_DRIVER_TYPE TMC2208
#define E0_DRIVER_TYPE TMC2208
//#define E1_DRIVER_TYPE TMC2208
//#define E2_DRIVER_TYPE TMC2208
//#define E3_DRIVER_TYPE TMC2208
//#define E4_DRIVER_TYPE TMC2208
//#define E5_DRIVER_TYPE TMC2208
//#define E6_DRIVER_TYPE TMC2208
//#define E7_DRIVER_TYPE TMC2208

oldman4U, 2020-10-07:
Cables seem to be connected right. The TFT talks to the mainboard.

IIIxpHunterIII <notifications@github.com> schrieb am Mi. 7. Okt. 2020 um
15:27:

> [image: 20201007_152252]
> <https://user-images.githubusercontent.com/16221970/95336800-5c8e5f00-08b1-11eb-83e3-f6a436686145.jpg>
> [image: 20201007_152312]
> <https://user-images.githubusercontent.com/16221970/95336807-5e582280-08b1-11eb-9565-c2ff9a1963a4.jpg>
> Now it's beeping like crazy saying ekk.
>
> Here's my config:
> /**
>
>    - Marlin 3D Printer Firmware
>    - Copyright (c) 2020 MarlinFirmware [
>    https://github.com/MarlinFirmware/Marlin]
>    -
>    - Based on Sprinter and grbl.
>    - Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
>    -
>    - This program is free software: you can redistribute it and/or modify
>    - it under the terms of the GNU General Public License as published by
>    - the Free Software Foundation, either version 3 of the License, or
>    - (at your option) any later version.
>    -
>    - This program is distributed in the hope that it will be useful,
>    - but WITHOUT ANY WARRANTY; without even the implied warranty of
>    - MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
>    - GNU General Public License for more details.
>    -
>    - You should have received a copy of the GNU General Public License
>    - along with this program. If not, see https://www.gnu.org/licenses/.
>    -
>
> */
> #pragma once
>
> /**
>
>    - Configuration.h
>    -
>    - Basic settings such as:
>    -
>    -
>       - Type of electronics
>    -
>       - Type of temperature sensor
>    -
>       - Printer geometry
>    -
>       - Endstop configuration
>    -
>       - LCD controller
>    -
>       - Extra features
>    -
>    - Advanced settings can be found in Configuration_adv.h
>    */
>    #define CONFIGURATION_H_VERSION 020007
>
>
> //===========================================================================
> //============================= Getting Started
> =============================
>
> //===========================================================================
>
> /**
>
[... 1956 lines cut ...]
> //#define TEMPERATURE_UNITS_SUPPORT
>
> // @section <https://github.com/section> temperature
>
> // Preheat Constants
> #define PREHEAT_1_LABEL "PLA"
> #define PREHEAT_1_TEMP_HOTEND 210
> #define PREHEAT_1_TEMP_BED 70
> #define PREHEAT_1_FAN_SPEED 0 // Value from 0 to 255
>
> #define PREHEAT_2_LABEL "ABS"
> #define PREHEAT_2_TEMP_HOTEND 240
> #define PREHEAT_2_TEMP_BED 90
> #define PREHEAT_2_FAN_SPEED 0 // Value from 0 to 255
>
> /**
>
>    - Nozzle Park
>    -
>    - Park the nozzle at the given XYZ position on idle or G27.
>    -
>    - The "P" parameter controls the action applied to the Z axis:
>    -
>    - P0 (Default) If Z is below park Z raise the nozzle.
>    - P1 Raise the nozzle always to Z-park heigh

zonalimitatore, 2020-10-07:
Oh Lord...

1 Red jumper to 5V and VDD
2 define baudrate 115200
3 define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
4 remove all unused stepper define (like @oldman4U say)
5 //#define NUM_M106_FANS 1 ?????????????????????????????

IIIxpHunterIII, 2020-10-07:
Whats the questionmarks all about,
and I shouldn't need to set it to 5v should I?

Also some of you guys said that I can use 250000, I'll use 115200 tho if you say so.

IIIxpHunterIII, 2020-10-07:
Oh and it changed nothing, seems like the display gets overloaded since it does not react when i click okay on the echo tab.

zonalimitatore, 2020-10-07:
5v or you power with usb
I tell my work with 115200 (do what you wanna)
???? It's because I never see that definition write in that way!

Your cable have one pin detach from others?

oldman4U, 2020-10-07:
Did you change the jumper on the mainboard?

Leave the baud rate. It works.

No need to define REPRAP it is a TFT VERSION TWO.

IIIxpHunterIII, 2020-10-07:
Yes, I've now changed it to 12v. Should I uncomment the user controlled fan thing or enable anything else?

IIIxpHunterIII, 2020-10-07:
Oh and no, there shouldnt be any detached cables, I've freschly crimped a jst connector on the tft cable. And yes the pins should be rightly oriented, atleast it is how ive always seen it done.

IIIxpHunterIII, 2020-10-07:
> So we see some progress. No "No printer attached" anymore?
>
> You can attach the files here directly, not as text but the files. This will help to keep this ticket readable, thank you.
>
> Are you sure [...]

Oh, I'm verry sorry, I've overlooked your reply as it seems. Yes the No reply thing is gone, now it freezes the second after the message disappears. If it helps, thre appears a small white dot in the top left corner of the screen after freezeup and a yellow one on the right.

zonalimitatore, 2020-10-07:
click on it

IIIxpHunterIII, 2020-10-07:
Trust me, I've done that ;P

zonalimitatore, 2020-10-07:
no response? Need to flash the bin with folder content the icons and fonts

IIIxpHunterIII, 2020-10-07:
> no response? Need to flash the bin with folder content the icons and fonts

Done it, now it says no connection again.

IIIxpHunterIII, 2020-10-07:
[config.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5341771/config.txt)

I suppose there's not but could there be something wrong here?

IIIxpHunterIII, 2020-10-07:
[image]
Back to the beginning, but those are the dots

zonalimitatore, 2020-10-07:
#  Note: Ensure that following options are enabled in Marlin firmware
#
#  M115_GEOMETRY_REPORT (in Configuration_adv.h)
#  REPORT_FAN_CHANGE (in Configuration_adv.h)
#  EMERGENCY_PARSER (in Configuration_adv.h)

Have you check it?
I don't think you can enable emergency parser with this board.

Anyway, you have the screen freezing,or it's responsive?

IIIxpHunterIII, 2020-10-07:
It is connecting but freezes instantly after saying ready. It sometimes gives an ecobefore freezing, which finally was displayed completely as echo unknown command m105

IIIxpHunterIII, 2020-10-07:
Oh and yes i've configured all 3 in adv

zonalimitatore, 2020-10-07:
That it's report temperature (m105)

IIIxpHunterIII, 2020-10-07:
I know but what does that tell me in regards to the freezing problem?

zonalimitatore, 2020-10-07:
I suggest what I say before.
In marlin set BAUDRATE to 115200,and the same in the screen

IIIxpHunterIII, 2020-10-07:
Already did that I've tried port -1 1 too

IIIxpHunterIII, 2020-10-07:
In conclusion: Just configured the bare minimum on a fresh marlin release, flashed it to the SKR. Gotten a fresh TFT file, flashed it to the screen. set the baud, the serial port and the three options in .adv and NOTHING CHAGED. I am starting to question myself or the BTT guys.

IIIxpHunterIII, 2020-10-07:
[image]
Last straw I'm holding onto rn is this, can that be the problem? Atleast thats how I've wired it up.

radek8, 2020-10-07:
Did you compile the FW for the display yourself?

radek8, 2020-10-07:
> [image]
> Poslední kapka, kterou se držím rn, je tohle, může to být problém? Alespoň tak jsem to zapojil.

it's alright

IIIxpHunterIII, 2020-10-07:
Yes,I have used the precompiled ones and a self compiled one. The wiring was not at fault.

radek8, 2020-10-07:
use speed 115000 as written by zonalimitatore!

precompiled for TFT35 V2.0 versions?

IIIxpHunterIII, 2020-10-07:
Yep

IIIxpHunterIII, 2020-10-07:
[image]

radek8, 2020-10-07:
you have a problem communicating on the serial line
how do you have a long RS232 cable?

IIIxpHunterIII, 2020-10-07:
What exactly do you mean? I used the original cable but with jst xh connectors crimped to both ends ( out of the box it is just one)

radek8, 2020-10-07:
Only if the cable is not over a meter long.
Do you have a quality source? Can't it cause interference?

IIIxpHunterIII, 2020-10-07:
I am not sure. I was under the impression that the original one should be good, but I can try recrimping.

radek8, 2020-10-07:
Try an even lower speed of 57600 if it doesn't solve the communication problem ...

radek8, 2020-10-07:
crimping the power supply cables will not help. I asked if you had any inflated capacitors

zonalimitatore, 2020-10-07:
It's very strange all that messages "unknown command". Too many

IIIxpHunterIII, 2020-10-07:
Soooooo 57600 worked...but only with nothing other than the screen connected and nothing configured. when i flashed my normal marlin and put in the stepperdrivers it dropped out even at 2400. and no there shouldnt be inflated caps. the crimping was about te five cables connecting tft and printer.

radek8, 2020-10-07:
so you have a huge problem with interference

zonalimitatore, 2020-10-07:
Very very bad problem. Maybe screen defect?

[154 more replies]

closed 2020-12-30 by stale[bot]