bigtreetech/BIGTREETECH-TouchScreenFirmware #1158: [Q] Does anyone know what causes this when browsing Gcode?

blueeagle69, 2020-10-06:
Hi.

I've noticed a weird issue when browsing GCode on my onboard SD card.
I think it's fine when files are on TFT media.
[image]

The file name shows as B1SENS~1.GCO when I select it.
The long filename is "B1 sensor_bracketWIP V3_1_0.2mm_PLA_29m.gcode".
I do wish the TFT would support long filenames when printing from onboard media.

zonalimitatore, 2020-10-07:
maybe dont like .GCO
Try rename to .GCODE

blueeagle69, 2020-10-07:
It has a .GCODE extensuin, but the TFT shortens filenames when printing from the Onboard SD  😒 All the rest are fine, but I always seem to get one that shows "????"

zonalimitatore, 2020-10-07:
Try to rename it,more short and without any point or symbol

blueeagle69, 2020-10-07:
Was about to try the same thing. Will  let you know

blueeagle69, 2020-10-07:
Nope I shortened the filename right down.
I also removed all underscores although the rest have them.

It still shows as "????"

zonalimitatore, 2020-10-07:
there is again 0.2 in filename?

blueeagle69, 2020-10-07:
Okay then.

I shortened the filename to B1 Sensor Bracket.gcode.
Same thing.
I then removed the B1 and left it as Sensor Bracket.
That worked.
I then put it back minus the space after B1, it came back as ???? on the TFT.
I then put the B back and left out the "1". It worked.
So for some reason the TFT doesn't like a number after the first character of the filename?

zonalimitatore, 2020-10-07:
Ahahahahahaha.....omg
One day...one bug

blueeagle69, 2020-10-07:
Confirmed. I created a dummy Gcode file and called it A3 Test.gcode.
It showed as ???? on the TFT.
I think I may have stumbled across a bug.
Only in TFT

zonalimitatore, 2020-10-07:
Ive name this without problem.....Now i try like you do

[image]

blueeagle69, 2020-10-07:
Yeah that would work. It just doesn't like a number after the first character for some reason.

zonalimitatore, 2020-10-07:
My work😯
[image]

blueeagle69, 2020-10-07:
I just tested again. The bug is limited to onboard SD printing 😔

zonalimitatore, 2020-10-07:
🙈🙈🙈🙈🙈

zonalimitatore, 2020-10-08:
Dude,change the label in BUG.It's confirmed

zonalimitatore, 2020-10-08:
I mean delete QUESTION and set like BUG😉

blueeagle69, 2020-10-08:
> I mean delete QUESTION and set like BUG😉

Apologies, lol.
I changed the {Q] label to [BUG] immediately, but for some reason the tag hasn't updated.
But it definitely shows [BUG] in the title.

blueeagle69, 2020-10-08:
I suppose I could close this one and start a nee bug report and mention this one?

oldman4U, 2020-10-08:
Not sure, because so far you the only one having this problem. I just tested it and can not reproduce it on my machine. Do you want to test the attached code file?

[C3_Dusche Mitnehmer.gcode.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5349351/C3_Dusche.Mitnehmer.gcode.zip)

I am able to see the right name when I try to print it from the SD card which is interred into the mainboards card reader.

Let me know

radek8, 2020-10-08:
How old is the FW in the display? This problem was resolved some six months ago.

oldman4U, 2020-10-08:
Am I remember right, that there was such an issue before... Unfortunately Git does not allow to search for "?", at least it shows no result even I can see it on the first page;-)

blueeagle69, 2020-10-08:
> Not sure, because so far you the only one having this problem. I just tested it and can not reproduce it on my machine. Do you want to test the attached code file?
> 
> [C3_Dusche Mitnehmer.gcode.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5349351/C3_Dusche.Mitnehmer.gcode.zip)
> 
> I am able to see the right name when I try to print it from the SD card which is inserted into the mainboards card reader.
> 
> Let me know

I just tried your gcode on my onboard SD.
It shows as "????" on screen.

oldman4U, 2020-10-08:
From when is your firmware?

blueeagle69, 2020-10-08:
Helloo again 😀
It's a pretty recent build I downloaded a little over a week ago.
The last commit was #1060

blueeagle69, 2020-10-08:
> How old is the FW in the display? This problem was resolved some six months ago.

It was a recent build, but I just downloaded the latest master again just for the hell of it.
Same issue when listing files from onboard media.
Any file with a number directly after the first character displays as ????. If I click it, it shows me the 8.3 filename.
(Long filenames from onboard media would ne nice).
Any file uploaded to TFT media displays as expected.
Also Marlin mode is unaffected.

oldman4U, 2020-10-09:
Which Marlin? I use v2.0.7 official release. And do we know your mainboard?

oldman4U, 2020-10-09:
And WHY did you change the picture??

blueeagle69, 2020-10-10:
Apologies for the delay in getting back to you.
I was bored so thought I would mess about with my profile image!
My Marlin version is 2.061
And my board is a SKR 1.4

blueeagle69, 2020-10-10:
I didn't actually realise 2.0.7 was released.
Downloading now 😀

blueeagle69, 2020-10-10:
Amazingly since updating to Marlin 2.0.7 the issue has gone.
Filenames that were showing as ???? before now show as normal.
Annoyingly though Gcode stored on the onboard SD still revert to 8.3 when selected.

oldman4U, 2020-10-10:
What a wonderful start into the day👍🏻

oldman4U, 2020-10-10:
Time to close this ticket 😁

blueeagle69, 2020-10-10:
😀

blueeagle69, 2020-10-11:
Well that's annoying and bizarre.
I just updated the printer to Marlin 2.0.7.1 and the ???? thing has returned 😣

oldman4U, 2020-10-12:
But this problem is obviously coming from Marlin. You should open a ticket there.

blueeagle69, 2020-10-13:
I was thinking the same thing sometime after posting.
Will close (again)

closed 2020-10-13 by blueeagle69

oldman4U, 2020-10-13:
And you know how to reopen it😁

Strange that only you can see this. Maybe it is a setting!??

blueeagle69, 2020-10-13:
🤣
Yeah personally I think my 3D printer is trying to drive me to the asylum.
I think for now I will simply remember not to name my files in a way that causes the blasted ???? 😉
All the best!

oldman4U, 2020-10-13:
Maybe a good decision. 👍🏻

oldman4U, 2021-01-31:
Verified on my printer using current Marlin and TFT fw, so I added it to the verified Bugs list.

guruathwal, 2021-03-08:
what's the status of this bug with reference to the latest commit? does it still exist?

oldman4U, 2021-03-08:
Was there end of Jan so I guess yes. Will verify and let you know asap.

oldman4U, 2021-03-08:
[M75 Lorem ipsum dolor sit amet.gcode.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6102999/M75.Lorem.ipsum.dolor.sit.amet.gcode.zip)

Hi. Please unzip and try to print from bSD On Board SD card slot. The absolutely same card (microSd to SD card adapter) shows the names correctly when trying to print from TFT SD card slot.

oldman4U, 2021-03-09:
Hi. Could you please check PR#1699 which likely solves the problem.

Thank you

oldman4U, 2021-03-09:
Hi birdy;-)

Could you please check Gurmeet's PR to see if it fixes the reported issue: https://github.com/guruathwal/BIGTREETECH-TouchScreenFirmware.git

Thank you

oldman4U, 2021-03-10:
Thank you Gurmeet. Verified that it works so I remove the issue from the Verified Bugs list.

PR #1699 (2021-03-10): Bugfix: '???' instead of Long file name for Onboard SD
- Fix: long file name not updated correctly causing '???' displayed in place of file name. (#1158)

resolves: #1158