bigtreetech/BIGTREETECH-TouchScreenFirmware #1159: TFT 35 V2.0

DNJ1980, 2020-10-07:
Hallo, ich habe auf meinem Display ein Update gemacht nun funktioniert mein Touchscreen nicht mehr. Ich habe schon einiges Probiert aber es will nicht funktionieren. Ich habe die .io geändert die config.h und neu Kompiliert aber das hat mich auch nicht weiter gebracht. Den Bootloader fix habe ich schon drauf und kann per SD- karte updaten. Es kommen auch keine Fehlermeldungen. Es ist auch alles richtig angeschlossen und kein Kabel ist ab. Vo dem Update hat es ja noch funktioniert. Bitte um Hilfe

MFG DNJ

oldman4U, 2020-10-07:
Hi. Kannst du auf englisch schreiben.

DNJ1980, 2020-10-08:
Hello. Unfortunately only with a translator.

Hello, I have made an update on my display now my touchscreen no longer works (I see everything but cannot select anything). I've tried quite a few things, but it won't work. I changed the .io, the config.h and recompiled it, but that didn't help me either. I already have the fix bootloader on it and can update via SD card. There are also no error messages. Everything is connected correctly and no cable is disconnected. It was still working before the update. Please help

MFG DNJ

oldman4U, 2020-10-08:
I hope it is ok for the community when I take this over in german. This probably will speed it up. 

Welche Firmware hast du genau aufgespielt und was war auf der SDCard genau drauf vor dem Update und danach?

DNJ1980, 2020-10-08:
Hallo. Ich habe die BIQU_TFT_APP1_V2.0.26x.bin drauf gespielt. Nach dem Update wurde der Ordner TFT35 in TFT35.CUR geändert.

DNJ1980, 2020-10-08:
Ich habe ein Video gemacht. 

https://youtu.be/VIJQeU3Lg0U

oldman4U, 2020-10-08:
OK, das scheint alles gut zu sein. Kannst du selbst Firmware kompilieren und welche Größe und Format hat die SD Karte?

DNJ1980, 2020-10-08:
Habe ich schon Probiert in der Plattformio.ini habe ich auch den Namen geändert bei Default_envs, aber das gleiche Problem. Die .bin Datei in .Pio/build habe ich natürlich auch verwendet.

oldman4U, 2020-10-08:
Versuch einmal eine SD Card kleiner, maximal gleich 8GB FAT32 oder FAT16.

DNJ1980, 2020-10-08:
Mit der 8GB Karte hatte ich es am Anfang probiert. Damit hatte ich das gleiche. Eine andere habe ich leider nicht. Marlin kann ich mit beiden Karten gut Flashen.

oldman4U, 2020-10-08:
So weit mir bekannt kann der TFT Bootloader nur mit 8GB oder kleiner umgehen, ich würde daher diese benutzen.

Mach bitte eine leere Textdatei, benenne sie Reset.txt und flashe damit einmal den TFT.

DNJ1980, 2020-10-08:
Also die Verbindung zum Drucker ist hergestellt alles funktioniert auch soweit nur ich kann nichts drücken um ins Menü zu kommen ich kann auch per PC drucken und die Extruder Temperatur ändern das wird mir dann auch auf dem Bildschirm angezeigt und es steigt dann auch nur nicht über den Touchscreen nicht. Ich Versuche das mit der Textdatei mal.was muss ich alles auf die Karte drauf machen damit er flasht?

oldman4U, 2020-10-08:
Nur die Textdatei. Danach sollte das Kalibrierungstool kommen.

DNJ1980, 2020-10-08:
Hallo. Ich habe das Reset gemacht und es Funktioniert jetzt alles. Vielen Dank für deine Hilfe:)

oldman4U, 2020-10-08:
Gerne!

Bitte schliesse dieses Ticket.

Danke

closed 2020-10-08 by DNJ1980