bigtreetech/BIGTREETECH-TouchScreenFirmware #1160: [FR] Add the ability to rotate screen by 90 as well not only 180

Atanasovgoran, 2020-10-08:
I am making a "pocket" sized printer, and I am using an SKR Mini board and TFT24 V1.1 for the electronics.
And it makes more sense in the design to put the display in vertical rather than horizontal (90 degrees rotation)

Or is there a way to make it already, but I guess it would be somewhat complicated since menu items would need to be repositioned.

Thanks

oldman4U, 2020-10-08:
Hi.

Please read the other Feature Requests related to Vertical screen orientation and close the ticket in case you believe this is already covered.

Please also keep in mind that this open source project only works because of the community and therefore also people like you and me working together to make this happen

Thank you

Atanasovgoran, 2020-10-08:
> Hi.
>
> Please read the other Feature Requests related to Vertical screen orientation and close the ticket in case you believe this is already covered.
>
> Please also keep in mind that this open source [...]

My bad, searched for 90 degree and not vertical. I see they teased it but so far no release.

Closing the ticket.

closed 2020-10-08 by Atanasovgoran

oldman4U, 2020-10-08:
Thank you and happy printing.