bigtreetech/BIGTREETECH-TouchScreenFirmware #1161: Configuration.ini dont make change to language and other function FTF35 E3 V3

islonina, 2020-10-08:
Today i flash this fork for MBL:

https://github.com/rebel1/BIGTREETECH-TouchScreenFirmware

The only problem is i cant make any changes to configuration.ini. I need to chang eto couple of function like load length filament and language but every time a make change on  FTF35 E3 V3. Did i miss something? Before my firmware works fine(official one fork) and configuration.ini also but doesn't change MBL function.

oldman4U, 2020-10-08:
Hi.

I believe you will have to contact rebel1 directly regarding this issue. We can help you in case you have a problem with the firmware from this side.

Thank you and please close this ticket in case you do not need it anymore.

Thank you

islonina, 2020-10-08:
Its a fork from official one what i can see "This branch is even with bigtreetech:master." Now even official one doesnt take change from configuration.ini

oldman4U, 2020-10-08:
Fork means that the base is from here, but he made some changes. So if you use software from rebel he will take over support, if you use software from here, we will try to help.

You read the manual from the beginning to the end, especially the information about max SDcard size and format as well as the parts which have to be on the TFT: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

Send us pictures which show the problem.

islonina, 2020-10-08:
Ok. Finally I make changes to configuration.h on firmware myself and everything works fine.

oldman4U, 2020-10-09:
Perfect. 

Please help the community and close the ticket once you do not need it anymore. Thank you

closed 2020-10-09 by bigtreetech