bigtreetech/BIGTREETECH-TouchScreenFirmware #1162: TMC 2208 UART FYSTEC

IIIxpHunterIII, 2020-10-08:
The question I've got is how do you enable UART on said driver in Marlin?

It reports no UART capable drivers on M119 even though i Put the Jumpers in like BIGTREETECH sais.
[Marlin_Configuration.h.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5351021/Marlin_Configuration.h.txt)

Patform:
SKR1.4
FYSTEC 2208 V1.2

traffic-light, 2020-10-08:
You know that this has nothing to do with the tft firmware?

radek8, 2020-10-08:
you need to measure the jumper correctly on both the SKR and the TMC2208 controller

radek8, 2020-10-08:
[image]

[image]

IIIxpHunterIII, 2020-10-08:
Is the solderbridge necessary for the V1.2?

[image]

radek8, 2020-10-08:
If you want to run in UART mode, yes
You don't have an F6 board

IIIxpHunterIII, 2020-10-08:
Ahh okay, I was wondering about that one.

radek8, 2020-10-08:
what are you solving?
It is a small ball of tin or you wrap 2 pins with wire

radek8, 2020-10-08:
https://www.makenprint.uk/3d-printing/3d-printing-guides/skr-v1-3-setup-guide/

In order to setup UART mode on the TMC2208 drivers themselves we need to enable the feature. It requires a light bit of soldering of two contacts as circled above and is a quick job. Time your time there is no rush.

If the TMC2208 driver you are using are version 3 from BIGTREE-TECH then skip the soldering part it is already done for you.

IIIxpHunterIII, 2020-10-09:
Ive done that yesterday after you told me to, tmc debugging (m119 if i remember correctly) says it works. Te only remaining prob would be that m119 only gives an ok but no other info such as vrev

zonalimitatore, 2020-10-09:
M119?

IIIxpHunterIII, 2020-10-09:
It was 122, I was mistaken. Could you maybe explain those terms, i couldn't find good explainations on them? And another question would be (I know, wrong thread) how do I configure dual Z endstops, it says I've gotten something wrong:

 #error "Z2_USE_ENDSTOP has been assigned to a nonexistent endstop!"

//
// For Z set the number of stepper drivers
//
#define NUM_Z_STEPPER_DRIVERS 2   // (1-4) Z options change based on how many

#if NUM_Z_STEPPER_DRIVERS > 1
  #define Z_MULTI_ENDSTOPS
  #if ENABLED(Z_MULTI_ENDSTOPS)
    #define Z2_USE_ENDSTOP          _XMAX_
    #define Z2_ENDSTOP_ADJUSTMENT   0
    #if NUM_Z_STEPPER_DRIVERS >= 3
      #define Z3_USE_ENDSTOP        _YMAX_
      #define Z3_ENDSTOP_ADJUSTMENT 0
    #endif
    #if NUM_Z_STEPPER_DRIVERS >= 4
      #define Z4_USE_ENDSTOP        _ZMAX_
      #define Z4_ENDSTOP_ADJUSTMENT 0
    #endif
  #endif
#endif

//
// Limit Switches
//
#ifdef X_STALL_SENSITIVITY
  #define X_STOP_PIN                  X_DIAG_PIN
  #if X_HOME_DIR < 0
    #define X_MAX_PIN                      P1_26  // E0DET
  #else
    #define X_MIN_PIN                      P1_26  // E0DET
  #endif
#else
  #define X_STOP_PIN                       P1_29  // X-STOP
#endif

#ifdef Y_STALL_SENSITIVITY
  #define Y_STOP_PIN                  Y_DIAG_PIN
  #if Y_HOME_DIR < 0
    #define Y_MAX_PIN                      P1_25  // E1DET
  #else
    #define Y_MIN_PIN                      P1_25  // E1DET
  #endif
#else
  #define Y_STOP_PIN                       P1_28  // Y-STOP
#endif

#ifdef Z_STALL_SENSITIVITY
  #define Z_STOP_PIN                  Z_DIAG_PIN
  #if Z_HOME_DIR < 0
    #define Z_MAX_PIN                      P1_00  // PWRDET
  #else
    #define Z_MIN_PIN                      P1_00  // PWRDET
  #endif
#else
  #ifndef Z_STOP_PIN
    #define Z_STOP_PIN                     P1_27  // Z-STOP
  #endif
#endif

IIIxpHunterIII, 2020-10-09:
Oh and that solved my question from befor 👍 
"TMC_DEBUG adds more reported information."

oldman4U, 2020-10-09:
What is the question? Dual z?

It is a function of the board. For Marlin you have to configure it like you would have a single z axis.

IIIxpHunterIII, 2020-10-09:
It doesnt compile bacause of this: #error "Z2_USE_ENDSTOP has been assigned to a nonexistent endstop!"

IIIxpHunterIII, 2020-10-09:
I have configured dual Z endstops, because...well problems.

//
// For Z set the number of stepper drivers
//
#define NUM_Z_STEPPER_DRIVERS 2 // (1-4) Z options change based on how many

#if NUM_Z_STEPPER_DRIVERS > 1
#define Z_MULTI_ENDSTOPS
#if ENABLED(Z_MULTI_ENDSTOPS)
#define Z2_USE_ENDSTOP XMAX
#define Z2_ENDSTOP_ADJUSTMENT 0
#if NUM_Z_STEPPER_DRIVERS >= 3
#define Z3_USE_ENDSTOP YMAX
#define Z3_ENDSTOP_ADJUSTMENT 0
#endif
#if NUM_Z_STEPPER_DRIVERS >= 4
#define Z4_USE_ENDSTOP ZMAX
#define Z4_ENDSTOP_ADJUSTMENT 0
#endif
#endif
#endif

IIIxpHunterIII, 2020-10-09:
> What is the question? Dual z?
> 
> It is a function of the board. For Marlin you have to configure it like you would have a single z axis.

And if i specify the Pins like this:
// Limit Switches
//
#define X_MIN_PIN P1_29
#define X_MAX_PIN P1_26
#define Y_MIN_PIN P1_28
#define Y_MAX_PIN P1_25
#define Z_MIN_PIN P1_27
#define Z_MAX_PIN P1_00

It rips the Z1 connector apart -> eg no Z1 Stop

oldman4U, 2020-10-09:
This exceeds what I know about Marlin, sorry. The machine I have which uses dual Z has TMC2209 and sensorless homing, so no physical endstops at all on min, and software endstops for max. Never configured dual motor drivers or endstops before.

My advise. Start with a simple setup and add features and functions later on. To start from zero to everything, incl. power control and filament sensor and LED and..... will not work and will leave you back frustrated.

My 2 cents

IIIxpHunterIII, 2020-10-09:
Thats a pity, but I wont start from the beginning... it took me two days to configure it since it's my first marlin build 😅 I'll figure it out somehow I guess

oldman4U, 2020-10-09:
Sounds like a big adventure to me. 👍🏻

radek8, 2020-10-09:
I also have two trapezoidal screws on the Z axis.
But I find the use of two engines unnecessarily complicated, so the second screw I have transferred to a pulley with a belt.
So I have no experience with your configuration

oldman4U, 2020-10-10:
Hi.

Please close the ticket once you do not need it anymore. The problem related to the TFT seems to be solved;-)

Thank you and happy printing

IIIxpHunterIII, 2020-10-10:
> I also have two trapezoidal screws on the Z axis.
> But I find the use of two engines unnecessarily complicated, so the second screw I have transferred to a pulley with a belt.
> So I have no experience with your configuration

I cannot do that, since the I3 has no space for that :(

closed 2020-10-10 by IIIxpHunterIII