bigtreetech/BIGTREETECH-TouchScreenFirmware #1163: BTT 2.8TFT Controller Panel firmware update?

JAdams12, 2020-10-08:
Hey everyone!

I have a [couple of these touch screens](http://www.bigtree-tech.com/shop/product/show/id/29) I got a while back that I'd like to install, but I hate the UI that comes with them and would rather use the new Unified one. 

I copied the config, TFT28 folder, and .bin file to an SD card, put it in the screen, and... nothing. It just starts up like normal. 

Anyone know what could be going wrong?

oldman4U, 2020-10-08:
Hi.

Please read the manual on the main page until the end: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

Please let us know

JAdams12, 2020-10-08:
Thank you for the reply @oldman4U but it looks like the issue was with the SD card I was using. I switched to another one and now the update is installing. 

Thank you for the help!

closed 2020-10-08 by JAdams12

oldman4U, 2020-10-08:
You are very welcome

Tipp, Search for "Round"  here in this ISSUE repository for a VERY NICE UI