bigtreetech/BIGTREETECH-TouchScreenFirmware #1164: [Q] I got error message notification

islonina, 2020-10-08:
I got 3 notification error:

-material hetup parameters
-PID settings
-filament load/unload lengths

Can't find them on menu

oldman4U, 2020-10-09:
Please help the community and close the other ticket. 

Thank you

oldman4U, 2020-10-09:
Thank you.

When did you get the notification after doing what?

Maybe a picture could also help.

Thank you

islonina, 2020-10-09:
I get this on power on printer every time. When I enter notification many on the main panel.[image]. When I push 117 ABL notification [image]

oldman4U, 2020-10-09:
It is a new feature. Previous version only showed the last Echo message.
The new version still shows one Echo message but in addition you can click
on the message „window“ or the upper bar to see the 3 last messages.

No bug but a feature.

;-)

islonina, 2020-10-09:
But I can't find where to I need to enter this value. Length and pod. I know that pod I probably pid value after auto tune but length load and unload? I have set this on firmware for marlin lcd

oldman4U, 2020-10-09:
These are information and not errors. Like I wrote before, with the older version you would not have seen the, even they always exist. Now you see them.

islonina, 2020-10-09:
I know it's information but I assume after : they should be a value like on fade high:5

oldman4U, 2020-10-09:
Yes, there should be everything shown related to a message, but I guess there is simply not enough space to show it. Those messages are usually very short and sometimes useful;-)

islonina, 2020-10-09:
Ok. Now i have problem. ftf doesnt save mesh value. When i run MBL on marlin mode everytking is saved when i run M503 i see all point value. When i run MBL on TFT the value doesnt exist even after manual M500 commend on TFT

oldman4U, 2020-10-09:
Close this ticket in case you do not need it anymore. Check existing open and closed tickets and make a new ticket if needed.

closed 2020-10-09 by islonina