bigtreetech/BIGTREETECH-TouchScreenFirmware #1165: [FR] Smart Filament Sensor Module with Repetier-Firmware V2

GenaUser, 2020-10-09:
Would it be possible to add sensor management for the Repetier-firmware V2?
https://github.com/repetier/Repetier-Firmware/tree/dev2

Information sent by the firmware:
GCodeSource :: printAllFLN (PSTR ("// action: out_of_filament T"), (int32_t) tool-> getToolId ());
GCodeSource :: printAllFLN (PSTR ("RequestPause: No filament detected!"));

It would also allow the use of smart filament sensors from multiple extruders connected to the board.
Thank you in advance to analyze this possibility.

traffic-light, 2020-10-09:
This screen is marlin, and not reptier, firmware based.

So I don't think it will be imported. To do that the complete firmware of the tft would need to be rebuild...

GenaUser, 2020-10-09:
@traffic-light 
Since their creation, these screens are perfectly compatible with Repetier-Firmware V1 or V2, whether in "Touch mode" or "Marlin mode".
My request concerns only the additional interpretation of the Gcode sent when an absence of the filament is detected.

traffic-light, 2020-10-09:
I'm sorry. My fault.

oldman4U, 2020-10-20:
Hello GenaUser.

Could you please be so kind and close this ticket. Your feature request is covered in the pinned Feature Request ticket #1177 

Thank you

closed 2020-10-20 by GenaUser