bigtreetech/BIGTREETECH-TouchScreenFirmware #1166: [FR] Pre heat beep/tune/tone to acknoledge the set temperatures have been reached

CBDesignS, 2020-10-09:
Is it possible to add a beep tone or tones to give an audible prompt once a pre-heat bed & extruder temperature has been reached ??

I have the slowest heating hot bed on the planet and it can some days take up to 7 mins to get up to 100c then heat the extruder to 235c or there abouts depending on fillament type can be 8 + mins. To help reduce the waiting times I use a pre heat option that heats the bed to 100 and extruder to 160 and this can slice a couple of min off the waiting time. I have pid tuned the bed many times but its still slow as heck. (130W 24v Silicone heatpad on 4mm Aluminium bed with 3mm Glass on top of it)

oldman4U, 2020-10-09:
Turn off PID for the bed. It is useless.

traffic-light, 2020-10-10:
> Turn off PID for the bed. It is useless.

I don't agree (completely) 😉 
When configured correctly its heating up pretty fast until it's almost the temperature you want.

After that it keeps it stable without heating it a lot.
(It also saves some power usage)

oldman4U, 2020-10-10:
Turning PID off for the bed will eventually heat it faster. Because of the
higher mass compared to the nozzle keeping the temp stable is also not a
problem. So what is it good for?

um 11:18:

oldman4U, 2020-10-10:
Btw. I thought you have to learn...😁

CBDesignS, 2020-10-10:
I have a build with bit bang for the bed and a build "configured" with pid and once I get the X Y gantry rebuilt and squared up I will test both and see whats the best. From memory the difference was not worth changing back to the other heating variant. a few seconds quicker on bit bang but top end temp waivered a little more on a 5 hour print. testing soon so will draw my conclusion once done. I just wish I knew how to code but alas basic python is about as far as I can go and if I here the word tuple I am out of there faster than a rat up a drain pipe ..  I suppose I need to hope somone with a greater knoledge than me can do it.

oldman4U, 2020-10-11:
The "problem" with such feature request is not that they are not worth, but that there are so many users waiting to get real bugs fixed. But even for this there is no developer available.

traffic-light, 2020-10-11:
> Turning PID off for the bed will eventually heat it faster. Because of the higher mass compared to the nozzle keeping the temp stable is also not a problem. So what is it good for? Peter van Weeterloo <notifications@github.com> schrieb am Sa. 10. Okt. 2020 um 11:18:
> […](#)

good explenation here: https://www.reddit.com/r/3Dprinting/comments/9xg2jm/heatbed_pid_vs_bangbang_which_one_is_easier_on/

and yes. ive got to learn. but not all the time ;-)

oldman4U, 2020-10-20:
Hi @CBDesignS.

Could you please be so kind and close this ticket. Your feature request is covered in the pinned Feature Request ticket #1177 
Thank you

CBDesignS, 2020-10-21:
Closed because the Feature Request has been added to the list.

closed 2020-10-21 by CBDesignS

traffic-light, 2020-11-14:
I've made this in my pr ( #1209) without knowing that there was a fr for it.

Will be on git with the next push

PR #1561 (2021-01-30): Beep when temp reached
### Description

This PR adds an acoustic acknowledge that the desired temperature is reached, given it to be the nozzle, the bed or the chamber.
It is working only if the temperature is set on the TFT by hand or automatically (Load/Unload, Extrude. Preheat, Heat, etc). It's not active if a gCode sets the temperature. It's not active if you set the temperature on the terminal or by any external command.
~It is falling into the rule of "alert_sound" in the Config.ini.~
It makes distinctive sounds, depending on the temperature if it has fallen or risen to the desired one.
 [... cut]