bigtreetech/BIGTREETECH-TouchScreenFirmware #1167: [BUG] MBL problem saving grid on TFT mode

islonina, 2020-10-09:
No possible way to save correct mesh point

### Steps to reproduce

Make a MBL from FTF screen- 9 point 

**Expected behavior**
Should save all point after allowing so save on EEPROM i TFT screen

**Actual behavior**
No mesh leveling point. Check with M503 only M420 S0 with missing grid point

### Hardware Variant
1. Marlin-2.0.x
2. TFT35 E3 V3.0 
### TFT Firmware Version & Main Board Firmware details
Creality Super Silent board 1.1.15 and Marlin 2.0.X
BIGTREE_TFT35_V3.0_E3.26.x.bin wit MBL enabled latest one from github

Echo from prontoface TFT mode after M503:
`SENDING:M503
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Steps per unit:
echo: M92 X80.00 Y80.00 Z400.00 E412.70
echo:Maximum feedrates (units/s):
echo:  M203 X200.00 Y200.00 Z12.00 E120.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X1000.00 Y1000.00 Z1000.00 E5000.00
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P500.00 R1250.00 T1250.00
echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
echo:  M205 B20000.00 S0.00 T0.00 J0.08
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Mesh Bed Leveling:
echo:  M420 S0 Z5.00
echo:Material heatup parameters:
echo:  M145 S0 H185 B45 F255
echo:  M145 S1 H240 B0 F255
echo:PID settings:
echo:  M301 P28.65 I2.75 D74.51
echo:  M304 P150.06 I29.55 D508.11
echo:Linear Advance:
echo:  M900 K1.40
echo:Filament load/unload lengths:
echo:  M603 L535.00 U490.00`

Echo from prontoface Marlin mode M503:
`SENDING:M503
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Steps per unit:
echo: M92 X80.00 Y80.00 Z400.00 E412.70
echo:Maximum feedrates (units/s):
echo:  M203 X200.00 Y200.00 Z12.00 E120.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X1000.00 Y1000.00 Z1000.00 E5000.00
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P500.00 R1250.00 T1250.00
echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
echo:  M205 B20000.00 S0.00 T0.00 J0.08
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Mesh Bed Leveling:
echo:  M420 S0 Z5.00
echo:  G29 S3 X1 Y1 Z-0.42500
echo:  G29 S3 X2 Y1 Z-0.25000
echo:  G29 S3 X3 Y1 Z0.02500
echo:  G29 S3 X1 Y2 Z-0.40000
echo:  G29 S3 X2 Y2 Z-0.07500
echo:  G29 S3 X3 Y2 Z0.32500
echo:  G29 S3 X1 Y3 Z-0.15000
echo:  G29 S3 X2 Y3 Z0.30000
echo:  G29 S3 X3 Y3 Z0.72500`

radek8, 2020-10-11:
Please send command M503 and M115 via the pronterface program
Save the printer's response here.
Thank you

digant73, 2020-10-11:
that problem will be fixed by PR #1154

islonina, 2020-10-11:
SENDING:M115
FIRMWARE_NAME:Marlin 2.0.7 (Oct 10 2020 00:51:44) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:SERIAL_XON_XOFF:0
Cap:BINARY_FILE_TRANSFER:0
Cap:EEPROM:1
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:0
Cap:RUNOUT:0
Cap:Z_PROBE:0
Cap:LEVELING_DATA:1
Cap:BUILD_PERCENT:1
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:1
Cap:PROMPT_SUPPORT:0
Cap:SDCARD:0
Cap:AUTOREPORT_SD_STATUS:0
Cap:LONG_FILENAME:0
Cap:THERMAL_PROTECTION:1
Cap:MOTION_MODES:0
Cap:ARCS:0
Cap:BABYSTEPPING:1
Cap:CHAMBER_TEMPERATURE:0

SENDING:M503
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius
echo:; Filament settings: Disabled
echo:  M200 S0 D1.75
echo:; Steps per unit:
echo: M92 X80.00 Y80.00 Z400.00 E93.00
echo:; Maximum feedrates (units/s):
echo:  M203 X200.00 Y200.00 Z12.00 E120.00
echo:; Maximum Acceleration (units/s2):
echo:  M201 X1000.00 Y1000.00 Z1000.00 E5000.00
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P1250.00 R1250.00 T1250.00
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
echo:  M205 B20000.00 S0.00 T0.00 J0.08
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; Mesh Bed Leveling:
echo:  M420 S0 Z3.00
echo:  G29 S3 I0 J0 Z-0.20000
echo:  G29 S3 I1 J0 Z0.20000
echo:  G29 S3 I2 J0 Z0.60000
echo:  G29 S3 I0 J1 Z-0.45000
echo:  G29 S3 I1 J1 Z0.02500
echo:  G29 S3 I2 J1 Z0.42500
echo:  G29 S3 I0 J2 Z-0.67500
echo:  G29 S3 I1 J2 Z0.07500
echo:  G29 S3 I2 J2 Z0.57500
echo:  G29 S4 Z0.00000
echo:; Material heatup parameters:
echo:  M145 S0 H185 B45 F255
echo:  M145 S1 H240 B70 F255
echo:; PID settings:
echo:  M301 P21.73 I1.54 D76.55
echo:; Filament load/unload lengths:
echo:  M603 L0.00 U0.00

islonina, 2020-10-11:
> that problem will be fixed by PR #1154

Fork from this issue have a problem compiling
{
	"resource": "/c:/Users/islon/Downloads/BIGTREETECH-TouchScreenFirmware-bugfix-2020-10-05/TFT/src/User/Configuration.h",
	"owner": "C/C++",
	"code": "1696",
	"severity": 8,
	"message": "#include errors detected. Please update your includePath. Squiggles are disabled for this translation unit (C:\\Users\\islon\\Downloads\\BIGTREETECH-TouchScreenFirmware-bugfix-2020-10-05\\TFT\\src\\User\\delay.c).",
	"source": "C/C++",
	"startLineNumber": 1,
	"startColumn": 1,
	"endLineNumber": 1,
	"endColumn": 26
}

radek8, 2020-10-11:
the compilation will be removed after merging PR #1138

islonina, 2020-10-12:
> the compilation will be removed after merging PR #1138

Just now i upload https://github.com/guruathwal/BIGTREETECH-TouchScreenFirmware

MBL works and show me that saved mesh to eeprom but stil on M503 no points

digant73, 2020-10-12:
the MBL points are reported in M503!

echo: G29 S3 I0 J0 Z-0.20000
echo: G29 S3 I1 J0 Z0.20000
echo: G29 S3 I2 J0 Z0.60000
echo: G29 S3 I0 J1 Z-0.45000
echo: G29 S3 I1 J1 Z0.02500
echo: G29 S3 I2 J1 Z0.42500
echo: G29 S3 I0 J2 Z-0.67500
echo: G29 S3 I1 J2 Z0.07500
echo: G29 S3 I2 J2 Z0.57500

If you want to see the grig type M420 T1 V1

islonina, 2020-10-12:
SENDING:M503
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius
echo:; Filament settings: Disabled
echo:  M200 S0 D1.75
echo:; Steps per unit:
echo: M92 X80.00 Y80.00 Z400.00 E93.00
echo:; Maximum feedrates (units/s):
echo:  M203 X200.00 Y200.00 Z12.00 E120.00
echo:; Maximum Acceleration (units/s2):
echo:  M201 X1000.00 Y1000.00 Z1000.00 E5000.00
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P1250.00 R1250.00 T1250.00
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
echo:  M205 B20000.00 S0.00 T0.00 J0.08
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; Mesh Bed Leveling:
echo:  M420 S0 Z3.00
echo:; Material heatup parameters:
echo:  M145 S0 H185 B45 F255
echo:  M145 S1 H240 B70 F255
echo:; PID settings:
echo:  M301 P21.73 I1.54 D76.55
echo:; Filament load/unload lengths:
echo:  M603 L0.00 U0.00
>>>M420 T1 V1
SENDING:M420 T1 V1
echo:Invalid mesh.

digant73, 2020-10-12:
ok, that bug is fixed by #1154. That PR will be merged on top of other PR solving the problem with FLASH space you got trying to  compile #1154

islonina, 2020-10-12:
OK. I will  be waiting till then

radek8, 2020-10-12:
The M115 command tells you
Cap: AUTOLEVEL: 0

islonina, 2020-10-12:
> The M115 command tells you
> Cap: AUTOLEVEL: 0

Yes after TFT MBL, when i do MBL on marlin mode 115 give me autolevel 1

islonina, 2020-10-12:
And some problem on printing 
echo:E0 Flow: 100%
echo:Unknown command: "ţM105"

radek8, 2020-10-12:
Looks like you're having trouble communicating on a serial line.
What communication speed are you set?
Do not use the supplied cable to connect the display to the Marlin board via RS232, but use a shielded cable.

islonina, 2020-10-12:
I use shielded one and 115200 baudrate

radek8, 2020-10-13:
Please remove the listing of the failed compilation.
The ticket is then confusing.
The problem was in the size of the compilation

> c: /users/islon/.platformio/packages/toolchain-gccarmnoneeabi/bin /../ lib / gcc /arm-none-eabi/7.2.1/../../../../arm-none->eabi/bin/ld.exe: oblast `FLASH 'přetekla 5676 bajtů

To eliminate the problem, a PR has already been developed, which is waiting to be merged

radek8, 2020-10-13:
> Používám stíněný a 115200 baudrate

Do you have a quality and sufficiently dimensioned power supply?
115200 and the shielded cable should not show interference.

islonina, 2020-10-14:
> > Používám stíněný a 115200 baudrate
> 
> Do you have a quality and sufficiently dimensioned power supply?
> 115200 and the shielded cable should not show interference.

Yes, already tried on stock ender one, poweredge server one and couple more. Never have a problem with them before. All shielded wire and grounded frame

oldman4U, 2020-10-20:
islonina. Could you please test #1188 to see if it solves your reported issue.

Thank you

islonina, 2020-10-20:
Sure no problem

oldman4U, 2020-10-21:
👍🏻

islonina, 2020-10-21:
Stuck on config screen
[image]

digant73, 2020-10-22:
@islonina It seems a problem when the settings are stored in the FLASH.
Can you try to comment out line 65 in config.c? This should avoid the freeze although the settings won't be stored in the flash

That means comment the following line in config.c:
//    storePara();

Please let me know the result.
Also what TFT model do you have?

Is there someone else having the same problem?

oldman4U, 2020-10-22:
It is a TFT35 E3 v3 and I have the same TFT. Should I test against #1188?

islonina, 2020-10-22:
> It is a TFT35 E3 v3 and I have the same TFT. Should I test against #1188?

Yest tft E3 and its from 1188

digant73, 2020-10-22:
> It is a TFT35 E3 v3 and I have the same TFT. Should I test against #1188?

yes please. I cannot test it by myself now. try also the scenario with PARA_SIZE set to 1024 (it is now set to 512).

islonina, 2020-10-22:
> PARA_SIZE

Ok after commenting 65 line no freeze all load to main screen. Bu building with uncomment 65 line and modify flashStorage.h from 512 to 1024 cant compile. I got Error:

Compiling .pio\build\BIGTREE_TFT35_E3_V3_0\src\src\User\API\flashStore.o
Compiling .pio\build\BIGTREE_TFT35_E3_V3_0\src\src\User\API\interfaceCmd.o
Compiling .pio\build\BIGTREE_TFT35_E3_V3_0\src\src\User\API\ledcolor.o
Compiling .pio\build\BIGTREE_TFT35_E3_V3_0\src\src\User\API\menu.o
TFT\src\User\API\config.c: In function 'getConfigFromFile':
TFT\src\User\API\config.c:55:5: error: unknown type name 'store'; did you mean 'short'?
     store custom codes count
     ^~~~~
     short
TFT\src\User\API\config.c:55:18: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'codes'
     store custom codes count
                  ^~~~~
Compiling .pio\build\BIGTREE_TFT35_E3_V3_0\src\src\User\API\parseACK.o
*** [.pio\build\BIGTREE_TFT35_E3_V3_0\src\src\User\API\config.o] Error 1
================================================================================================== [FAILED] Took 7.49 seconds ==================================================================================================

Environment            Status    Duration
---------------------  --------  ------------
BIGTREE_TFT35_E3_V3_0  FAILED    00:00:07.494
============================================================================================= 1 failed, 0 succeeded in 00:00:07.494 =============================================================================================The terminal process "C:\Users\islon\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

digant73, 2020-10-22:
@islonina that is a problem in your envirnment. I compiled it with no problem but I cannot test it due I'm on office now.
However, pretty musch sure the problem is due to a buffer overflow on "u8 data[PARA_SIZE];" used on storePara function.
With the new parameters added for the Mesh Editor the required buffer size should be 516 at least (against the current 512)

oldman4U, 2020-10-22:
Hi.

I saw the same issue and can confirm that changing the site to 1024 solves the problem.Are you sure this fix will work on all TFTs, also the TFT24 and TFT28? 28 I can test, but not 24 if this would help.

digant73, 2020-10-22:
Already asked to the creator of flashStore.c (guruatwhal) if he had some reasons to limit to 512 byte. I think it was the initial limit when the config.ini was introduced. A lot of parameters were added in the meanwhile. It should be possible to increase that limit (e.g. 768)

islonina, 2020-10-22:
Without this modification 1024 I can build and flash firmware without a problem. After I change that value I get this error. Every other brunch works also fine.

digant73, 2020-10-22:
@islonina ok. Try to reduce the value (e.g. 768) or so on. However the value must be at least 516.

PR #1188 (2020-10-28): Mesh Editor + improvements + bugfix (replace PR #1154)
**NEW FEATURES:**
* resolves #288
* resolves #594
* resolves #777
* Implemented #288 and #777 adding a Mesh Editor (under BL menu) and Mesh Tuner menus providing a visual mesh grid (gradient colors are configurable in config.ini) and a fine tune tool for mesh editing. The Mesh Editor supports:
1) MBL
2) UBL
3) ABL Bilinear

**BUG FIXES:**
* Fixed #1167 MBL process causing "Invalid mesh" error trying to load the meshes with M420 S1. MBL process is now based on use of standard move gcode
* Fixed #967 related to wrong Software Endstops enabli [... cut]
closed 2020-10-28 by bigtreetech