bigtreetech/BIGTREETECH-TouchScreenFirmware #1168: [BUG] Extrude menu problems

GenaUser, 2020-10-11:
I am using the TFT35 V3 with the latest firmware version (compiled by myself). 
My printer has 2 extruders.
In the "Extrude" menu when we click on the "nozzle" icon, the selection does not return to "E0" after "E1", but continues to scroll through unrecognized characters until the system freeze.

[image]

radek8, 2020-10-11:
Did you also upload the config.ini file while uploading the new FW?

GenaUser, 2020-10-12:
The config.ini is downloaded with the firmware.

Each click on the "Nozzle" icon scrolls through the values ​​indicated in Configuration.h:
#define TOOL_CHANGE {"T0", "T1", "T2", "T3", "T4", "T5"}
#define EXTRUDER_ID {"E0", "E1", "E2", "E3", "E4", "E5"}
It starts with “E0” through “E5”, then “T0” through “T5” and then unrecognized characters.
The selection is not limited to the value indicated in config.ini
//Number of extruder
//Options: [1 to 6]
ext_count: 2

or possibly in Configuration.h:
#define HOTEND_NUM 2 // set in 1 ~ 6
#define EXTRUDER_NUM 2 // set in 1 ~ 6

radek8, 2020-10-12:
Can you save here the config.ini you used?

GenaUser, 2020-10-13:
You can find the config.ini here:
https://1drv.ms/u/s!Ak8QYQq1MpUzmHKN82XytAsex7mz?e=uRIH65

radek8, 2020-10-13:
I couldn't simulate your problem.
I have one hotend with two extruders. The extruder menu switches between E0 and E1.

GenaUser, 2020-10-14:
Problem solved.

The bug does not appear in older versions of the firmware.
In more recent ones, the "Extruder" menu does not read the values ​​displayed by the firmware of the screen to determine the number of extruders, but it queries the printer using the M115 function.
The information communicated by M115 did not indicate the number of extruders, but the name of its variable.
The error was that I had only enabled the tool variable but not the extruder variable in the printer firmware.

Thank you for your help.

closed 2020-10-14 by GenaUser