bigtreetech/BIGTREETECH-TouchScreenFirmware #1169: [BUG] Menu "Parameter Settings" problems

GenaUser, 2020-10-11:
I am using the TFT35 V3 display with the latest firmware version (compiled by myself). 
My printer has 2 extruders.
The " offset 2nd Nozzle" menu does not store the values ​​entered.
As soon as you exit the menu the values ​​are reset to "0", even if you choose "Save settings" immediately afterwards.

[image]

radek8, 2020-10-11:
Please send command M503 and M115 via the pronterface program
Save the printer's response here.
Thank you

GenaUser, 2020-10-12:
The problem indicated is not related to the firmware of the printer board, but to that of the screen.
The parameters of "Offset 2nd Nozzle" are not saved in the screen memory.
All other parameters in the "Parameter settings" menu are saved correctly, even if the display is not connected to a printer board.

radek8, 2020-10-12:
The display does not store these values on the TFT.
It reads them from Marlin and, after changing the values, sends them back to marlin.
The problem will be in the Marlin configuration.
Can you please save the required data M503 and M115?
Thank you

radek8, 2020-10-12:
The display just shows what Marlin is sending

radek8, 2020-10-12:
> All other parameters in the "Parameter settings" menu are saved correctly, even if the display is not connected to a printer board.

If the TFT is not connected to the Marlin board, no printer parameters can be changed ...

GenaUser, 2020-10-13:
With my screen not connected, the parameters modified in the submenus of the “Parameter parameters” menu remain saved on the screen until it is restarted. Of course, he cannot send them without connection.
This is not the case with the values ​​in the "2nd nozzle offset" menu. 
If I leave the menu and come back, they are no longer present, whether the screen is connected or not.
This is not very annoying, so I will close this topic.
Thank you for your help

closed 2020-10-13 by GenaUser

radek8, 2020-10-13:
Why are you closing the ticket? Why don't you send me the required data?
I added this item to the menu and if there's a problem with that, I'd like to fix it.

radek8, 2020-10-13:
Also, your second issue #1168 points to some non-standard Marlin settings