bigtreetech/BIGTREETECH-TouchScreenFirmware #1170: [FR] Draft FR List

oldman4U, 2020-10-11:
@traffic-light 

Hi Peter. What do you think to make this a companion to your ReadMe ticket.

(FR) Read me before posting a Feature Request!

Dear reader,

This Feature Requests list has two "functions":

1: It allows users to check, if a certain feature has already been requested, before starting a new ticket.

2: To mark features which are already under development. This could help to avoid that several developers are working on the same function.

General Features:
#290 Add a character generator in order to use klipper
#524 Vertical Screen GUI
**#634 The TFT display does not support the progress LED**
#804 Transfer a file via Wifi ESP3D to the TFT-SD and/or TFT USB-Stick 

Touchscreen Related Feature:
**#288 Ability to edit bed-leveling mesh grid in touch mode**
#296 Idle and back to main menu
#601 Invert filelist
**#707 (feature request title) Show Chamber Temp.**
**#732 Simplified Status Screen while Printing**
#763 Recover possibility to have the custom menu with icons
#810 Octoprint Print Status Screen
#834 Filament change buttons
**#865 Print finished dialog and jump back to Info Screen on TFT35**
**#1026 Alternate elapsed/remaining time during printing**
**#1046 Manual way to enter temperature in touchscreen mode NOTE- WHY LIMITING TO TEMPERATURE**
#1053 Option to deactivate the TFT LCD menu item
#1102 standard deviation on repeatability test
#1112 Ability to Scroll GCode Command Display Window
#1121 Runout Sensor On Board
#1137 Add ability to get to home screen with 1 click from any screen please
#1150 Include G34 Auto-Alignment to Movement/BedLevel UI
#1166 Pre heat beep/tune/tone to acknowledge the set temperatures have been reached

Marlin Screen Related Features:
#301 Colour Mix menu
#594 Add button for Marlin Manual Mesh Bed Leveling 

Additional features and functions:
#358 use #pragma once
#361 remote firmware deploy with build.py
#365 move language.c into language.ini
#598 Implement "LittlevGL"
#1115 Add option for temp sensors
#1165 Smart Filament Sensor Module with Repetier-Firmware V2

Bug, Feature or Known Issue?:
#362 printing menu and echo
#438 move language.c into language.ini
#494 Can M117 commands issued from Octoprint make it to the status screen in touch mode?
#571 Firmware assumes heated bed
#574 Optional' Enhancements (for CNC) Firmware assumes Tool, Extruder, Fan
#597 ESP32 CAM Compatibility
#958 WIFI REPETIER support
#1050 How to display M117 messages in touch mode info area on TFT35 E3 V3.0

Feature Requests shown in **Bold** are under development and if possible, we will also write the responsible developer next to it.
New feature requests will be added as fast as possible.
Notes from the community are added to each FR in CAPITAL LETTERS.

traffic-light, 2020-10-11:
I think it's  a really good idea ;-)

You could keep it as a separate thing i think?

oldman4U, 2020-10-11:
I hope BTT would add it to the pinned tickets.

Then we need a known issue.

Because yours is more a Frequently asked questions and how the report an
issue right documentation.

um 19:39:

oldman4U, 2020-10-12:
> I think it's a really good idea ;-)
> 
> You could keep it as a separate thing i think?
> 
> And maybe the possibility for everyone to tell what they are building? Don't think that is really needed because you set everything to bold. But it could be handy so that you don't need to change it every time when someone is doing a pr.

Hi Peter. Could you please delete your reply to make space for other replies, thank you.

oldman4U, 2020-10-13:
> I think it's  a really good idea ;-)
>
>
>
> You could keep it as a separate thing i think?
>
>
>
> And maybe the possibility for everyone to tell what they are building? Don't think that is really needed [...]

Hi Peter. Please delete your answer so I can release this ticket. 

Thank you

oldman4U, 2020-10-19:
Hi @traffic-light. Could you please remove your comments we start with an empty ticket. Thank you

@bigtreetech Please pin this ticket, so I can ask all users with FR's to close their tickets.

Kind regards

oldman4U, 2020-10-20:
@bigtreetech  Thank you BTT!!!

JeremyKennedy, 2020-10-21:
What you are trying to do here goes against best practices for managing issues. Issues should remain open if they are unresolved, so that people who are searching for something can see if it has been resolved or not. You seem to think that having a single issue to contain other issues is a good idea maybe as an overview, but really that is what the main GitHub list of issues is for, this is completely unnecessary and confusing.

You should not be asking people to close unresolved issues, it is going to cause a ton of confusion. Issues should be standalone, rather than having some crazy "master issue" as you've created here. If you want to categorize feature requests so it is easier to find them, apply a label using GitHub's built-in functionality for that. Then you can search the issues list for that label, exactly as designed and intended by GitHub and all other issue trackers.

oldman4U, 2020-10-21:
Hi Jeremy.

Thank you for taking the time to answer me. It seems that you know GitHub much better than me, so it would be great if you could convince BTT to add such a structure.

I am in the production and business industry since more than 25 years, half of the time as a product manager. For all our products there are documents like "Known Issues" and a "Knowledge Base" covering the most important topics at a single place. And most of the other products and developments I know have something similar.

It is also best practise, that someone starting a ticket is taking care of it and in case he is not interested anymore, he closes this ticket. In case it is unused for some time, usually the owner of the repo closes it, or a Bot. This is all not the case here. 

Many tickets are also opened, because the search function is not working well. There is a ticket #1158 which has "????" in its headline, search for it and check if you get a result. I get nothing.

Another reason is, that we had the situation that two engineers have been working on the same function "Mesh Bed Editor", without knowing from each other. My personal feeling is, that there are not too many engineers working on the software, so waisting time this way is far away from optimal.

Closing a ticket on page 2 or higher, and been listed in a pinned ticket instead is a good deal for me. The same time it makes the whole ticket system more useful. So I can see many benefits using this system for the price, that tickets which would never be found or which have not been used since months, have to be closed. Sounds like a good deal for me.

But lets see if we both agree on one thing. Look at this repository: https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues Do you think that unmanaged repositories are working good enough, so that actions like this are not needed. Once we agree that this is not the case and we find a better way to avoid this, I am willing to do what ever is needed to introduce this better system.

Again, I really appreciate your input.

Thank you

traffic-light, 2020-10-22:
I agree with @JeremyKennedy .

But you have to take in mind that a lot of users here doesn't know how Github works. So it's not really handy to have a lot of tickets open with FR in it.

This way you keep the issues clean and hopefully more organized.

_The biggest problem in this project is the lack of project management._

oldman4U, 2020-10-22:
I agree with him too. It is not the best solution but hopefully better than nothing. A few weeks ago we had around 250 open tickets and it was not usable anymore. Hope this helps to improve it further.

traffic-light, 2020-10-26:
There is a PR (not finished yet) for 634 The TFT display does not support the progress LED

oldman4U, 2020-10-26:
> There is a PR (not finished yet) for 634 The TFT display does not support the progress LED
> 
> 1209 ;-)

Updated. Sorry it took so long;-)

traffic-light, 2020-10-26:
@oldman4U  it was not that long ago that I made it :-P

Stellarspace, 2020-10-27:
> I agree with him too. It is not the best solution but hopefully better than nothing. A few weeks ago we had around 250 open tickets and it was not usable anymore. Hope this helps to improve it further.

Why won't @bigtreetech add a management bot or something?

blueeagle69, 2020-10-29:
Feel free to add my https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1135 to the list 😀

Stellarspace, 2020-10-29:
> Feel free to add my #1135 to the list 😀

It’s a bug, not a feature.

blueeagle69, 2020-10-29:
> > Feel free to add my #1135 to the list 😀
> 
> It’s a bug, not a feature.

Thank you, I was aware it was a bug.
Or the other hand, it's something that isn't currently supported.

SlackHacky, 2020-11-11:
Any update or clues to how to do on #524 Vertical Screen GUI ?

oldman4U, 2020-11-11:
Hello SlackHacky.

Sorry for the late answer, has been a busy day...

As far as I know, there is currently no developer working actively on this feature. Are you an engineer which could develop the functionality, or are you a user who would like to use the screen in vertical direction?

traffic-light, 2020-11-11:
@oldman4U my pr(#1209) resolves the next issues:

    resolves #634
    resolves #1072
    resolves #1178
    Resolves #358

oldman4U, 2020-11-11:
> @oldman4U my pr(#1209) resolves the next issues:
> 
> ```
> resolves #634
> resolves #1072
> resolves #1178
> Resolves #358
> ```

GREAT! Thank you for letting me know. I will update the tickets asap.

traffic-light, 2020-11-11:
@oldman thank you for your work!!!

traffic-light, 2020-11-15:
I'll cover #1166 in my PR (#1209 )

oldman4U, 2020-11-27:
Close this ticket now that it is pinned. Hopefully the Github app one day is able to show pinned tickets;-)

closed 2020-11-27 by oldman4U

readeral, 2020-12-09:
This is such a disaster @oldman4U

oldman4U, 2020-12-09:
Hi. What is a disaster?

readeral, 2020-12-09:
Consolidating issues like this. If you want something to list issues prioritise them, that's what Github Projects are for. Otherwise, duplicate projects get marked as duplicate and closed, while the original issue stays open. This is normal github etiquette.

oldman4U, 2020-12-09:
How long are you watching THIS repository?

readeral, 2020-12-09:
That's irrelevant, but if you truly want a dick-swinging contest in the realm of "how long" - you've been a user of github for only 12 months. Look into the projects feature, it's a far better approach than what you're trying to do.

oldman4U, 2020-12-09:
Just to add.

Duplicate items are NOT deleted here, like the once which are orphaned for more than 1 year and where the owner does not answer anymore. PR's are waiting for weeks to get pushed and without the new bot we would have 350+ tickets. The owner of this repository does not answer tickets and .....

It is relevant, because we have no access and we can not make the things right without the support of the owner of this repo and because you have no glue how it "works" here - but only how it works in other repos.

Please feel free to improve it the way you think it is right. I do not say that this is the best way to do it, so please show me how it works.

Thank you

readeral, 2020-12-09:
I've literally suggested that the Projects feature is how to go about it. Start here to learn: https://github.com/features#features-project-management

oldman4U, 2020-12-09:
So you want to tell me what I have to do!?

Why you are not doing it? Are you too good for this?

readeral, 2020-12-09:
I literally came to this repo from the Klipper repo to see how I might be able to contribute. I can't get a true sense of what's going on because a significant number of issues that are live issues are closed and compiled into this one and conversations are now dead. *shrug* If you really want a little kingdom, have at it.

oldman4U, 2020-12-09:
I want to hand this kingdom over to someone who knows better, maybe someone like you!

What I do not want is someone new to THIS repo, knowing how it should be, seeing that it is not like this but NEVER tried to make it right.

Please stay here for more than one week and see by yourself and then let's discuss, ok?

oldman4U, 2020-12-09:
Klipper support is highly requested and has nothing to do how this repo is used or administrated. So it would be great if you could take this over.

traffic-light, 2020-12-09:
@readeral maybe you should know that not everyone knows how github works.
A lot of users here do not know anything. just placing their request and don't know how to search for it.
With this list it could be easier for users to find out if their FR is already there.

Yes. It's  far from perfect. But it's better then nothing.

I agree that FR's shouldn't be closed because there is a list. It should be closed trough a PR.

And I think the question: How long you're watching this repo is a good question. You're just showing in a few messages that you DON'T know anything how this repo is.

I'm not trying to scare you away from this repo. But you need to know more about the average user here.

oldman4U, 2020-12-10:
Thank you.

3 Questions:
1. What makes you believe, that duplicated items are automatically deleted in this repository.
2. How can Project features be introduced into a repository as a new feature without being the owner of the same repository.
3. How is it going with the Klipper integration?

oldman4U, 2020-12-14:
readeral seems to be a nine day wonder....

I delete my comments above to keep this ticket usable.

traffic-light, 2020-12-14:
@oldman4U I think so... I'll remove my comments too

readeral, 2020-12-14:
Or maybe it takes more than 9 days to wrap my head around a new code base in a language I have barely used. I'm more likely to make a PR after two months. This is the pace of open source.

oldman4U, 2020-12-14:
Not talking about a PR but a short answer.

readeral, 2020-12-14:
Clearly wasn't worth my effort to stroke your ego

oldman4U, 2020-12-14:
Based on your introduction I expected not a lot, but even this was too
much.

oldman4U, 2020-12-23:
@Stellarspace @blueeagle69 @SlackHacky

Could you please be so kind and close your comments above as a part of my housekeeping work.

Thank you and Merry Christmas

radek8, 2021-01-30:
# 1298 On/Off terminalACK button
you can delete, it's done
Thank you

oldman4U, 2021-01-30:
Hi radek.

Thank you for letting me know.

Happy printing

bigtreetech, 2021-02-03:
@oldman4U Hello, Thank you very much for your help in the project. Do you have Facebook to communicate with? If it is not convenient to publish, you can send your contact information to our email “Tech@biqu3d.com”

oldman4U, 2021-02-26:
@readeral 

I hope you are doing fine.

It is now more than 2 months ago where you not only enlighted us with your appearance, but where you also told us that this amount of time is also the usual timeframe for a PR in the OS community. Now I ask myself how much substance has been behind your comments here and I am sure, also the whole Klipper community can not wait to see what you have done for them.

So please let me know when you will show us your contribution to this project.

Thank you

labotecno, 2021-02-26:
hello i get the G26 test leveling Working i need to finish some icon in other screen , how i can pull request ? [https://github.com/labotecno/BIGTREETECH-TouchScreenFirmware](url)

oldman4U, 2021-02-26:
Sounds great!

I use GitHub Desktop for pushing PRs. Try it out and let us know in case you have a problem.

labotecno, 2021-02-26:
i will add babystep menu when test leveling is enabled ! its working too ! like we can adjust offset z at the moment like prusa printer ! 

> 
> 
> Sounds great!
> 
> I use GitHub Desktop for pushing PRs. Try it out and let us know in case you have a problem.

ok i ill tell you soon thank you

Domush, 2021-03-01:
This consolidated list does not replace feature requests. 

None of the original feature requests should be closed as none of them have been implemented.

Closing the linked feature requests betrays the entire purpose of GitHub and makes it **more** difficult to fulfill feature requests, as they can no longer be properly tracked.

All of the linked feature requests must be reopened in their original requests. No wonder development of these TFT's has slowed, nobody can issue pull requests because all of the linked feature requests are closed.

oldman4U, 2021-03-01:
@Domush

why are you posting this here?

traffic-light, 2021-03-01:
@oldman4U I think he's new at this repo. So he doesn't know why this is here.

readeral, 2021-03-01:
He's posting it here because he actually understands GitHub

oldman4U, 2021-03-01:
readeral

Are you here to present the results of your great work?

I am sure your friend Edward understand GitHub and you both are invited to improve this repo and make it a "real" GitHub.

So far all you and your friend left here is hot air. Not a single line of new code or a bug which could be fixed because of you.

oldman4U, 2021-03-01:
@Domush 

I already tried to explain the situation here to your friend readeral before. Obviously he did not understand, so I'll try it again.

The time I joined this repository there have been almost 300 open tickets. Most of them have been not been updated for weeks and months, almost like in this repo: https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues

Unfortunately BIGTREETECH was too busy doing other things and therefore nobody was taking care. So some users like radek8, traffic-light,Tygrys-1, Gurmeet, and myself tried to organise this repo without the help of the owner. A few months later a bot was installed which closes tickets after 9 weeks of inactivity. To make sure that important tickets are still available without the need to search for closed tickets, two summary tickets have been opened.

To answer directly to your ticket above:

This consolidated list does not replace feature requests.
No, it does not. But what makes you believe that this could be the case?

None of the original feature requests should be closed as none of them have been implemented.
Agreed. It would be like this in an ideal world where the owner of this repo (or the user which opened a ticket) closes the ticket once it is not needed anymore. Unfortunately this is not the case.

Closing the linked feature requests betrays the entire purpose of GitHub and makes it more difficult to fulfill feature requests, as they can no longer be properly tracked.
Why does it make it more difficult to fulfill feature requests?
In an ideal world feature requests would be collected here, brought over to a development tool like JIRA once it is approved and closed automatically once fixed. But here it is sodom and gomorra, where bugs and features which are done are still in open tickets and vice versa and where users start tickets which are completely unrelated. See the SKR v1.3 or Marlin repo.

All of the linked feature requests must be reopened in their original requests. No wonder development of these TFT's has slowed, nobody can issue pull requests because all of the linked feature requests are closed.
Feel free to ask BTT, the owners of the tickets or someone else to open them again. I expect you to stay then here to take care - which would be great! We need people here actively working on this project. What we do not need here are users who tell others what to do and what is right and wrong, but not willing or able to contribute. Not sure I understand the need of an open feature request and the ability to issue a PR. Please explain.

I also have a question to you. It seems that you are a very smart developer with a successful history and I am sure you have a lot to do. What makes this little unimportant repository suddenly so important for you that you took the time to delight us with your presence?

Kind regards

oldman

oldman4U, 2021-03-01:
Sorry, last question.

Do you really exist or are you a fake person because readeral has no friends?

7 repositories, with a total of 2 contributions in the last year and the domain on your profile: http://trulyfallacious.com can be bought.

An active GitHub user looks different. But I am sure you have several accounts, right!?

readeral, 2021-03-01:
> readeral
>
> Are you here to present the results of your great work?
>
> I am sure your friend Edward understand GitHub and you both are invited to improve this repo and make it a "real" GitHub.
>
> So far [...]

If you look at my github activity, you'll note it's almost entirely javascript. I'm not a full-time programmer, and so learning C and C++ sufficiently to contribute is going to take me some time. Hopefully I'll be able to contribute to fix my issue in 2021, but BOY would I LOVE someone who is proficient in C to have VISIBILITY OF ISSUES so they can just jump in and say "Oh yeah I can do that" but no, the way you've molested this repo has made that almost impossible.

readeral, 2021-03-01:
Also, having looked at Domush's profile, see the "Pro" down the bottom there? That probably means they have been a github user for a while, and so paid to have private repositories (prior to them becoming free in the past year).

[31 more replies]