bigtreetech/BIGTREETECH-TouchScreenFirmware #1172: [FR] Different menu layouts for the larger touch panels

cbaugher, 2020-10-13:
As of right now the same menu layout is used for all of the TFT sizes, from 2.8 through 7.0.  This seems like a total waste of potential.  What good is having a larger screen if you aren't going to utilize the extra space for extra functionality?  The current layout works fine for the 3.5 and smaller screens, but for the 4.3 and up screens the extra size gets you literally nothing (unless you have terrible vision or REALLY fat fingers)

Personally I would like to see more features for IDEX printers and printers with multiple extruders.  It would be nice to show temperatures and fan speeds for all active extruders at once, instead of rotating through them in a single icon.  How about being able to color code extruders?  Or assign filament types to extruders?  How about easy access to a change tools button?

Or for any printer; how about being able to display an image of the part to be printed on the screen, like in the file list?  Or maybe show a picture of the actual printer in the move menu, that highlights the axis that is moving?  That last one is a bit frivolous.  But the point is there are so many cool things that could be done on a larger screen.  Yet they aren't.  Why not?

C|

oldman4U, 2020-10-13:
I could imagine that this will come and as soon as it is available some will argue they bought the bigger screen for bigger buttons.

Keep in mind that bigger in this case not automatically means more pixels and this somehow limits the possibility to show more.

cbaugher, 2020-10-13:
> 
> I could imagine that this will come and as soon as it is available some will argue they bought the bigger screen for bigger buttons.
> 
> Keep in mind that bigger in this case not automatically means more pixels and this somehow limits the possibility to show more.

Fair point about the screen resolutions.  I was assuming that the resolution would increase with the size of the screen, which appears to be not the case.  In fact, looking into this further I find myself rather disappointed.  The various sizes and resolutions are as such:

70 = 800 x 480
50 = 480 x 272
43 = 480 x 272
35v1 = 480 x 320
35v2 = 480 x 320
35v3 = 480 x 320
32 = 320 x 240
28 = 320 x 240
24 = 320 x 240

The 4.3 and 5.0 screens in fact have _worse_ resolution than the 3.5 screens.  I would have expected one of them to at least be in the middle, like 640 x 400.

C|

oldman4U, 2020-10-14:
Those displays are a designers nightmare.

From very very small to unusual big with different pixel sizes and aspect ratios. And all should support the same functionality.

For me the current design is not that bad and I hope the focus will be on stability, better navigation and more features in the foreseeable future.

my 2cent

cbaugher, 2020-10-15:
> Those displays are a designers nightmare.
>
> From very very small to unusual big with different pixel sizes and aspect ratios. And all should support the same functionality.
>
> For me the current design [...]

I also noticed the aspect ratios were not consistent.   [not impressed face]

Overall I would agree the current menu layout is generally ok.  I just kind of wish it was a little more friendly towards IDEX printers.  With some optimization I think it could be.

C|

oldman4U, 2020-10-19:
The displays are a bit more than 1 year available now, and for this short period of time they work very well. I guess more features are coming very soon, now the the basics are more or less covered.

Please close the ticket once you feel comfortable to do so. This would help to keep this system usable.

Thank you

cbaugher, 2020-10-21:
I guess I'll close this and then maybe create new feature requests for more specific things.  This is kind of vague and not really that actionable.

C|

closed 2020-10-21 by cbaugher

oldman4U, 2020-10-21:
Perfect.

Thank you