bigtreetech/BIGTREETECH-TouchScreenFirmware #1173: Unknown command: "M503 (S0)"

alibaaab, 2020-10-13:
After update from older version (Jun 17 2020) to latest (BIGTREE_TFT35_V3.0_E3.26.x.bin) on my TFT35-E3 am I am still getting echo about M503 like the title says. It is quite disturbing. I tried to add it to knownEcho to ignore it, but there is no more space on the board for this change. Is there some fix for it? Do you know why it is happening?

radek8, 2020-10-14:
Can you attach a photo?

alibaaab, 2020-10-15:
So, just after boot:
[image]

Detail screen of errors after boot (no other action between):
[image]

After first (basically after each command):
[image]

For now i just hide it by marlin
`#define DISABLE_M503   `

radek8, 2020-10-15:
You must enable M503 in Marlin

alibaaab, 2020-10-15:
Just for take this pictures i did this change to marlin 
`// #define DISABLE_M503   `
but i this case i got still this errors, so i hide it by:
`#define DISABLE_M503   `

radek8, 2020-10-15:
set:
#define EEPROM_SETTINGS     // Persistent storage with M500 and M501
//#define DISABLE_M503        // Saves ~2700 bytes of PROGMEM. Disable for release!

#define EXTENDED_CAPABILITIES_REPORT

alibaaab, 2020-10-15:
OK in configuration.adv.h i have 
`#define EXTENDED_CAPABILITIES_REPORT`
and in configuration.h i have 
`#define EEPROM_SETTINGS       // Persistent storage with M500 and M501
// #define DISABLE_M503        // Saves ~2700 bytes of PROGMEM. Disable for release!
// #define EEPROM_CHITCHAT       // Give feedback on EEPROM commands. Disable to save PROGMEM.
#define EEPROM_BOOT_SILENT    // Keep M503 quiet and only give errors during first load
#if ENABLED(EEPROM_SETTINGS)
  //#define EEPROM_AUTO_INIT  // Init EEPROM automatically on any errors.
#endif`

radek8, 2020-10-15:
you have to comment
// #define DISABLE_M503

alibaaab, 2020-10-15:
I have it in comment (enable feature) M503, but in this case i always got message 
`Unknown command: "M503 (S0)`
I want to use 503, but seem like problem is not on display but on board since same error i got by manual call gcode M503

radek8, 2020-10-15:
The problem will be in the Marlin configuration
please attach files
configuration.h
configuration_adv.h

alibaaab, 2020-10-15:
[Configuration_adv.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5383505/Configuration_adv.txt)
[Configuration.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5383507/Configuration.txt)

alibaaab, 2020-10-15:
i have other changes on pin setup for set properly 3d touch sensor, but should not be related

radek8, 2020-10-15:
Try deleting to comment on this volume
#define EEPROM_CHITCHAT

radek8, 2020-10-15:
Máte Desku BTT_SKR_V1_4_TURBO
Proč máte povolený ovladač pro konektor EXP3?
 #define CR10_STOCKDISPLAY
správě by jsi měl povolit 
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
a použít 2 ploché vodiče EXP1 a EXP2
Aby ti fungoval imulovaný Marlin mod

radek8, 2020-10-15:
Enable these 2 values
#define EMERGENCY_PARSER
#define EXTRA_FAN_SPEED

radek8, 2020-10-15:
if you are using the TMC2208 in UART mode
turn on
#define HYBRID_THRESHOLD

alibaaab, 2020-10-15:
nice now retro simulation is working, thanks radku but :
> Y stepper motor is much louder even when i return #define HYBRID_THRESHOLD back into comment
> 503 error still appear
For change confirm i add new version of configs:
[Configuration.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5387350/Configuration.txt)
[Configuration_adv.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5387351/Configuration_adv.txt)

radek8, 2020-10-15:
I understand that your display is already working in Marlin mode and the error with the M503 is still appearing. 
I didn't understand the other one. Can you be more accurate?

radek8, 2020-10-15:
If you have ender3 and you haven't changed the pulleys, you have the wrong stride lengths

[image]

radek8, 2020-10-15:
I would allow
#define RESTORE_LEVELING_AFTER_G28

radek8, 2020-10-15:
you still haven't commented on this choice !!!!!

//#define DISABLE_M503

radek8, 2020-10-15:
comment the choice
//#define SLIM_LCD_MENUS

alibaaab, 2020-10-15:
> 
> 
> If you have ender3 and you haven't changed the pulleys, you have the wrong stride lengths
> 
> [image]

This was changed because from some reason i got differents between controller coordinate agains real one. So i calculate these numbers and now it is occured because of that.

> 
> 
> comment the choice
> //#define SLIM_LCD_MENUS

Sorry for that, already did.

alibaaab, 2020-10-15:
Sorry for my insolence, but can we call?

radek8, 2020-10-15:
enable option
#define LONG_FILENAME_HOST_SUPPORT
#define SCROLL_LONG_FILENAMES

radek8, 2020-10-15:
I don't speak English

alibaaab, 2020-10-15:
> 
> 
> I don't speak English

but i think radek is czech name and i am speaking cesky ;)

radek8, 2020-10-15:
724064325

alibaaab, 2020-10-16:
I suggest for all to settup config according on rader's comments

closed 2020-10-16 by alibaaab

radek8, 2020-10-16:
The steps on the X-axis are strange.
Can you please disconnect the TMC2208 from the board and take a picture of the socket?
The x-axis does not appear to be in UART mode, but has a manually set microstep at 8x

hiskynet, 2022-04-04:
> You must enable M503 in Marlin

thanks helped in my e-step calibration for Geeetech A10M