bigtreetech/BIGTREETECH-TouchScreenFirmware #1175: BIGTREETECH TFT35 V2.0 BITMAPS MISSING ISSUE IN NON-UNIFIED UPDATE

fxlaptop, 2020-10-13:
AFTER FAILED UPDATES OF THE TFT35 V2 FIRMWARE I NOTICED THAT THE BMP FILES ITS ASKING FOR ARE MISSING FROM THE BMP FOLDER.. THE UNIFIED FIRMWARE HAS ALL OF THE BMPS BUT THE OTHER ONE, I GUESS ITS THE WORDED TYPE AND NOT PICTURE ICON UNIFIED BECAUSE IT WORKS.. SO THE ORIGINAL WORDED ICON BITMAP FOLDER HAS BITMAPS MISSING... @BIGTREETECH PLEASE FIX THE TFT35 MISSING NON-UNIFIED BITMAPS.. IF IM INCORRECT THEN PLEASE TELL ME WHAT THE ISSUE IS. I FOLLOWED ALL POSTED ADVICE ON HERE BUT NOTHING WORKS EXCEPT USING THE UNIFIED FOLDER BITMAPS..  I PREFER THE OTHER ONES, NOT THE UNIFIED PICTURE TYPES..  THANKS FOR ANY INSIGHT.

chapu96, 2020-10-14:
Saludos tengo el mismo problema, alguna solucion?, gracias

traffic-light, 2020-10-14:
I think your caps lock is broken fxlaptop.

But oh well...

Did you try to build your own firmware? That could solve some problems you've got.

And are you sure it's a 35v2? I don't see any pre-compiled version for that one.

oldman4U, 2020-10-14:
Not sure he really means PRECOMPILED but the NON Unified Theme in general.

Maybe it would be best to declare this theme EOL asap. There is an ongoing development of the firmware and new features sometimes require new bmp's. In case there is no user building the new templates, they will be missed and finally issues will come up.

WHO wants to maintain this old theme and make it compatible with the new TFT's and add the bmp's for new features?

Could you do this @fxlaptop? This would be great.

oldman4U, 2020-10-20:
@fxlaptop 

Have you seen the Hybrid Red Menu Material theme or The round miracle: #935 They come with all the bmp's and maybe you like them.

Please give it a try

Thank you

PR #1251 (2020-11-27): Changed the ReadMe to cover the new language.ini, added "The Round Miracle" and the "Hybrid Red" themes and removed the "Classic" theme
The new language.ini works different than it is described in the ReadMe today, which could confuse the user.
Added information about the cabling, because there are some users which do not understand how many cables to connect.

I removed the Classic menu, because it is not supported by the last firmware and added two themes which are supporting all TFTs from 24 to 70.

resolves #896 
resolves #1175
closed 2020-11-27 by bigtreetech