bigtreetech/BIGTREETECH-TouchScreenFirmware #1177: [BUG] esp8266 2.1.1 esp3d and ftf35 BIGTREE TFT35 V3.0 E3.26.x

islonina, 2020-10-14:
On AP setup i dont see wifi when connected to TFT35(correct pinout)
Working wifi when connecting only CHEN and power

**Expected behavior**

**Screenshots**
If applicable, add screenshots to help explain your problem.

ESP3D Firmware:
- ESP core version: [2.1.1]
- Version:[e.g 2.1b4]
- Web UI Version:[e.g 2.1b14]   
- Flash method:
Arduino
Platform IO
ESP8266 tool

- Any change done in FW
  No changes

Target Firmware:
 - Name: Marlin 2.0.7
- TFT35 E3.26.x 
 

Board used (please complete the following information):
 ESP8266 ESP-01
 Flash size: 1M 512kb

oldman4U, 2020-10-14:
ESP-01S or without S?

islonina, 2020-10-14:
> ESP-01S or without S?

Hard to tell no silkscreen on the board, but looking of the layout of the component without S

islonina, 2020-10-15:
for future reference. If u have old ESP-01 u need to solder CHEN and 3.3v on the ESP to make it "boot". Without that it only gets power. Weird hardware design but hay, it works now

closed 2020-10-15 by islonina