bigtreetech/BIGTREETECH-TouchScreenFirmware #1178: [BUG] usb printing not able to control Neopixel PRINTER_EVENT_LEDS on tft35 v3 and poss other screens with usb drive ports

CBDesignS, 2020-10-14:
Once we get neopixels actally working (by changing from the default pin P1_24 to another pin on the skr 1.4 turbo) and rewiring things etc I now find that there is no way to use PRINTER_EVENT_LEDS !!! when printing direct from a removable usb drive. It seems that the only way to use them is by printing from the sd slot or by command line with direct connection.

printing from usb drive just ignores M190 Sxx.x that turns the neopixel string on & starts the neopixels changing from blue to violet and M109 Sxxx.x that changes from violet to red when heating the extruders, and when the print starts turns them white. I know Marlin 2x does not support the end of print green colour neopixels as it uses sdPrint.complete to trigger it but a little gcode tweak can fix that.
this is not a marlin problem, this is a Touchscreen tft35 v3 (and possibily other screens) problem as it works on other devices by other makers according to the bug reports on the Marlin git.

Printing from the sd slot on the tft35 v3 works fine and seems to send the M109 & M190 commands correctly

traffic-light, 2020-10-14:
It's because your printing from tft. It only works in marlin.

I did write some code for it that it works in tft mode. The only problem is that I can't test it.

See #1072

CBDesignS, 2020-10-15:
sweet. I just noticed a strange thing, may have just been a coincidence. when I started the print from sd it was in "old style windows file names 8+3" naming and the pixel string worked as it should marlin wise but the next test print the screen started showing long file names and the neopixel failed to light but using the touch menu they still took the colour changes . I will test a little more tomorrow.

CBDesignS, 2020-10-15:
Done a little more testing and now its getting freaky. with lcd in marlin mode I select a test print from sd "good old dos 8+3 naming style" and start the print. Neopixels light as they should and the lcd jumps over to tft mode for the print and at the end of the print the neopixels turn green and the buzzer makes horrid noises instead of playing my beeper tones and freezes with bed and extruder still heating, press reset button and lcd boots back in marlin mode. I will flash last btt compiled ver of tft35 v3 firmware to see if any of the recent user updates have messed up. or maybe eeprom has corrupt.. who knows

traffic-light, 2020-10-15:
Is that with my code?

It could be a older bug

traffic-light, 2020-10-15:
I also hope it will arrive soon.

It's already 1.2 months ago that they shipped it out from the airport. An I didn't knew it was such a long flight 🤣:

 2020-09-15
depart from airport
17:47
2020-08-31
Depart from facility to service provider.
21:28
2020-08-25
Shipment arrived at facility and measured.
21:28
2020-08-25
4PX picked up shipment.

CBDesignS, 2020-10-15:
the code I used was downloaded from the BIGTREETECH-TouchScreenFirmware git on28th Sept 2020. and not sure if #1058 fixed the screen jumping as that is what was happening to me yesterday. I have turned serial off and it is behaving. I will turn it back on and see if #1058 was fixed on v3 screens

CBDesignS, 2020-10-15:
@traffic-light I cloned your git from the link you provided and compiled then uploaded to my tft35 v3. The added menu entry looks ok, pic included but it seems that abl resets to off after a power off or reset, my temps never changed even tho the extruder was at 150c and the bed was 50c. no sign of neopixel activity either. Also when I try to jump over to Marlin mode I just get a blank screen.
[image].
I don`t know if I screwed up somewhere but its not looking good atm. will re-compile your code and try again.
[image]

traffic-light, 2020-10-15:
It could be something wrong with merging and adjusting.

I've got to take a look at it I think...

oldman4U, 2020-10-20:
Any news on this? Feature or bug?

CBDesignS, 2020-10-20:
still as is @oldman4U . waiting on @traffic-light and his screen to try and push us forward. I just use marlin mode if I want the neopixels on.

oldman4U, 2020-10-22:
Is this about external neopixels or the knobs LEDs?

If helpful, I could give it a try.

Let me know

oldman4U, 2020-10-23:
Btw. the jump from Marlin to Touchscreen was  a bug and has been fixed with Gurmeets last PR.

In your case this was almost a feature;-)))

CBDesignS, 2020-10-24:
Screen jumping may be fixed or not but neopixel is not working in tft mode the same way as it works in Marlin mode. I have just compiled the current code from the master branch and I still have the same problem with neopixels not working as they should.
Once this print is complete I will enable serial data again and see if screen jumping is fixed. not that I can get serial connection to an Orangepi Zero working on uart3 or uart4 that runs my Octoprint server (only as a monitoring device so I can make sure it`s running ok when I am not in the room)

traffic-light, 2020-10-24:
It's not working because it's not implemented, yet.
I received my screen today and the GitHub is updated with a new language implementation. So i'll work on it, again, when I have spare time

oldman4U, 2020-10-24:
> Screen jumping may be fixed or not but neopixel is not working in tft mode the same way as it works in Marlin mode. I have just compiled the current code from the master branch and I still have the [...]

This why I only mentioned the jump issue and not the neopixel.

CBDesignS, 2020-10-25:
yea screen jumping seems to be fixed oldman4U and it`s for external neopixels (the tft35 v3 does not have an onboard knob led)

traffic-light, 2020-10-25:
Yes, it will be for neopixels.
I'm implementing it for neopixel AND the rotary knob.

oldman4U, 2020-10-25:
Because some TFTs do have an illuminated knob😁

Thank you

CBDesignS, 2020-10-25:
again then I have been fed the incorrect information. I was told that the v3 did not have an led knob and it was only the E3 that had one fitted. It seems that everything I was told about btt has proven to be utter bull crap.

traffic-light, 2020-10-26:
When the only thing left is that you want the neopixels show the heat up state, go from blue to red, white while printing and green when finished, you can close this ticket 

It's already covered in another ticket and in the pinned fr request list.

I'm currently working at it when I've got some time.

See my git.

oldman4U, 2020-10-28:
> again then I have been fed the incorrect information. I was told that the v3 did not have an led knob and it was only the E3 that had one fitted. It seems that everything I was told about btt has proven to be utter bull crap.

As far as I know, the TFT35 has no encoder LED on board, so what traffic-light is working on will work with the TFT35 E3 v3 and I believe all with all new TFT bigger than that.

Sorry in case I confused you with my answers.

@traffic-light Does your code also include the general ability to control a neopixel LED, independent of the screen? Sorry that I ask...

CBDesignS, 2020-10-28:
@oldman4U reading the PR **traffic-light** states :- _The knob-led, when the user got one, will also be changing color_ so I take it that when he gets the code and related bug fixed then it should work for the neopixels and the knob led if the screem has one and if no knob led then just the neopixels.

traffic-light, 2020-10-28:
That's correct!

Outlook for Android<https://aka.ms/ghei36> downloaden

CBDesignS, 2020-11-11:
A pleasure to close this ticket now that it has been resolved by @traffic-light and his skillfull coding also not forgetting the helpers along the way.

closed 2020-11-11 by CBDesignS