bigtreetech/BIGTREETECH-TouchScreenFirmware #1179: [FR] (feature request title)

Portzal, 2020-10-15:
Including a toggle in .h for multiple Z steppers. 
This will allow separate settings in TMC options for Z2,Z3,Z4 etc.

Portzal, 2020-10-16:
I have just noticed that there is no option to turn off StealthChop either.

radek8, 2020-10-17:
If your drivers use UART mode, you have the option to set HYBRID_THRESHOLD from the display menu

radek8, 2020-10-17:
As for multiple Z-axis motors.
The display FW should be able to set the same TMC controller parameters to two motors on the same axis. But I don't have this configuration to verify it.

Portzal, 2020-10-20:
I have made some progress with sorting this issue for myself, but I am still working through the code to find the relevant areas to update. [image]

radek8, 2020-10-20:
The current parameter for the second Z axis is not displayed on the touchscreen because both stepper motors are assumed to operate with the same setting.
Therefore, use unmodified FW displays.
- Check the current to motor Z and Z2 via the terminal with command M906.
- Change the Z value via the touchscreen.
- Check via the terminal with the M906 command whether the current has also changed on the Z2 stepper motor

If it works as I write, it's the expected state, if not, there's a mistake somewhere :-)

Portzal, 2020-10-20:
@radek8 I am already well aware of everything you have commented. I prefer the functionality I have in the Marlin simulation mode and see no reason why the touch screen mode does not have the same settings. [image]

radek8, 2020-10-21:
You did not answer me if the current setting works as I described or there is a problem with it.

I understand that more views in Marin suit you. If it is better for you, use it.
BTT produces HW. The software does not solve much, it is rather compiled by the users themselves according to their needs. You can set up to 6 extruders in the configuration file. Unfortunately, practically only 2 can be configured and with other axes X, Y, Z it is similar.

The space for the code in TFT is also limited and the requirements and functions are many. 95% of users of this display do not have a printer with multiple axes or extruders. Those who do not have a weaving printer consider it unnecessary to have other parameters for axes in the menu. Many users also criticize the possibility of setting the current for the second extruder because it does not have it.

The current logic does not seem so bad to me. If I have two stepper motors on the axis, they must be the same and I will need to set the same parameters on them. This can also prevent incorrect operation if you forget to set the current on the second motor and he would then lose the steps.

radek8, 2020-10-21:
I want a bid for the second extruder, you want a bid for the second z axis
someone will come up with the fact that it has 4 motors on the Z axis and someone else will have two motors on all axes ....
submenu in the Parameters menu currently has the ability to edit 5 items. It would have to be put in more offers. That would be confusing. Or the entire Parameters menu would have to be redesigned.

Portzal, 2020-10-22:
Another missing piece in the puzzle appears to be the ability to adjust Junction deviation.

radek8, 2020-10-22:
PR is already ready for the Junction parameter #1192

Can you answer my question above?

You look for bugs and you call it a circus.
You can help clear the errors or stop using the display. No one is forcing you.

radek8, 2020-10-22:
The display lacks a lot of functions, but they can be supplemented.
So do you want to get involved and help improve and test new features?

radek8, 2020-10-22:
> Udělal jsem určitý pokrok v třídění tohoto problému pro sebe, ale stále pracuji v kódu, abych našel příslušné oblasti k aktualizaci. [image]

Send the file you edited, I'll tell you what you're missing :-)

radek8, 2020-10-22:
> @ radek8 Už jsem si dobře vědom všeho, co jsi komentoval. Dávám přednost funkcím, které mám v simulačním režimu Marlin, a nevidím důvod, proč režim dotykové obrazovky nemá stejná nastavení.![obraz](https://user-

Marlins compiles for a specific printer configuration.
The display must be as universal as possible in order to be able to control as many different printers with different configurations ...

closed 2020-12-28 by stale[bot]