bigtreetech/BIGTREETECH-TouchScreenFirmware #1180: BIG Problem with config.ini

CBDesignS, 2020-10-15:
BIG Problem with config.ini

the last lines in config.ini are (cut n paste) from the latest build
#### Cancel G-code - run this G-code after canceling print
# maximum length 50 characters
cancel_gcode:M104 S0\nM140 S0\nG28 XY R10\n

why is there a G28 homing command in the cancel_gcode ?????

If you cancel a build say half way through you will have an item on the build platform, if you use 3d touch and poss other devices as Z endstop and they uses the center of the build plate to activate you will run into a slight problem of crashing into the printed part on the platform when it tried to home z ..

radek8, 2020-10-15:
G28 is with parameter XY
Therefore, the Z axis will not be guided home

radek8, 2020-10-15:
If this does not suit you, you can change the sequence of characters according to your needs.

CBDesignS, 2020-10-15:
must be a bug then as on my freaky machine it homes xy & z if I have it enabled and try to use it.

CBDesignS, 2020-10-15:
I have changed it @radek8. because I am trying to rebuild this old machine with modern hardware there are times I need to scrap a test print and without that code active the machine just stops in it`s tracks spewing fillament out into what it was printing. This was the second time I have canceld a print and it homes all 3 axis even tho it should only home x & y. it is the last time tho. 
It is maybe related to all the rest of the bugs in the touch tft side of the tft35 v3 screens.

CBDesignS, 2020-10-15:
I am no coder but all these strange bugs could they all originate from thing. is there a white space bug or something corrupting a command with spaces in it ?? 
nG28 XY R10
maybe it is only sending the G28 and the space between XY is going corrupt or even not sending the XY part of the command

radek8, 2020-10-15:
Check Marlin if you do not have the Z axis enabled in this row

#define EVENT_GCODE_SD_ABORT "G28XY"      // G-code to run on SD Abort Print (e.g., "G28XY" or "G27")

radek8, 2020-10-15:
I tested with myself. if I interrupt printing, only home position guidance is performed on the X and Y axis only

CBDesignS, 2020-10-15:
> 
> 
> Check Marlin if you do not have the Z axis enabled in this row
> 
> #define EVENT_GCODE_SD_ABORT "G28XY" // G-code to run on SD Abort Print (e.g., "G28XY" or "G27")

yup already enabled.  I have just re compiled the latest source and noticed a slightly different config.ini but the only change was fan related so I doubt it has much to do with this problem.
#### Fan Controller Count
# Options: 0 to 2
fan_ctrl_count:0

I am about to re flash the tft and see how it goes.