bigtreetech/BIGTREETECH-TouchScreenFirmware #1181: [FR] In Terminal, move the letter F from the ABC page to 123 page

StarNamer, 2020-10-16:
**Is your feature request related to a problem? Please describe.**
I've recently refitted my 3DTouch a couple of times and each time needed to type the command G1 F60 Z0. But 'F' is on the alphabetic screen so requires 3 screen taps instead of 1.

**Describe the solution you'd like**
Move the ABC to bottom right and move the SPACE down to where ABC was. Move Z to where SPACE was. Move E to where Z was. Add F where E was.

Also rearrange ABC screen similarly and duplicate E, S & T so they are available of both screens since these are common letters in text.

`
1 2 3 M G DEL`
`
4   5   6   X   Y   Z`
`
 7   8   9   E   F  SPC`
 `
.   0   -   S   T  ABC`

 `
A   B   C   D   E  DEL`
`
 H   I   J   K   L   N`
`
 O   P   Q   R   S  SPC`
`
 T   U   V   W   Y 123`
 

**Describe alternatives you've considered**
Keep switching back and forth between screens

**Additional context**
There is space for the additional F key on the numeric screen and it is commonly used in GCode commands

oldman4U, 2020-10-16:
Hi.

As far as I know, you are a very good programmer. Why not changing it and push a PR?

oldman4U, 2020-10-19:
I started a FR list which covers all the requests from the past. Please close this ticket and let me know if you are going to change this or not, so I can mark it correctly.

Thank you

oldman4U, 2020-10-20:
I did my part, seems you are working also on it;-)

The FR list is now pinned, so it is easily to see all FR. Please check #1170 and let me know.

Thank you

StarNamer, 2020-10-20:
@oldman4U I've actually produced a version which has the F key on the 123 page. But since I have a TFT35 I went a bit further and added I, J, P & R as well by adding an extra column. See [this commit](https://github.com/StarNamer/BIGTREETECH-TouchScreenFirmware/commit/a67bf8e69a4dc05d1e2450a2ac8a889a8a5f9450#diff-ad56d3a5dadb5fd3edad7e4de4f3198f76ad94ca0faa98f237c6b95cd771dea6).

However, it occurred to be that the TFT24 etc may not be able to reasonably support a 7 by 4 grid of keys. Should I produce a simpler version or just submit a PR for this?

oldman4U, 2020-10-21:
Hi.

Sorry, I am no developer, so I asked someone from the dev. team to help. Hope this will happen.

THANK YOU

StarNamer, 2020-10-21:
@oldman4U Thanks. I submitted a [PR ](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1191 )anyway.

oldman4U, 2020-10-21:
Saw it, GREAT!

guruathwal, 2020-10-22:
increasing the grid cell count is not good for small displays like TFT24 because the touch does not work well near the edges.  There is an empty cell on the first keyboard view. you can use it to place the required `F` alphabet.

oldman4U, 2020-10-23:
Thank you Gurmeet.

I have ordered a TFT24 to see how it works. But I can see that the current solution is already very small on the TFT28 and adding an extra column makes it almost unusable. Don't think this be much better on the 24...;-(

I believe it would be a good idea to take Gurmeets approach and add F to the first page and not to add an extra column.

StarNamer, 2020-10-24:
As I commented on the [PR](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1191) I've already started looking at varying the grid size depending on screen size (320x240, 480x320, etc). I am currently thinking of 6x4 (as now + F) for the 320x### sizes and 7x4 for the 480x### sizes. The TFT70 (800x480) could probably just have a 10x6 and only 1 page!

oldman4U, 2020-10-28:
This sounds great. Is your PR complete and is there anything I could test for you?

StarNamer, 2020-10-28:
I've coded a version with different layouts for different screen sizes, but am still testing and know I need to make a few more changes to clean up the code and make it maintainable going forward. At the moment the TFT70 layout would be 10x5, but since the only punctuation which anyone might use outside of filenames is "!" & "#" for M32, I don't see any point in adding any more than I have unless someone asks.

I hope to push a new commit and hence update this PR in the next couple of days - real work and outside life permitting!

By the way, I'm very interested in [PR #1171](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1171), [Ability to Scroll GCode Command Display Window #1112
](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1112) and am trying make sure there are no conflicts with that.

oldman4U, 2020-11-08:
Hello StarNamer.

Thank you for your contribution. Today I found some time to check it on the 24,28 and 35 TFT's I have and wanted to send you the attached pictures.

Would be great if you could find some time for additional contributions.

Kind regards

PS: Maybe you want to close this ticket;-)
[image]
[image]

PR #1231 (2020-11-10): Improved Terminal menu based on TFT size and resolution
Based on the idea of StarNamer on his PR #1191, just to speed up the merge on BTT main repo due to my last changes already applied on Terminal menu.
I don't have any TFT sample from BTT to test the layout on all the TFT variants, so I ask you to test in your TFT and eventually provide your feedback. I simple verified that the 3 layouts are correctly displayed in my TFT and the terminal correctly works.
Thanks to StarNamer!

**IMPROVEMENTS:**

* **Improved Terminal menu:**
 1) three different layouts based on TFT size and resolution:
  * T [... cut]
closed 2020-11-10 by bigtreetech